Koenkk/zigbee2mqtt #19424: [New device support]: Mercator Ikuu Combination sensor (modelID: 'TS0601', manufacturerName: '_TZE200_agumlajc')

mscottco, 2023-10-26:
### Link

https://www.ikuu.com.au/product/combination-sensor-2/

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x943469fffe44b9fe","nwkAddr":60197,"manufId":4098,"manufName":"_TZE200_agumlajc","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"y\u0005�,iy\u0005�,iy\u0005�,iz\u0005�,i�\u0005�,i�\u0005�,i�\u0005�,i�\u0005�,i","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_agumlajc","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":65,"powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1698279424090,"defaultSendRequestWhen":"immediate"}

### Comments

Hi, I'm hoping for some assistance with creating a converter to support the Mercator Ikuu Combination Sensor. I've tried to write a converter (copied below) without much success. I also had a look at the existing mercator products in the codebase but they all look different to the instructions on the "Support New Devices" instructions.

The device has the following functions:
- PIR sensor (possibly datapoint 104)
- Temperature
- Humidity (possibly datapoint 2)
- Lux (
- A relay for turning a light on and off. This appears to match to datapoint 105 where 0 = On, 1 = Off, 2 = Auto (based on motion)

I've scanned the logs and found the following entries to assist with the datapoint mappings:
Datapoint 1 not defined for '_TZE200_agumlajc' with value 222
Datapoint 2 not defined for '_TZE200_agumlajc' with value 58
Datapoint 10 not defined for '_TZE200_agumlajc' with value 500
Datapoint 11 not defined for '_TZE200_agumlajc' with value -100
Datapoint 12 not defined for '_TZE200_agumlajc' with value 88
Datapoint 13 not defined for '_TZE200_agumlajc' with value 18
Datapoint 14 not defined for '_TZE200_agumlajc' with value 2
Datapoint 15 not defined for '_TZE200_agumlajc' with value 2
Datapoint 101 not defined for '_TZE200_agumlajc' with value 0
Datapoint 102 not defined for '_TZE200_agumlajc' with value 1000
Datapoint 103 not defined for '_TZE200_agumlajc' with value 0
Datapoint 104 not defined for '_TZE200_agumlajc' with value 0
Datapoint 105 not defined for '_TZE200_agumlajc' with value 1
Datapoint 106 not defined for '_TZE200_agumlajc' with value 50
Datapoint 107 not defined for '_TZE200_agumlajc' with value 1
Datapoint 108 not defined for '_TZE200_agumlajc' with value 0
Datapoint 109 not defined for '_TZE200_agumlajc' with value 0

Mercator have been approached to provide the DPID's and datapoint specification although I'm not expecting much luck...

Happy to provide additional information as needed.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_agumlajc'}],
    model: 'SSWMPIR-ZB',
    vendor: 'Mercator Ikuü',
    description: 'Combination Sensor',
    fromZigbee: [fz.occupancy, fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.occupancy(), e.temperature(), e.humidity()],
    configure: tuya.configureMagicPacket,
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            //[1, ],
            [2, 'humidity', tuya.valueConverter.raw],
            //[10, ],
            //[11, ],
            //[12, ],
            //[13, ],
            [101, 'temperature', tuya.valueConverter.raw ], //temperature?
            //[102, ],
            //[103, ],
            [104, 'occupancy', tuya.valueConverter.raw],
            [105, 'light setting', tuya.valueConverterBasic.lookup({'On': 0, 'Off': 1, 'Auto': 2}) ] //Light motion setting
            //[106, ],
            //[107, ],
            //[108, ],
            //[109, ],
        ],
    },
};

module.exports = definition;
```

mscottco, 2023-10-26:
I had another stab at getting the code to work and made some progress (edited version in original comment above). What works:
- motion
- humidity
- temperature (although it appears to be reading ~5 degrees higher than other sensors in the room)

What I'm having issues with:
- illuminance: A value shows under the device 'exposes' tab on z2m but the value hasn't showed up in Home Assistant
- Getting the relay to work. It seems to be exposed on datapoint 105 with the values mapped above but I haven't figured out a way you can set that from z2m.
- The motion is also very quick to clear, is there an easy way to control the timeout?

roastcom, 2023-11-15:
> * illuminance: A value shows under the device 'exposes' tab on z2m but the value hasn't showed up in Home Assistant
Having exactly the same issue with illuminance on a different device. Let me know if you've managed to work it out.

sabaramo, 2024-06-03:
> What I'm having issues with:
> 
> * illuminance: A value shows under the device 'exposes' tab on z2m but the value hasn't showed up in Home Assistant

I reckon both illuminance and motion timer is not exposed. See my comment below for my theory. 

> * Getting the relay to work. It seems to be exposed on datapoint 105 with the values mapped above but I haven't figured out a way you can set that from z2m.

I dont think the relay is exposed either. It's probably just electrically coded to align with on, off or auto. 

> * The motion is also very quick to clear, is there an easy way to control the timeout?

I believe both illuminance and timer are controlled by the small knobs above the on/off button. They're probably not exposed via mqtt but rather hard coded to the knob on the board.

Tekno-man, 2024-07-14:
Did anyone end up getting anywhere with this?

mscottco, 2024-07-20:
Below is the current converter file I'm using which seems to do well. Exposes to homeassistant as shown in the below picture.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_agumlajc'}],
    model: 'SSWMPIR-ZB',
    vendor: 'Mercator Ikuü',
    description: 'Combination Sensor',
    fromZigbee: [fz.occupancy, fz.temperature, fz.humidity, fz.illuminance, fz.on_off, tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.occupancy(), e.temperature(), e.humidity(), e.illuminance(), e.light(), 
        e.enum('state', ea.STATE_SET, ['ON', 'OFF', 'AUTO']).withDescription('Light Mode')
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'genOta', 'genTime', 'genGroups', 'genScenes']);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'manuSpecificTuya']);
    },
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'humidity', tuya.valueConverter.raw],
            //[10, ],
            //[11, ],
            //[12, ],
            //[13, ],
            [101, 'illuminance', tuya.valueConverter.raw ], //illuminance
            //[102, ],
            //[103, ],
            [104, 'occupancy', tuya.valueConverter.trueFalse1],
            [105, 'state', tuya.valueConverterBasic.lookup({'ON': tuya.enum(0), 'OFF': tuya.enum(1), 'AUTO': tuya.enum(2)}) ] //Light motion setting
            //[106, ],
            //[107, ],
            //[108, ],
            //[109, ],
        ],
    },
};

module.exports = definition;
```

[image]

Koenkk, 2024-07-20:
@mscottco would you mind making a pr to add out of the box support for this device?

therealchunkyd, 2024-08-28:
Hi All - Did anyone have luck with this? @mscottco as a newbie to z2m, do I save the content of your converter to a .js file next to the configuration.yaml, add as external converter in z2m, the restart?

mscottco, 2024-08-30:
@Koenkk Sorry it's taken so long - life gets busy and programming fell by the wayside. I've just submitted a pull request for the file, I hope it goes through but will monitor / fix the CI issues as best I can.

@therealchunkyd - that sounds about right. As I recall, create the js file then in your configuration you have to tell zigbee2mqtt to load the file.

closed 2025-01-12 by github-actions[bot]

ausfas, 2025-04-20:
> [@Koenkk](https://github.com/Koenkk) Sorry it's taken so long - life gets busy and programming fell by the wayside. I've just submitted a pull request for the file, I hope it goes through but will [...]

Any idea how do I add that to ZHA in my HA ?

volvok, 2025-10-06:
Updated for latest versions of z2m 
https://github.com/Koenkk/zigbee-herdsman-converters/issues/10220