Koenkk/zigbee2mqtt #19427: Lights show wrong state after power cycle

Brumhilde, 2023-10-26:
### What happened?

We have some table/floor lights that we sometimes turn on by using the switch on the cord. By turning the switch off and on the lights turn on. Z2M shows the lights as off when they are on after the power cycle. I am just migrating from Deconz to Z2M and noticed this difference in behaviour. In Deconz the state is updated immediately. I have used Deconz and Conbee for several years with this scenario. I have tried different configurations with Availability but it is not working as expected. I use mainly Trådfri lights.

### What did you expect to happen?

I expect Z2M to show the correct state (ON) in the dashboard.

### How to reproduce it (minimal and precise)

Power cycle a light that turn on when power is restored. The light is turned on but Z2M show state OFF in dashboard.

### Zigbee2MQTT version

1.33.1 commit 7e63039

### Adapter firmware version

0x26580700

### Adapter

Conbee 2

### Debug log

Nothing in the log when I power cycle the light.

Koenkk, 2023-10-27:
Can be fixed by enabling the availability feature: https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#state-retrieval

Brumhilde, 2023-10-27:
I specifically write that Availability does not work. For availability to retrieve state the device has to go offline first and that requires it to be powered off for more than 1 minute. Setting Availability timeout to 0 does not work either.

Koenkk, 2023-10-27:
can you provide the debug log when the device reconnects to power?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Brumhilde, 2023-10-27:
Nothing is logged when I power cycle the lights

Koenkk, 2023-10-28:
I would expect that you see a `deviceAnnounce` in the debug log, otherwise z2m has no way of knowing that the bulbs turned on. What kind of bulbs are this?

Brumhilde, 2023-10-28:
I have tried with these trådfri lights:
https://www.zigbee2mqtt.io/devices/LED1545G12.html#ikea-led1545g12 Firmware 2.3.095
https://www.zigbee2mqtt.io/devices/LED1842G3.html#ikea-led1842g3 Firmware 2.3.093

I can give it a new try on Tuesday.

Brumhilde, 2023-10-30:
If I toggle the power off/on nothing is shown in the log. I have to wait for the device to go offline before returning power for the device to "reconnect". To grab the below log I set availability timeout to 1 minute and waited for device to go offline.

Warning 2023-10-30 08:14:11Failed to ping 'EOS Floor Light' (attempt 1/2, Read 0x000b57fffe9a8d5a/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (98)))
Warning 2023-10-30 08:14:24Failed to ping 'EOS Floor Light' (attempt 2/2, Read 0x000b57fffe9a8d5a/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (99)))
Debug 2023-10-30 08:14:27Active device 'EOS Floor Light' was last seen '1.80' minutes ago.
Info 2023-10-30 08:14:27MQTT publish: topic 'zigbee2mqtt/EOS Floor Light/availability', payload '{"state":"offline"}'
Debug 2023-10-30 08:14:31Device 'EOS Floor Light' reconnected
Info 2023-10-30 08:14:31MQTT publish: topic 'zigbee2mqtt/EOS Floor Light/availability', payload '{"state":"online"}'
Debug 2023-10-30 08:14:33Retrieving state of 'EOS Floor Light' after reconnect
Debug 2023-10-30 08:14:33Received Zigbee message from 'EOS Floor Light', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-10-30 08:14:33MQTT publish: topic 'zigbee2mqtt/EOS Floor Light', payload '{"brightness":190,"color_mode":"color_temp","color_options":null,"color_temp":283,"linkquality":151,"power_on_behavior":"on","state":"ON","update":{"installed_version":587814449,"latest_version":587814449,"state":"idle"},"update_available":false}'

Brumhilde, 2023-10-30:
I found the reason for the unexpected behaviour. Since I am migrating now I have permit_join true in my configuration. If I disable join the device announce itself as expected:

Debug 2023-10-30 08:23:34Device 'EOS Floor Light' announced itself
Info 2023-10-30 08:23:34MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"EOS Floor Light","ieee_address":"0x000b57fffe9a8d5a"},"type":"device_announce"}'
Info 2023-10-30 08:23:34MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"EOS Floor Light"},"type":"device_announced"}'
Debug 2023-10-30 08:23:36Retrieving state of 'EOS Floor Light' after reconnect
Debug 2023-10-30 08:23:36Received Zigbee message from 'EOS Floor Light', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-10-30 08:23:36MQTT publish: topic 'zigbee2mqtt/EOS Floor Light', payload '{"brightness":190,"color_mode":"color_temp","color_options":null,"color_temp":283,"linkquality":191,"power_on_behavior":"on","state":"ON","update":{"installed_version":587814449,"latest_version":587814449,"state":"idle"},"update_available":false}'

Brumhilde, 2023-10-31:
Yeah I'm happy with the solution.

closed 2023-10-31 by Brumhilde