Koenkk/zigbee2mqtt #19430: HA 2023.11.0b0: Entity sensor.mini_cave_power_outage_memory cannot be added as the entity category is set to config

hmmbob, 2023-10-26:
### What happened?

Running HA 2023.11.0b0, these errors popped up and entities aren't added to HA anymore:

```py
2023-10-26 13:49:36.826 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform mqtt
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.grijze_lamp_power_outage_memory cannot be added as the entity category is set to config
2023-10-26 13:49:36.830 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform mqtt
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.pubercave_power_outage_memory cannot be added as the entity category is set to config
2023-10-26 13:49:36.835 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform mqtt
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.mini_cave_power_outage_memory cannot be added as the entity category is set to config
2023-10-26 13:49:36.841 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.grijze_lamp_power_outage_memory cannot be added as the entity category is set to config
2023-10-26 13:49:36.849 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.pubercave_power_outage_memory cannot be added as the entity category is set to config
2023-10-26 13:49:36.852 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.mini_cave_power_outage_memory cannot be added as the entity category is set to config
```

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20220103

### Adapter

Sonoff

Koenkk, 2023-10-27:
I think this is fixed after https://github.com/Koenkk/zigbee2mqtt/pull/19434 / https://github.com/Koenkk/zigbee2mqtt/pull/19435 , please check with the latest dev-branch (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

hmmbob, 2023-10-28:
Unfortunately, it is not - but I also see no changes in #19434 relating to the specific sensor that's triggering the error in my case (`_power_outage_memory`):

```
2023-10-28 14:15:47.826 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform mqtt
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.mini_cave_power_outage_memory cannot be added as the entity category is set to config
2023-10-28 14:15:47.870 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.mini_cave_power_outage_memory cannot be added as the entity category is set to config
```

hmmbob, 2023-10-28:
I've manually deleted the unused `sensor.xxx_power_outage_memory` entities from HA, and now the error is gone.

@frenck Appears that some old config might still be lingering around in HA after the updates?

hmmbob, 2023-10-28:
Ah, not really: the deleted sensors just come back in HA but with "disabled" status, and when made available again the error is back:

```
2023-10-28 14:54:56.064 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform mqtt
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.mini_cave_power_outage_memory cannot be added as the entity category is set to config
2023-10-28 14:54:56.070 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1021, in add_to_platform_finish
await self.async_internal_added_to_hass()
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 279, in async_internal_added_to_hass
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.mini_cave_power_outage_memory cannot be added as the entity category is set to config
```

frenck, 2023-10-29:
> Appears that some old config might still be lingering around in HA after the updates?

> discovery information: for the SENSOR: homeassistant/sensor/0x84ba20fffe65518d/power_outage_memory/config (which should be diagnostic)

Well, Home Assistant isn't the one filling the discovery topic @ MQTT, that is Z2M. I'm not sure when those update?

#19434 set the entity category for every sensor that might have been `config` to `diagnostic` in the discovery payload.

frenck, 2023-10-29:
Nevermind, Koen pointed me into the right direction. My change is incomplete. Will make another PR.

../Frenck

hmmbob, 2023-10-29:
Thanks as always, @frenck

hmmbob, 2023-11-01:
'Fixed by #19474

closed 2023-11-01 by hmmbob