Koenkk/zigbee2mqtt #19432: Device recognized correctly but stange behaviour -> related to #18640

rickoltan, 2023-10-26:
### What happened?

MHCOZY YWB 4ch-RF is recognized correcly but shows strange behaviour. Was already mentioned in #18640.

If all switch are all off, nothing happens.
If a switch is turned on, that switch will switch continuously between on/off until after a certain while it stays on off.
Final state is always off.

### What did you expect to happen?

That switching the switch on in UI (or by hand on the device) will keep the switch in the on position and not starting switching continuously and ending in a off state.

### How to reproduce it (minimal and precise)

Happens always. Just connect the device to Zigbee2MQTT and try to switch on one of the switches.

### Zigbee2MQTT version

 1.33.1 commit: 7e63039

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Debug log

2023-10-26 17:49:34Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:34MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:34+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Debug 2023-10-26 17:49:34Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:34MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:34+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Debug 2023-10-26 17:49:34Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:34MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:35+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Debug 2023-10-26 17:49:35Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:35MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:35+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Debug 2023-10-26 17:49:35Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:35MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:35+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Debug 2023-10-26 17:49:36Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:36MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:36+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Debug 2023-10-26 17:49:36Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:36MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:36+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Debug 2023-10-26 17:49:36Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:36MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:37+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Debug 2023-10-26 17:49:37Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:37MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:37+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Debug 2023-10-26 17:49:37Received Zigbee message from 'swpopo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-10-26 17:49:37MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"202 [... cut ...]
Info 2023-10-26 17:49:39MQTT publish: topic 'zigbee2mqtt/swpopo', payload '{"device":{"applicationVersion":80,"dateCode":"","friendlyName":"swpopo","ieeeAddr":"0xa4c138fefb7cccaa","manufacturerID":4417,"manufacturerName":"_TZ3000_u3oupgdy","model":"TYWB 4ch-RF","networkAddress":5443,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2023-10-26T17:49:40+02:00","linkquality":255,"power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'

badrpc, 2023-10-29:
I was trying (unsuccessfully) to configure this device into a sort of countdown mode (it is probably what is called inching mode) and while doing so I was able to get the device into similar weird state. I don't know whether this will help or not, but perhaps try experimenting with different values of manuSpecificTuya_3 cluster attribute switchType. I have seen the following behaviour:

0 - trigger on release of the manual button, latch
1 - trigger on touch of manual button
2 - follow manual button
3 - same as 2
4 - manual buttons don't work

Combining writes to manuSpecificTuya_3 switchType and pushing of "Mode" button on device resulted in a weird state where device would continuously send on/off events to Z2M.

If you happen to know how to get this device into inching mode and configure corresponding delay please let me know, I will appreciate any pointers in this directions.

rickoltan, 2023-10-29:
I found out it has as a fact to do with some configuration settings. Pressing specific two buttons on the device (button 4 and M), the device is in another mode now and reacts as expected.

closed 2023-10-29 by rickoltan