Koenkk/zigbee2mqtt #19433: no occupancy since for TuYa 809WZT

mpuff, 2023-10-26:
### Is your feature request related to a problem? Please describe

this feature is nice and should be able to enable to all motion sensors

### Describe the solution you'd like

please add the no occupancy since feature to this device, because it is a motion sensor

### Describe alternatives you've considered

this feature is nice and should be able to enable to all motion sensors

### Additional context

thats all

Chartreusito, 2023-10-27:
Hi, 809WZT is already supported with 1.33.1 of z2m : https://www.zigbee2mqtt.io/devices/809WZT.html#tuya-809wzt
I've got 2 of these and they're functionnal.

juandelarocha, 2024-01-10:
I have just bought a 809WZT and won't join my zigbee network. Zigbee2MQTT version is 1.35.1-1. Any advice?

Rubeast, 2024-03-02:
> I have just bought a 809WZT and won't join my zigbee network. Zigbee2MQTT version is 1.35.1-1. Any advice?

Might be related to this:
https://github.com/Koenkk/zigbee2mqtt/issues/20814

closed 2024-09-29 by github-actions[bot]