Koenkk/zigbee2mqtt #19440: FL 122 C innr smart light strip 2m color

Ricardo544, 2023-10-27:
### Link

https://www.innr.com/fr/produit/flex-strip-colour-2m-or-4m/

### Database entry

{"id":25,"type":"Router","ieeeAddr":"0x00158d00028e395f","nwkAddr":23682,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"FL 122 C","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"FL 122 C","manufacturerName":"innr","powerSource":1,"zclVersion":2,"appVersion":48,"stackVersion":33,"hwVersion":2,"dateCode":"20221010","swBuildId":"3.0.1"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":48,"stackVersion":33,"hwVersion":2,"dateCode":"20221010","swBuildId":"3.0.1","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1698425910230,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
It seems to be identical to the OFL 122 C for outdoors, I hope it supports power_on_behavior.

### Supported color modes

1000lm -1800K 6500K

Koenkk, 2023-10-27:
Added! Will be supported in the 1 November release.

closed 2023-10-27 by Koenkk

Ricardo544, 2023-10-27:
Thank you for your answer :o))