Koenkk/zigbee2mqtt #19448: ZY M100 Presence Sensors Not Updating - ONLY LUX SENSOR WORKING

Adaaks, 2023-10-27:
### What happened?

I tried to change the presence sensitivity to 1 to see the effect it has, and then the presence sensors stopped updated, only the lux sensor works even thought the presence sensitivity is back to max... I tried removing the device, re-installing zigbeem2qtt

The sensor used to kinda work (although I feel like small movements were not detected...)

Model = 24G MmWave radar human presence motion sensor
Model = _TZE204_ijxvkhd0

### What did you expect to happen?

For presecence sensors to work

### How to reproduce it (minimal and precise)

Change  presence sensitivity

### Zigbee2MQTT version

1.33.1-dev commit: 0e55b38

### Adapter firmware version

20210708

### Adapter

Sonoff 3.0 zigbee usb dongle

### Debug log

[image]

Fehuv, 2023-10-31:
I have the same thing

igoralexeev, 2023-11-03:
same thing

Adaaks, 2023-11-03:
what fixed it for me is connecting the presence sensor to smart life and pairing it to home assistant again )requires tuya hub...however i stopped using this presence sensor as it detects presence when there is nobody in room sometimes (not even in close to sensor) and literally no movement. this sensor is too much problem, and smart homes are a pain in the ass.

Fehuv, 2023-11-04:
> то же самое

Hi. I don't remember which forum I got it from, but the seller gave this code to some forum member on Aliexpress. In general, you need to copy the file to this path /config/zigbee2mqtt/ It should be in the same folder with configuration.yaml 

[ext_converter.js.ts.zip](https://github.com/Koenkk/zigbee2mqtt/files/13256466/ext_converter.js.ts.zip)

I copied the permissions ext_converter.js

also in the file configuration.yaml need to add

[image]

GRamsaroep, 2023-12-11:
> > то же самое
> 
> Hi. I don't remember which forum I got it from, but the seller gave this code to some forum member on Aliexpress. In general, you need to copy the file to this path /config/zigbee2mqtt/ It should be in the same folder with configuration.yaml
> 
> [ext_converter.js.ts.zip](https://github.com/Koenkk/zigbee2mqtt/files/13256466/ext_converter.js.ts.zip)
> 
> I copied the permissions ext_converter.js
> 
> also in the file configuration.yaml need to add
> 
> <img alt="Снимок экрана 2023-11-04 в 14 20 04" width="293" src="https://private-user-images.githubusercontent.com/147280555/280462788-00d7b8f9-adff-4c78-b1d4-5a30d18fc9ef.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIyODA5OTUsIm5iZiI6MTcwMjI4MDY5NSwicGF0aCI6Ii8xNDcyODA1NTUvMjgwNDYyNzg4LTAwZDdiOGY5LWFkZmYtNGM3OC1iMWQ0LTVhMzBkMThmYzllZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxMVQwNzQ0NTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01YjdkMzUxYmRkZDljOTg3NjZhZGMxMzdlMzI0NzNiY2FmODhlMDhjY2QzMjJiMWM2NGI2YmE3ZTJiYTU2YTdhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5FULZ3rJdpmQ_R-q3wddM0XXM-CA6p69x4A67zaqewM">

Did this work? I have the same problem and want to test this out today.

tjhubz, 2023-12-27:
Thank you to @Fehuv for posting a fix to this problem! It works for me now. If someone else happens to stumble upon this issue, make sure that you: 

1. Download the zip file that they posted, unzip it
2. Copy the code from the `ext_converter.ts.js` file into a .js file, not a .ts file (I named mine `tuya-converter.js` for organization).
3. Upload the new .js file into the ~/zigbee2mqtt/ directory, the same directory that contains the zigbee2mqtt `configuration.yaml`.
4. Add the following lines to the zigbee2mqtt `configuration.yaml`:
```yaml
external_converters:
  - tuya-converter.js
```
5. Restart zigbee2mqtt, the device should work now! If any values are null, adjust all exposed settings back to normal.

It would be great if this code was built into zigbee2mqtt by default, hopefully it will be added later!

eldonky, 2024-01-27:
It works for me.

Biero1, 2024-02-07:
> With z2m 1.35.2, everything should work out-of-the-box, can somebody confirm this?

Both sensors work good but it's impossible to change sensitivity. They both return back to default after a few seconds.
HAOS, SONOFF Zigbee Dongle Plus-E 7.3.1.0 build 176,  Z2M 1.35.3, _TZE204_ijxvkhd0

Skyhawker231, 2024-02-07:
thank you @Fehuv ! this fixed it

Carrotpro, 2024-02-18:
i still have this issue any one else?

JackTalisker, 2024-02-19:
I have the same issue but I only noticed this post today.
Didn't try the above ext.converter yet. Maybe this evening.

Carrotpro, 2024-02-19:
> I have the same issue but I only noticed this post today. Didn't try the above ext.converter yet. Maybe this evening.

Hey I found there using zigbee2mqtt edge it's on the add on page

JackTalisker, 2024-02-20:
Yesterday evening I tried to re-pair the device using the above custom converter. 
Now the radar of my ZY-M100-24G is operational again.

rVlad93, 2024-03-01:
Just got the sensor today but I'm facing the same problem even with the external converter. What's weird is that I managed to change the radar sensitivity to 10 and now I can't change it back to default or any other value.

Carrotpro, 2024-03-01:
> Just got the sensor today but I'm facing the same problem even with the external converter. What's weird is that I managed to change the radar sensitivity to 10 and now I can't change it back to default or any other value.

Do you have add on zigbee2mqtt edge?

rVlad93, 2024-03-01:
> > Just got the sensor today but I'm facing the same problem even with the external converter. What's weird is that I managed to change the radar sensitivity to 10 and now I can't change it back to default or any other value.
> 
> Do you have add on zigbee2mqtt edge?

No, I'm on the regular version. Do I need that in order for this to work?

Carrotpro, 2024-03-01:
> No, I'm on the regular version. Do I need that in order for this to work?

In my case yes don't no about everyone else

rVlad93, 2024-03-02:
I can confirm it's working on the latest verion without the need for the external converter. Thank you!

Carrotpro, 2024-03-02:
> I can confirm it's working on the latest verion without the need for the external converter. Thank you!

Your welcome 😁

alexianos11, 2024-05-29:
as of 1.37.1, it does not work, the same symptoms. Reverting to 1.36.0 works with forementioned external converter.
Could anyone tell me which version works without external converter? "latest" does not help.

Th3M1k3y, 2024-06-07:
I am on [1.38.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.38.0) and getting this error

`Exception while calling fromZigbee converter: Value '5' is not allowed, expected one of 0,1,2}`

jesmnm23, 2024-06-11:
en la actualización lo reconoce pero no funciona la presencia...Alguien conoce solucion? Gracias

eldonky, 2024-06-11:
> Thank you to @Fehuv for posting a fix to this problem! It works for me now. If someone else happens to stumble upon this issue, make sure that you:
>
> 1. Download the zip file that they posted, unzip [...]

A mi me funcionó esto.

Mrsash2020, 2024-06-23:
On 1.38.0-1

Ok so I have these sensors previously. Had the issue where I could not change sensitivity. Seller on Aliexpress provided the above .js file which I am not sure did much apart from able to being able to commit slider. 

Yesterday tried to add 2 more of these presence sensors. OMG what a pain 
Exception while calling fromZigbee converter: Value '5' is not allowed, expected one of 0,1,2}

Thought it was me. Followed a lot of suggestions, did not help.

This morning went to Edge and had to reboot VirtualBox instance and it finally started showing presence and movement data. Something in the normal version still is a problem

anoblet, 2024-06-26:
I'm receiving this error on Z2M Edge without the external converter: `Invalid value for number.bedroom_presence_presence_sensitivity: 0.6 (range 1.0 - 10.0)`

Mrsash2020, 2024-06-26:
For me it works in the Edge but the distance needs to be multiplied by 10 to correct it with our without the external converter js file.  The extrenal converter however was causing the time stamp of last motion to showing as 10 hours off. So dropped the external converter.

Th3M1k3y, 2024-06-26:
Trying to use the converter on [1.38.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.38.0) is giving me this error

```
2024-06-26 17:32:36] error: 	z2m: Failed to load external converter file 'tuya-converter.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js)
[2024-06-26 17:32:36] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
[2024-06-26 17:32:36] error: 	z2m: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
```

After some searching I found the mention of having to change a bit in the top of the converter, so it looks like this

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const legacy = require('zigbee-herdsman-converters/lib/legacy');

const tuya = require('zigbee-herdsman-converters/lib/tuya');
```

After that change the error has gone away, including the note about the usage of external converters, and the sensor still isn't working.

This error is still showing in the log
`Exception while calling fromZigbee converter: Value '5' is not allowed, expected one of 0,1,2}`

Mrsash2020, 2024-06-26:
I was provided this one a few days ago for the ZY-M100-24GV2

[ms.zy24g.js.txt](https://github.com/user-attachments/files/15996962/ms.zy24g.js.txt)

Th3M1k3y, 2024-07-03:
After updating mine is now detecting movement and such, and is identified as [ZY-M100-24GV2](https://www.zigbee2mqtt.io/devices/ZY-M100-24GV2.html#tuya-zy-m100-24gv2), but move and presence sensitivity is acting up.

Setting it to anything over 1 will revert it to 1, and setting it to 0 leaves it at 0.

tanus10, 2024-07-03:
try to set the sensitivity value x 10.
Like 30 or 50 or 60. It settles with 3, 5, 6 respectively. Not sure why.

I have ditched the device because of MQTT message flooding problem and moved on to FP1. My system became too unstable, such as slow responses, z2m container crashes, restarts. Especially when log is set to INFO.

Irrelevant but, I am still using wifi version of the same product, it works without such problems.

Mrsash2020, 2024-07-03:
I was provided this yesterday
[2ms.zy24g.js.txt](https://github.com/user-attachments/files/16091318/2ms.zy24g.js.txt)

Seems to correct the distance but some unit naming issues

anoblet, 2024-07-04:
For me, these units behave under par. I wish I didn't go for Zigbee. I bought an LD-2410C and it's so much better.

jonny190, 2024-07-09:
I'm also getting the log spam with
[image]
From the same device, the roof mounted ZY-M100-24G
Could it just be a case of amending the device template?

danielo515, 2024-10-25:
> I can confirm it's working on the latest verion without the need for the external converter. Thank you!

Which version is that?

danielo515, 2024-10-25:
By the way, extensions are expected to export a class, so the provided JS snippet is not valid

IronButterfly, 2024-11-02:
After the last update Z2M to 1.41.0 the whole log was flooded with messages like:

2024-11-02 18:03:33.707 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.human_kitchen_motion_sensitivity: 90 (range 1.0 - 10.0)
2024-11-02 18:03:34.653 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.human_kitchen_occupancy_sensitivity: 90 (range 1.0 - 10.0)

ZY-M100-24G

What can I do with these values?

closed 2025-01-11 by github-actions[bot]