Koenkk/zigbee2mqtt #19449: [New device support]: SC500ZB-v2 LoraTap Shutter Switch

buldoog, 2023-10-27:
### Link

https://pl.aliexpress.com/item/1005004988690810.html

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c1387c33dd5521","nwkAddr":61772,"manufId":4417,"manufName":"_TZ3000_femsaaua","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1,242],"endpo ints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[4,5,6,258,57345,0],"outClusterList":[25,10],"clusters":{"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercenta ge":10,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":0,"moesCalibrationTime":106}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"tuyaBacklightS witch":1,"tuyaBacklightMode":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"conf iguredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1698440315440,"defaultSendRequestWhen":"i mmediate"}

### Comments

The device is detected as SC400 and only partially works. I can open and close the roller blind, but setting any position between open and close does not work. For example, the roller blind is completely closed and I click 50% and it will open completely anyway. I performed the calibration correctly and set the opening time of the entire roller shutter to 25 seconds.

closed 2024-05-25 by github-actions[bot]

albertomercurio, 2025-01-05:
Same problem for me. We should need a firmware update

dmbuil, 2025-01-14:
Same here! Any Ext. Converter possible?

petjek, 2025-02-24:
Same for me. All 5 new devices are reported as Tuya TS130F.