Koenkk/zigbee2mqtt #19457: MQTT error: connect ECONNREFUSED

jaydhary14, 2023-10-28:
### What happened?

Consistently getting the "MQTT error: connect ECONNREFUSED"  error message on my vm running on ProxMox. This has has started recently and have never seen this error before. 

I have mqtt running on vm as well in a proxmomx 

I am using latest version of homeassistant running on proxmox

### What did you expect to happen?

Not sure why I am getting this message, I would expect this message to go away 

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20210708

### Adapter

zStack3x0

redpixy, 2023-10-29:
disabled ipv6 on your server if your network doesn't support it.

redpixy, 2023-10-29:
did you try using programs like mqttx to check whether you can access your mqtt server?

jaydhary14, 2023-10-29:
Disabled IPV6 on zigbee and mqtt , restarded the server , however still getting the error

***@***.***

redpixy, 2023-10-30:
1. is your network_mode: host ?
2. have you tried programs like mqttx to test the connection to your mqtt server?
3. is your mqtt using anonymous connection?

Buttars, 2023-10-31:
I had to set the IP to 127.0.0.1 instead of local host. The containers do not have the proper hosts files set.

closed 2024-05-29 by github-actions[bot]