Koenkk/zigbee2mqtt #19459: [New device support]: Thermostat Avatto TS0601  _TZE204_lzriup1j

poky0611, 2023-10-28:
### Link

https://www.aliexpress.com/item/1005004149361837.html?spm=a2g0o.productlist.main.7.33b574e0fKVUhl&algo_pvid=4d80c551-5cbb-4ef1-a0a9-2e2b1390c7b1&aem_p4p_detail=202310280653406094599874110440001983376&algo_exp_id=4d80c551-5cbb-4ef1-a0a9-2e2b1390c7b1-3&pdp_npi=4%40dis%21CZK%211055.80%21686.34%21%21%2144.18%21%21%40210384db16985012208764468e5c35%2112000033940405467%21sea%21CZ%210%21AB&curPageLogUid=4Gk4YmDj1Uf6&search_p4p_id=202310280653406094599874110440001983376_4

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0xa4c138ff9bedf68f","nwkAddr":13877,"manufId":4417,"manufName":"_TZE204_lzriup1j","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���,i���,i���,i���,i���,i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1698500204419,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,

I bought wall thermostat Avatto in Zigbee2Mqtt it shows as TS0601  _TZE204_lzriup1j and is unsupported. Can you please help me to make it work?

Thank You!
Robert Pokorný

roon7, 2023-10-31:
Here is what I use now temporary,  (saved in zigbe2mqtt folder TZE204_lzriup1.js)
it's for sure not perfect but most important settings are working.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const valueConverter = {
    systemMode: tuya.valueConverterBasic.lookup({'heat': true, 'off': false}),
    runningState: tuya.valueConverterBasic.lookup({'heat': 1, 'idle': 0}),
    preset: tuya.valueConverterBasic.lookup({'manual': 0, 'program': 1}),
    backlight: tuya.valueConverterBasic.lookup({'off': 0, 'low': 1, 'medium': 2, 'high': 3}),
    sound: tuya.valueConverterBasic.lookup({'off': 0, 'on': 1}),
    divideBy10: tuya.valueConverterBasic.divideBy(10),
};

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_lzriup1j',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Fill in a description of the device here',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 60, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET)
            .withSystemMode(['off', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat'], ea.STATE)
            .withPreset(['manual', 'program'], ea.STATE_SET),
            exposes.enum('backlight', ea.STATE_SET, ['off', 'low', 'medium', 'high'])
            .withDescription('Screen brightness'),

        
            e.child_lock(),
            exposes.binary('sound', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Switches beep sound when interacting with thermostat'),
            exposes.binary('frost', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Antifreeze function'),

            // <- added the humdity expose
        // Here you should put all functionality that your device exposes
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'system_mode', valueConverter.systemMode],
            [19, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1], 
            [4, 'preset', tuya.valueConverterBasic.lookup(
                {'manual': tuya.enum(0), 'program': tuya.enum(1)})],
            [101, 'running_state', valueConverter.runningState],
            [2, 'current_heating_setpoint', valueConverter.divideBy10],
            [15, 'upper_temperature', tuya.valueConverterBasic.raw],
            [3, 'local_temperature', valueConverter.divideBy10],
            [15, 'heating_temperature_limit', valueConverter.raw], 
            [9, 'child_lock', tuya.valueConverter.lockUnlock],
            [102, 'flost', tuya.valueConverter.onOff],
            [105, 'sound', tuya.valueConverter.onOff],
            [110, 'brightness_state', tuya.valueConverter.backlightMode],
        ],
    },
};

module.exports = definition;

poky0611, 2023-11-02:
Everything works fine now with your JS, thank you so much, you are the best!

closed 2023-11-02 by poky0611

Koenkk, 2023-11-03:
@roon7 would you mind to make a pull request such that this code is integrated?

roon7, 2023-11-03:
Hi @Koenkk 
Unfortunately this is not ready, missing many settings:  sensor, schedule, week, brightness_state, sound, frost_protection, factory_reset, heating_temp_limit, deadzone_temperature.
I couldn't devote too much time to it, so I focused on basic settings such as basic temperature control. The rest didn't really want to work for me.
This TZE204_lzriup1 is selled as Avatto ZWT100 3A, so I think it's very similar or even the same as 
**Beok TGR85-ZB, AVATTO ZWT-100-16A**
This one is supported now, but it's hard for me to check, as I coudn't find code for it.

vmrk00, 2023-12-03:
I tried what @roon7  suggested, but it didn't work for me. What could I have done wrong? Thanks!

roon7, 2023-12-03:
Hi everything looks good,
it's my mistake, I forgot that you have add to the config.yaml in the existing in zigbee2mqtt folder lines:
external_converters:
  - TZE204_lzriup1.js
  then it will works for sure.

vmrk00, 2023-12-03:
It's works!  Thank you, you are awesome!

vmrk00, 2023-12-07:
Hi @roon7 , 
Again me. Is it possible that since I set this up, my other devices aren't connecting to Zigbee2MQTT anymore?
All of other devices disabled and don't communicate with my HA.

roon7, 2023-12-07:
Hi @vmrk00 
it is not possible for this change to have this effect, so may be you did some changes in config file by miskate.
Do you have copy of the config file before you make change, if then try to compate it.
You can also undo the changes you made to add the thermostat and see if it changes anything.

MrEcosse, 2023-12-08:
This is the completed converter with all the options now available such as the key sound, the ability to choose the internal or external (floor) sensor, and the ability to set the maximum temperature of the floor.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const valueConverter = {
    systemMode: tuya.valueConverterBasic.lookup({'heat': true, 'off': false}),	
    runningState: tuya.valueConverterBasic.lookup({'heat': true, 'idle': false}),
    divideBy10: tuya.valueConverterBasic.divideBy(10),
};

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_lzriup1j']),
        model: 'ZWT100-16A',
        vendor: 'TuYa',
        description: 'Avatto wall thermostat for electric heating',
        onEvent: tuya.onEvent({timeStart: '1970'}),
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [{...tuya.tz.datapoints, key: [...tuya.tz.datapoints.key, 'key_sound', 'sensor_choice', 'sensor_limit', 'inverse']}],
        whiteLabel: [{vendor: 'AVATTO', model: 'ZWT-100-16A'}],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Full factory reset, use with caution!'),
            e.binary('key_sound', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Switches sound when interacting with thermostat'),
            e.child_lock(),
            e.climate()
                .withSystemMode(['off', 'heat'], ea.STATE_SET)
                .withPreset(['auto', 'manual', 'temporary_manual'])
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperature(ea.STATE)
                .withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET),
            e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Antifreeze function'),
            e.max_temperature_limit()
                .withUnit('Â°C')
                .withValueMin(15)
                .withValueMax(90)
                .withValueStep(1)
                .withPreset('default', 60, 'Default value')
                .withDescription('Maximum upper temperature'),
             e.numeric('sensor_limit', ea.STATE_SET)
                .withUnit('Â°C')
                .withValueMin(35)
                .withValueMax(60)
                .withValueStep(1)
                .withPreset('default', 60, 'Default value')
                .withDescription('Floor sensor temperature limit'),
            e.numeric('deadzone_temperature', ea.STATE_SET)
                .withUnit('Â°C')
                .withValueMax(10)
                .withValueMin(0.5)
                .withValueStep(0.5)
                .withPreset('default', 1, 'Default value')
                .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
            e.binary('inverse', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Primary and secondary reverse selection.  ON is Auxillary reversal'),
            e.enum('sensor_choice', ea.STATE_SET, ['inside', 'outside', 'all'])
[... 27 lines cut ...]
                [109, null, tuya.valueConverter.ZWT198_schedule],
                [109, 'schedule_weekday', tuya.valueConverter.ZWT198_schedule],
                [109, 'schedule_holiday', tuya.valueConverter.ZWT198_schedule],
                [110, 'backlight_mode', tuya.valueConverter.backlightModeOffLowMediumHigh],
                [111, 'inverse', tuya.valueConverter.onOff], 

		
                // These are the schedule values in bytes, 8 periods in total (4 bytes per period).
                // For each period:
                // 1st byte: hour
                // 2nd byte: minute
                // 3rd, 4th bytes: temperature multiplied by 10
                // On the device last 2 periods are ignored if schedule_mode is 7day. When schedule_mode is disabled,
                // scheduling can't be configured at all on the device.
                // For example, if schedule_mode is weekday/sat+sun and this byte array is received:
                // [6,10,1,144,8,10,0,170,11,40,0,170,12,40,0,170,17,10,0,230,22,10,0,170,8,5,0,200,23,0,0,160]
                // Then the schedule is:
                // Mon-Fri: 6:10 --> 40C, 8:10 --> 17C, 11:40 --> 17C, 12:40 --> 17C, 17:10 --> 23C, 22:10 --> 17C
                // Sat-Sun: 8:05 --> 20C, 23:00 --> 16C
            ],
        },
    };

module.exports = definition;
```

roon7, 2023-12-08:
Looks like everything works now, thank you @MrEcosse 
Great job!

MrEcosse, 2023-12-08:
No problem

tispokes, 2023-12-12:
With ZHA it's possible to sync time/weekday with HA, seems like this - via z2m - does not?

tispokes, 2023-12-12:
Okay thanks. Also there are some writing mistakes (copied https://github.com/Koenkk/zigbee2mqtt/issues/19459#issuecomment-1846785112 as it is the latest post) according to the temperature unit
[image]

tispokes, 2023-12-20:
@MrEcosse another issue I've to address:
Setting the `current_heating_setpoint` with 0.5 steps is possible on the device but not via z2m. When you set a x.5 value, the value step back to 0 degrees.

MrEcosse, 2023-12-20:
Working OK for me on both my stats.  Changed set point from the HA Gui to 18.5 and that's what the Stat now shows

tispokes, 2023-12-20:
Interesting, maybe it's my Sonoff Plus dongle (ERF32)?

MrEcosse, 2023-12-20:
Can you see anything in the MQTT messages when you change the temp?  This is the message I see...

{"backlight_mode":"medium","child_lock":"UNLOCK","current_heating_setpoint":18.5,"deadzone_temperature":0.5,"factory_reset":"OFF","faultalarm":0,"frost_protection":"ON","inverse":"OFF","key_sound":"ON","linkquality":72,"local_temperature":20,"local_temperature_calibration":1,"max_temperature_limit":30,"preset":"manual","running_state":"idle","schedule_holiday":"08:00/20.0°C 23:00/14.0°C","schedule_weekday":"06:00/19.0°C 08:00/16.0°C 11:30/17.0°C 12:30/16.0°C 17:00/20.0°C 22:00/14.0°C","sensor_choice":"outside","sensor_limit":36,"system_mode":"heat","working_day":"5-2"}

MrEcosse, 2023-12-20:
And on my other battery one I get:

{"backlight_mode":"high","child_lock":"UNLOCK","current_heating_setpoint":14.5,"deadzone_temperature":0.5,"factory_reset":"OFF","frost_protection":"ON","linkquality":142,"local_temperature":17.3,"local_temperature_calibration":0,"max_temperature_limit":30,"preset":"manual","running_state":"idle","schedule_holiday":"08:00/20.0°C 22:00/14.0°C","schedule_weekday":"06:00/19.0°C 08:30/16.0°C 11:30/18.5°C 13:30/16.0°C 16:00/20.0°C 22:00/14.0°C","system_mode":"heat","working_day":"5-2"}

tispokes, 2023-12-20:
Hey, I'm pretty sorry, I was on the wrong device in my head. ZWT100 works just fine. Thanks for the effort. BHT6 makes these problems.

tispokes, 2023-12-20:
@MrEcosse have you tested 0 degree deadzone? Is this working? Because the lag between the stopping of heating and stopping of radiating heat is about 1 hour in my case. I would even prefer negative settings, for the upper limit.

MrEcosse, 2023-12-20:
I'm not sure the device itself supports a zero deadzone.  IIRC it wasn't possible to set it so low on the Tuya app.  (I tend to pair with Tuya first just so I can see what capabilities I need to get working in HA).  I guess you are using underfloor heating?

Akrotkiv, 2023-12-23:
Hi, I am new in HA and its workaround. I want to use thermostat:
BEOK TDR-89B-ZB-EP
It is represented as model TS0601 manufacturer _TZE204_lzriup1j

I tried all above, but doesn't work for me.
Should I do something else because of other "brand" or should it work?
I thing it's the same device as BEOK TGR-85 which was supported already (I think)
Product page: https://www.beok-controls.com/room-thermostat/underfloor-heating-thermostat/zigbee-underfloor-heating-thermostat.html

tispokes, 2023-12-23:
Hey @Akrotkiv have you added the JavaScript file for this device (see screenshot attached)

The content you can use from this: https://github.com/Koenkk/zigbee2mqtt/issues/20072#issuecomment-1842641288

It's a bit newer. But you have to replace the Å characters with spaces, see my screenshot some posts above this post.

[image]

You can adjust the Name in line 25
After adding this file you have to add it into the z2m config like this (line 28 and 29) and the reboot HA

[image]

Akrotkiv, 2023-12-23:
Thanks @tispokes 
Yes I add .js and edited config. (see screenshots).

I'm not sure with "underscore" in begining and "j" in the end of name/filename.js
[image]
[image]

roon7, 2023-12-23:
Hi @Akrotkiv,
I'm not sure if this could have impact .. but I have space before TZE204 in config
[image]

tispokes, 2023-12-23:
@Akrotkiv I've recognized the space as well and never tried it without a space between dash and file name. Used yaml at work a lot, but never tested w/o the space.

You can use the simpler file editor with included syntax check.

Akrotkiv, 2023-12-23:
The space doesn't solve anything.
If filename is not important, what defines exactly this device? fingertip?
Can I write "anything" (my device ID) into:
model:
vendor:
whitelabel:

roon7, 2023-12-23:
fingerprint: tuya.fingerprint('TS0601', ['_TZE204_lzriup1j']) this define device.
If you add new device and it's not recognized you can see in logs it fingerprint like TZE204_lzriup1j in this particular situation.
I have no idea why it's not working for you.

Akrotkiv, 2023-12-24:
Thanks @roon7 , @tispokes for your care. ;)
I just realized, that In was using an "edge" addon. Uninstalled, installed regular Zigbee2mqtt addon, and device was recognised as supported.
Wish you all Merry Christmas day,

tispokes, 2023-12-24:
You're welcome and you too

ljakab, 2024-01-13:
I know that this issue is close, but I have the gas version of the exact same thermostat, model number ZWT-BH-3A. @MrEcosse, since I'm a newbie, how would I go about discovering the differences needed? There is no floor temperature for example.