Koenkk/zigbee2mqtt #19462: [New device support]: Tuya thermostat TS0601  _TZE204_pcdmj88b

Tilt7238, 2023-10-28:
### Link

https://de.aliexpress.com/item/1005006066560400.html

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0xa4c138c059067de9","nwkAddr":46572,"manufId":4417,"manufName":"_TZE204_pcdmj88b","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":56,"65508":0,"appVersion":73,"modelId":"TS0601","manufacturerName":"_TZE204_pcdmj88b","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1698512485137,"defaultSendRequestWhen":"immediate"}

### Comments

I bought 2 of these thermostats and tried it with the existing TV02-Zigbee external converter in [Tuya.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts) , but it didn't work. I also tried the [converter ](https://gist.github.com/Koenkk/3993885d51baf82ed334afe0781c7d55)below which I found in another [issue](https://github.com/Koenkk/zigbee2mqtt/issues/19103)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_a4bpgplm'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_dv8abrrz'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_z1tyspqw'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_rtrmfadk'},
    ],
    model: 'TS0601_thermostat_1',
    vendor: 'TuYa',
    description: 'Thermostatic radiator valve',
    whiteLabel: [
        {vendor: 'Unknown/id3.pl', model: 'GTZ06'},
    ],
    onEvent: tuya.onEventSetLocalTime,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(), e.child_lock(), e.max_temperature(), e.min_temperature(),
        e.position(), e.window_detection(),
        e.binary('window', ea.STATE, 'CLOSED', 'OPEN').withDescription('Window status closed or open '),
        e.binary('alarm_switch', ea.STATE, 'ON', 'OFF').withDescription('Thermostat in error state'),
        e.climate()
            .withLocalTemperature(ea.STATE).withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
            .withPreset(['auto', 'manual', 'off', 'on'],
                'MANUAL MODE ☝ - In this mode, the device executes manual temperature setting. ' +
            'When the set temperature is lower than the "minimum temperature", the valve is closed (forced closed). ' +
            'AUTO MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature. ' +
            'ON - In this mode, the thermostat stays open ' +
            'OFF - In this mode, the thermostat stays closed')
            .withSystemMode(['auto', 'heat', 'off'], ea.STATE)
            .withRunningState(['idle', 'heat'], ea.STATE),
        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
        e.binary('boost_heating', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Boost Heating: press and hold "+" for 3 seconds, ' +
            'the device will enter the boost heating mode, and the ▷╵◁ will flash. The countdown will be displayed in the APP'),
        e.numeric('boost_time', ea.STATE_SET).withUnit('min').withDescription('Countdown in minutes')
            .withValueMin(0).withValueMax(1000),
    ],
    meta: {
        tuyaDatapoints: [
            [1, null,
                {
                    from: (v) => {
                        utils.assertNumber(v, 'system_mode');
                        const presetLookup = {0: 'auto', 1: 'manual', 2: 'off', 3: 'on'};
                        const systemModeLookup = {0: 'auto', 1: 'auto', 2: 'off', 3: 'heat'};
                        return {preset: presetLookup[v], system_mode: systemModeLookup[v]};
                    },
                },
            ],
            [1, 'system_mode', tuya.valueConverterBasic.lookup({'auto': tuya.enum(1), 'off': tuya.enum(2), 'heat': tuya.enum(3)})],
            [1, 'preset', tuya.valueConverterBasic.lookup(
                {'auto': tuya.enum(0), 'manual': tuya.enum(1), 'off': tuya.enum(2), 'on': tuya.enum(3)})],
            [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [3, 'local_temperature', tuya.valueConverter.divideBy10],
            [4, 'boost_heating', tuya.valueConverter.onOff],
            [5, 'boost_time', tuya.valueConverter.countdown],
            [6, 'running_state', tuya.valueConverterBasic.lookup({'heat': 1, 'idle': 0})],
            [7, 'window', tuya.valueConverterBasic.lookup({'OPEN': 1, 'CLOSE': 0})],
            [8, 'window_detection', tuya.valueConverter.onOff],
            [12, 'child_lock', tuya.valueConverter.lockUnlock],
            [13, 'battery', tuya.valueConverter.raw],
            [14, 'alarm_switch', tuya.valueConverter.onOff],
            [15, 'min_temperature', tuya.valueConverter.divideBy10],
            [16, 'max_temperature', tuya.valueConverter.divideBy10],
            [17, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [18, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [19, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [20, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [21, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [22, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [23, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [101, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
            [102, 'position', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;
```

ha-jos, 2023-11-01:
Same issue, I ordered the zigbee gateway so I can map the correct data.
The expected delivery will be by the end of next week.

If I find time and the correct mapping data i will try to contribute it so others can use is too.

T0ytoy, 2023-11-07:
I'm very much intersted as I bought multiple of these radiator valves and can't use them right now :(
I'd love to help in any way but I don't have a tuya gateway nor the knowledge to developp an adapter for new devices.

I'll follow this thread with care though!

Erwsaym, 2023-11-07:
I'm interested too !
Thanks in advance if you find the way to bring compatibility !

jim-fx, 2023-11-08:
I have a tuya zigbee gateway and followed the steps outlined in the zigbee2mqtt wiki to [Find Tuya Datapoints](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

Through this I obtained the following data points for _TZE204_pcdmj88b:

```json
{
  "2": "Mode",
  "4": "Set temperature",
  "5": "Current temperature",
  "6": "Battery capacity",
  "7": "Child lock",
  "8": "Temperature scale",
  "9": "Set temperature ceiling",
  "10": "The lower limit of temperature",
  "14": "Window check",
  "16": "Window temp",
  "17": "Window time",
  "18": "Backlight brightness",
  "19": "Factory data reset",
  "21": "Holiday temperature",
  "24": "Home temp",
  "25": "Leave temp",
  "28": "Week program",
  "29": "Week program Tuesday",
  "30": "Week program Wednesday",
  "31": "Week program Thursday",
  "32": "Week program Friday",
  "33": "Week program Saturday",
  "34": "Week program Sunday",
  "35": "Fault alarm",
  "36": "Frost protection",
  "37": "Rapid warming",
  "38": "Rapid heating countdown",
  "39": "Switch Scale",
  "47": "Temperature correction",
  "48": "Valve testing",
  "49": "State of the valve",
  "101": "111"
}
```

I will try to write an external converter myself, but I am not sure if i can do it :)

tnako, 2023-11-09:
Here is mine
```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE204_pcdmj88b'},
        ],
        model: 'TS0601_thermostat_3',
        vendor: 'TuYa',
        description: 'Thermostatic radiator valve',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.child_lock(), e.battery_low(),
            e.climate()
                .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withSystemMode(['auto', 'heat', 'off'], ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperatureCalibration(-3, 3, 1, ea.STATE_SET),
            ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
            e.binary('scale_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('If the heat sink is not fully opened within ' +
                'two weeks or is not used for a long time, the valve will be blocked due to silting up and the heat sink will not be ' +
                'able to be used. To ensure normal use of the heat sink, the controller will automatically open the valve fully every ' +
                'two weeks. It will run for 30 seconds per time with the screen displaying "Ad", then return to its normal working state ' +
                'again.'),
            e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('When the room temperature is lower than ' +
                '5 °C, the valve opens; when the temperature rises to 8 °C, the valve closes.'),
            e.numeric('error', ea.STATE).withDescription('If NTC is damaged, "Er" will be on the TRV display.'),
        ],
        meta: {
            tuyaDatapoints: [
                [2, 'system_mode', tuya.valueConverterBasic.lookup({'auto': tuya.enum(0), 'heat': tuya.enum(1), 'off': tuya.enum(2)})],
                [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [5, 'local_temperature', tuya.valueConverter.divideBy10],
                [6, 'battery', tuya.valueConverter.raw],
                [7, 'child_lock', tuya.valueConverter.lockUnlock],
                [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
                [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
                [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
                [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
                [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
                [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
                [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
                [35, 'fault_alarm', tuya.valueConverter.errorOrBatteryLow],
                [36, 'frost_protection', tuya.valueConverter.onOff],
                [39, 'scale_protection', tuya.valueConverter.onOff],
                [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
                [49, 'valve_state', tuya.valueConverterBasic.lookup({'closed': tuya.enum(0), 'open': tuya.enum(1)})],
            ],
        },
    };

module.exports = definition;
```
Works/tested:
- changing target temp `current_heating_setpoint`
- collecting current temp `local_temperature`
- valve state `valve_state` open/closed

Caveats
- reused model/image 'TS0601_thermostat_3'
- schedule not tested
- `system_mode` looks like preset

T0ytoy, 2023-11-09:
@tnako thank you very much for implementing the converter, I tested it and can confirm that changing target temperature works and triggers the valve. Local temperature is fine too and it can be adjusted +/-3°C. With that and some Home Assistant automation, I think we can pretty much do anything.

It would be perfect if we could directly act on the valve state (percentage opened ?) to rely on any external thermometer, but I can work around that.

Thanks again!

jim-fx, 2023-11-09:
@tnako thank you very much :)

I might be wrong, but I think DP 48 refers to the thermostat opening and closing the valve to the max/min for testing.

Koenkk, 2023-11-09:
@tnako could you create a PR?

tnako, 2023-11-09:
> @tnako could you create a PR?

Sure. First time is best time and closer to weekends.

If I disappear after this promise:
<details><summary>Details</summary>

- added presets
- added boost mode
- changed `valve_state` to `running_state` and heat/idle

```typescript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE204_pcdmj88b'}, // Mine
        ],
        model: 'TS0601_thermostat_3',
        vendor: 'TuYa',
        description: 'Thermostatic radiator valve',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.child_lock(),
            e.battery_low(),
            e.climate()
                .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withPreset(['schedule', 'holiday', 'manual', 'comfort', 'eco'])
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperatureCalibration(-3, 3, 1, ea.STATE_SET),
            ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
            e.binary('scale_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('If the heat sink is not fully opened within ' +
                'two weeks or is not used for a long time, the valve will be blocked due to silting up and the heat sink will not be ' +
                'able to be used. To ensure normal use of the heat sink, the controller will automatically open the valve fully every ' +
                'two weeks. It will run for 30 seconds per time with the screen displaying "Ad", then return to its normal working state ' +
                'again.'),
            e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('When the room temperature is lower than ' +
                '5 °C, the valve opens; when the temperature rises to 8 °C, the valve closes.'),
            e.numeric('error', ea.STATE).withDescription('If NTC is damaged, "Er" will be on the TRV display.'),
            e.binary('boost_heating', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Boost Heating: the device will enter the boost heating mode.'),
            ],
        meta: {
            tuyaDatapoints: [
                [2, 'preset', tuya.valueConverterBasic.lookup({'schedule': tuya.enum(0), 'holiday': tuya.enum(1), 'manual': tuya.enum(2), 'comfort': tuya.enum(3), 'eco': tuya.enum(4)})],
                [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [5, 'local_temperature', tuya.valueConverter.divideBy10],
                [6, 'battery', tuya.valueConverter.raw],
                [7, 'child_lock', tuya.valueConverter.lockUnlock],
                [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
                [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
                [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
                [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
                [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
                [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
                [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
                [35, 'fault_alarm', tuya.valueConverter.errorOrBatteryLow],
                [36, 'frost_protection', tuya.valueConverter.onOff],
                [37, 'boost_heating', tuya.valueConverter.onOff],
                [39, 'scale_protection', tuya.valueConverter.onOff],
                [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
                [49, 'running_state', tuya.valueConverterBasic.lookup({'idle': tuya.enum(0), 'heat': tuya.enum(1)})],
            ],
        },
    };

module.exports = definition;
``` 

</details>

T0ytoy, 2023-11-10:
@tnako I updated with your last version, thank you very much. 

It seems that the presets all have a fixed target temperature value (except manual mode) : eco is 20°C (how tf is that "eco"), comfort is 25°C (🔥 ), holiday ('away') is 15°C. 

@jim-fx you have a tuya gateway, right? Do you know if those values are updatable at all with the official tuya app, or are they hard-coded and the only solution in homeassistant is to use manual mode?

Thanks again for all the work put into this :)

jim-fx, 2023-11-10:
Not sure, I would guess the values are hardcoded. But if you connect the thermostat to home assistant you dont't need presets anymore? I accidentaly bought two, I dont mind sending you one 4 free if you would like to play around yourself :)

T0ytoy, 2023-11-10:
@jim-fx that's very nice of you but not necessary, I already own 5 of them, only one is installed yet, though 😄 
You're right that manual mode is enough, but preset would have been a nice "cherry on top" feature, for example in Home Assistant when no one is at home you trigger the "away" preset instead of setting the manual mode to 17°C, it's a bit easier to understand in the UI.

Anyway 'tis but a detail, thank you for answering ;)

T0ytoy, 2023-11-10:
Thank you for looking it up! Those TRV went from unusable to good home assistant product in a few day, that is so great.

Looking up at your converter, I was wondering if the third parameter of :  
`.withLocalTemperatureCalibration(-3, 3, 1, ea.STATE_SET)` was the increment size. If that is the case, it would be better at 0.5 or even 0.1, as 1C is a bit high for calibration. But maybe there is a hardware limitation and it cannot be done.

Thanks again!

tnako, 2023-11-10:
Yes, '1' is precision and you can put any step there. For this valve only 1C calibration works in my tests.

Personally I use home assistance and calibration way easy to do with `template.sensors`. Example of usage hoas `filter` on raw data: Red is raw valve values and Blue is 5min moving average.
[image]

tnako, 2023-11-11:
And merged to main branch. Thank everybody! With datapoints here was easy to make convertor 👍

j4m3s, 2023-11-16:
Thank you very much for doing this @tnako, it's really good of you.    I've got one of these devices and have been playing with the device & the converter:
  * it appears that the temperature calibration can actually range between -12 and 12 from my testing. 
  * I think it's [this device](https://www.aliexpress.com/item/1005006220325821.html?spm=a2g0o.productlist.main.75.5f18cgyFcgyFOo&algo_pvid=6e696f4b-1641-42c4-849d-70f6c93d51f7&algo_exp_id=6e696f4b-1641-42c4-849d-70f6c93d51f7-37&pdp_npi=4%40dis%21GBP%21118.40%2159.2%21%21%21131.50%21%21%402103864c17001564181656437e67fd%2112000036338445747%21sea%21UK%212756995310%21&curPageLogUid=41Yw41Ao4V8N) rather than the one in the picture.  I own both, the one pictured in zigbee2mqtt is a _TZE200_hue3yfsn I think.
  * The number of transition-points in the day schedule is hardcoded at 4 in `tuya.ts` [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/82b3b1524fa0f11fc1507479db7ec1f99cf8c386/src/lib/tuya.ts#L695) - I don't know how to know if that's a genuine limitation in the device or not.  According to the AliExpress link above it supports 6 transitions each for 5+2, 6+1 or 7 individual days, the same as the TZE200_hue3yfsn.  If that's the case, there should be another datapoint for "schedule" like [this one](https://github.com/Koenkk/zigbee-herdsman-converters/blob/8c13303704e4de9dbf5c6df149431330c5fd653e/src/devices/tuya.ts#L2730).

dockerpirate, 2023-11-22:
i tested dev branch, some functions need to be corrected:

- 'current_heating_setpoint', 5, 45, 0.5
- .withLocalTemperatureCalibration(-7, 7, 1, ea.STATE_SET)
- frost_protection -> doesnt work
- scale_protection -> is not present as setting on the device
- system_mode state -> wrong value for "heat / off"

not tested
- battery low (battery status seems to work)
- error state
- shedule

Maybe for description modell nr. is "BAB-1413Pro-E" and it looks similar to Tuya TV02, but it isnt
[image]

[image]

shyney7, 2023-11-27:
> How do you even get those devices in pairing mode?
> Only possible via Tuya gateway?

In the manual it says to press the reset button for >5seconds

apedance, 2023-11-27:
Not sure what manual you have but mine doesn't have this option.

[image]

elmargrote, 2023-11-27:
There is a very small black button above the display which you can only see when you have device open.
You have to push that until the wlan symbol blinks.

apedance, 2023-11-27:
I have a small black button.
The only thing it does when pressed once is change the orientation of the display.
When pressed longer nothing happens other than it turning off the display after 10s of pressing.

I can see the Wifi symbol. I have no idea how to pair this device.
Comms should be Zigbee.

[image]
[image]

philanmar, 2023-11-27:
I have the same devices, to pair them, you have to "activate" the TRV by pressing a button, so the display is active (took me a little bit to figure this out cause i wondered why it did not start pairing), then you have to press that little, black button for a few seconds, then a blue LED should start to blink, which means the device is in pairing mode.

apedance, 2023-11-27:
Yes, well, I think.
I finally have pairing mode. Thank you guys.
But zigbee2mqtt can't see those devices yet.

apedance, 2023-12-01:
I updated to the latest version which was released. (stable)
Support is in the changelog.

Still if I enter pariring mode (blue light blinking) - I cannot see the devices in zigbee2mqtt.
Any ideas?

shyney7, 2023-12-01:
> I updated to the latest version which was released. (stable)
> Support is in the changelog.
> 
> Still if I enter pariring mode (blue light blinking) - I cannot see the devices in zigbee2mqtt.
> Any ideas?

Its working for me and was added after allowing new devices

shyney7, 2023-12-01:
> This is weird. Anything I am missing in z2m?
> [image]
> 

Should work after enabling join. Maybe reset the device and then try again?

A problem that I am having is that the device just turns to off mode after a while although it's set to manual mode and was set to heat mode.

[image]

shyney7, 2023-12-01:
@apedance 
[image]

T0ytoy, 2023-12-01:
Same model number on my manual, everything works great on z2m with a Zigate+. The fact that both TRV can't connect might mean the problem iis with z2m or your dongle?

apedance, 2023-12-01:
> Same model number on my manual, everything works great on z2m with a Zigate+. The fact that both TRV can't connect might mean the problem iis with z2m or your dongle?

hmmm. I am using the recommended SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P with crrent version: 1.34.0-1 zigbee2mqtt.
Hue Sensors and other TRVs are working fine. Can this be related to the dongle specifically?

@shyney7 please do me a favour and delete the quoted image. ;) thanks

apedance, 2023-12-01:
> > This is weird. Anything I am missing in z2m?
> > ![image](https://private-user-images.githubusercontent.com/4264240/287277126-d0107230-4a9f-4db3-b7ff-c787f177c726.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE0Mzg5NDUsIm5iZiI6MTcwMTQzODY0NSwicGF0aCI6Ii80MjY0MjQwLzI4NzI3NzEyNi1kMDEwNzIzMC00YTlmLTRkYjMtYjdmZi1jNzg3ZjE3N2M3MjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDFUMTM1MDQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjhjMDEwZjk2ZmZhNmQwNGYzYTlmMmE3NjBiMmZhYWMxYzNmZDAwZjgxNDAzODA2N2MzMWI5NmFlYjZlOWI4YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.6d_swqezdhm_ZWSRyRSyKxojdVcvOr07K8FCR0sxnLo)
> 
> Should work after enabling join. Maybe reset the device and then try again?
> 
> A problem that I am having is that the device just turns to off mode after a while although it's set to manual mode and was set to heat mode.
> 
> ![Screenshot_20231201_142919_Home Assistant.jpg](https://private-user-images.githubusercontent.com/17785769/287278072-5101f8f9-2ba1-48fe-9959-9bde180b4a11.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE0Mzg5NDUsIm5iZiI6MTcwMTQzODY0NSwicGF0aCI6Ii8xNzc4NTc2OS8yODcyNzgwNzItNTEwMWY4ZjktMmJhMS00OGZlLTk5NTktOWJkZTE4MGI0YTExLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjAxVDEzNTA0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM2ZjJjOWVhODVkYThjZmYyNWNhOGMxMjhiYjRmM2NkMzE2NGM0Y2YyMjEwNDc5ZmM1ZDY4ZWFiNDU3ZTAxZWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uAvaxMdzwBBeTz0nwJlrP8YcaOKiCe06p6Wdk31kzRE)

this one

shyney7, 2023-12-01:
It seems that it switches to off after about 2-3 minutes.
[image]

j4m3s, 2023-12-01:
Looking at your graph that's because the actual current temperature (the blue line) is already above the target temperature (the pink line), so the TRV is doing what it's supposed to, turning off the heat. Or am I misunderstanding your question?

apedance, 2023-12-01:
Testing two right now.
They work fine so far. They have far worse signal rates LQI at the same conditions than a SONOFF TRVZB device for instance.

shyney7, 2023-12-01:
> Looking at your graph that's because the actual current temperature (the blue line) is already above the target temperature (the pink line), so the TRV is doing what it's supposed to, turning off the heat. Or am I misunderstanding your question? 

It's not the same behavior like my other tuya device. Its stays on even after reaching the target temperature. That makes more sense to me to keep the desired level. You don't want to close the valve completely after reaching the temperature and only opening it after the temperature is falling below level again.

Here is how it is on the other device (_TZE200_hue3yfsn):

[image]

j4m3s, 2023-12-01:
The _TZE200_hue3yfsn doesn't expose its valve state though - the herdsman code for that device always reports the valve as open (unless the device is set to Heating Stop iirc).  So the orange shading on the first graph will look that way even if you set the target temperature to 15 degrees and the valve is actually fully closed.

dockerpirate, 2023-12-02:
> > Looking at your graph that's because the actual current temperature (the blue line) is already above the target temperature (the pink line), so the TRV is doing what it's supposed to, turning off the heat. Or am I misunderstanding your question?
> 
> It's not the same behavior like my other tuya device. Its stays on even after reaching the target temperature. That makes more sense to me to keep the desired level. You don't want to close the valve completely after reaching the temperature and only opening it after the temperature is falling below level again.
> 
> Here is how it is on the other device (_TZE200_hue3yfsn):
> 
> ![
> 
> This is also the reason why the temperature is overshooting as you can see in the graph below. At 16:00 I installed the new thermostat. It should look like before a steady temperature. Since the thermostat is closing completely and then opening completely the temperature is overshooting the target temperature all the time. ![

But your problem has nothing to do with z2m.  z2m sends the target value to the device and the device adjusts it.  It's like a switch, if the switch flickers it has nothing to do with z2m

shyney7, 2023-12-02:
@j4m3s thanks for the clarification 👍. I didn't know that the _TZE200_hue3yfsn model doesn't transfer the valve state and thought that heating is the normal state. So everything is working fine from the z2m side . Although I think that these thermostats are useless with such a high delta. Not even the standard thermostats that just work mechanically without electronics work better than this. Is it possible to force on off from HA so that I can use a PID controller e.g. to control the thermostat and use an external temperature sensor?

j4m3s, 2023-12-02:
This is one of the reasons why I've decided I don't like these devices and stopped using them - I've bought another _TZE200_hue3yfsn instead (although I really wish it exposed the valve state!). 

To answer your question: you could create an automation which sets the target temperature to a high value when your external sensor shows the temperature is too low, and vice versa when the sensor shows it's too low. But I think you'd still see big swings due to the thermal inertia in the radiator. To avoid the swings you would want to control the % open of the valve to have it open a small amount when it's close to target. That's not possible with this device.  The other alternative is to use an automation to adjust the temperature calibration value up or down according to the delta between the temperature the device detects and the external sensor. Unfortunately that doesn't work well on this device either - not only is the local device temperature resolution poor but the calibration can only be adjusted in 1 degree steps too.

So tnako has done a great job of adding support for the device to z2m, but IMO they're not very good devices so I dont use them.

dockerpirate, 2023-12-02:
> @j4m3s thanks for the clarification 👍. I didn't know that the _TZE200_hue3yfsn model doesn't transfer the valve state and thought that heating is the normal state. So everything is working fine from [...]

Maybe the problem is a high value for "blind spot"?

[image]

shyney7, 2023-12-02:
@dockerpirate since it's already set to 1° by default I don't see any optimization potential here!?

j4m3s, 2023-12-08:
What we really need is better firmware for the devices so that they use 0.1 degree increments for temp and calibration, and expose the valve open state. Then they'd be OK :)

closed 2024-07-06 by github-actions[bot]

stillax, 2024-12-29:
Hi, I could buy some TRVs (BAB-1413Pro-E), but first I want to know, what is the actual state of the TRVs and HA? Thanks a lot.

qveensi, 2025-02-19:
> Thank you very much for doing this [@tnako](https://github.com/tnako), it's really good of you. I've got one of these devices and have been playing with the device & the converter:
>
> * it appears [...]

Hello, I also purchased this model and checked that in reality the calibration works from -12 to 12 in increments of 0.5

I created the corresponding request pool: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8840

sebastiendussaux, 2026-01-18:
Hello everyone,
I have these faucets but I'm having a problem with the system_mode in Z2M. It shows "heat" when the valve is closed and "off" when the valve is open. In Home Assistant, it turns off the thermostat as soon as it's supposed to heat up. How can I reverse this behavior?

Thanks in advance.
Sorry, I don't speak English, so I translated it using Google Translate.