Koenkk/zigbee2mqtt #19463: Lights turn off or on very slowly or some times not working

avdgaag1984, 2023-10-28:
### What happened?

Operating the lights suddenly becomes very slow
When i read the log some times when i hid a light button i get the error: Public 'set' 'state' to 'Glazen Kast' failed: Error: Command 0x000d6ffffef34f72/1 genOnOff.on({},{"sendWhen":"immediate","timeout"failed (Data request failed with error: 'MAC channel access failure'(
))'

### What did you expect to happen?

The lamp must turn on or off

### How to reproduce it (minimal and precise)

Restart Home Assistant

Remove the zigbee stick from the USB and put it back in

Restart my synology Nas

### Zigbee2MQTT version

Current version: 1.33.1-1

### Adapter firmware version

20220219

### Adapter

CC2652RB

### Debug log

Info 2023-10-28 20:27:43MQTT publish: topic 'zigbee2mqtt/Glazen kast', payload '{"linkquality":58,"power_on_behavior":null,"state":"OFF","update":{"installed_version":537019939,"latest_version":537019939,"state":"idle"},"update_available":false}'
Error 2023-10-28 20:27:52Publish 'set' 'state' to 'Glazen kast' failed: 'Error: Command 0x000d6ffffef34f72/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
Info 2023-10-28 20:27:52MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Glazen kast' failed: 'Error: Command 0x000d6ffffef34f72/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"Glazen kast"},"type":"zigbee_publish_error"}'

lux73, 2023-11-01:
try to Update your Coordinator Stick with latest Firmware -> https://www.zigbee2mqtt.io/guide/adapters/#flashing-cc1352-cc2652-cc2538-based-adapters

jb137, 2023-11-30:
It seems I am experiencing the same problem since this morning. I am running a Sonoff CC2652P based device. After updating to the latest coordinator version (20230507) the issue continues.

The logs show "Data request failed with error: 'MAC channel access failure' (225)".

The coordinator seems to work fine receiving events. When I toggle switches directly (outlet linked to Zigbee switch) the state of the targeted outlet is updated immediately in Zigbee2MQTT web interface. However when I try to control a switch from the web interface the error comes up after a few seconds.

How to proceed?

jb137, 2023-11-30:
The issue has been solved for me. I run the software on a rpi with external ssd. The antenna was located 30cm from the rpi and ssd. This worked for over a year. Pulling these further apart did the trick. And everything is working again like before.

closed 2024-06-28 by github-actions[bot]