Koenkk/zigbee2mqtt #19464: [New device support]: eWeLink SNZB-04 is supported, but not detected as such

Emyrk, 2023-10-28:
### Link

https://www.amazon.com/gp/product/B0C2HX5W7R/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0xa4c138e0d32800ab","nwkAddr":27551,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-04","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SNZB-04","manufacturerName":"eWeLink","powerSource":3,"zclVersion":3,"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffec6e5f1","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"genOnOff":{"attributes":{}},"genIdentify":{"attributes":{"identifyTime":0}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec6e5f1","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec6e5f1","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1698530300937,"defaultSendRequestWhen":"active","checkinInterval":3600}

### Comments

This is the Sonoff SNZB-04 door contact sensor defined in the vendor list here:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/sonoff.ts#L112-L133

I think it just has to be added to the white label `eWeLink`, but I had some issues getting that to work. I have tested this with the Sonoff SNZB-04 and everything is working as intended. I am unsure if I did this correctly, so did not open a new PR.

It just needs to be added to the existing door sensor under the new manufacturer name.

### External converter

```shell
const definition =     {
  fingerprint: [
      // ModelID is from the temperature/humidity sensor (SNZB-02) but this is SNZB-04, wrong modelID in firmware?
      // https://github.com/Koenkk/zigbee-herdsman-converters/issues/1449
      {type: 'EndDevice', manufacturerName: 'eWeLink', modelID: 'TH01', endpoints: [
          {ID: 1, profileID: 260, deviceID: 1026, inputClusters: [0, 3, 1280, 1], outputClusters: [3]},
      ]},
  ],
  // **** MY CHANGE ****
  // I had to add this to the 'zigbeeModel' because it was not working with just adding it to the white label list.
  zigbeeModel: ['DS01',"SNZB-04"],
  model: 'SNZB-04',
  vendor: 'SONOFF',
  // **** MY CHANGE ****
  // Added to white label list
  whiteLabel: [{vendor: 'eWeLink', model: 'RHK06'},{vendor: 'eWeLink', model: 'SNZB-04'}],
  description: 'Contact sensor',
  exposes: [e.contact(), e.battery_low(), e.battery(), e.battery_voltage()],
  fromZigbee: [fz.ias_contact_alarm_1, fz.battery],
  toZigbee: [],
  configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
      await reporting.batteryVoltage(endpoint, {min: 3600, max: 7200});
      await reporting.batteryPercentageRemaining(endpoint, {min: 3600, max: 7200});
  },
};
```

Koenkk, 2023-10-29:
Integrated the fix, thanks! Will be included in the 1 November release.

closed 2023-10-29 by Koenkk

Emyrk, 2023-10-30:
@Koenkk thanks! Is there documentation on how to use all the fields like `whiteLabel` and `fingerprint`? 

I would love to contribute and add these accessories as I get them, just not 100% sure how to use the fields. Was going to deep dive the JS when I get time.

Koenkk, 2023-10-30:
Not really but:
- fingerprint -> is used for detecting the device (like zigbeeModel)
- whiteLabel -> is used to re-use the definition to represent another device. E.g. there are many TuYa devices which work with the same definition, using the whitelabel they can use the same definition and just specify a different model/vendor/description there (so it e.g. shows correctly in the frontend and z2m docs)

Emyrk, 2023-10-30:
In this specific case, I am curious why it had to be added to the `Model` field. I would have guessed I only needed to add it the `whiteLabel` :thinking: