Koenkk/zigbee2mqtt #19465: [New device support]: TuYa RGBW dimmer recognized as RGBCCT bulb

zagi988, 2023-10-28:
### Link

https://www.aliexpress.com/item/1005004630523125.html

### Database entry

{"id":27,"type":"Router","ieeeAddr":"0xa4c138fe3dbb3d59","nwkAddr":6738,"manufId":4107,"manufName":"eWeLight","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"zclVersion":3}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454,"colorMode":2,"currentX":11272,"currentY":48955,"colorTemperature":454,"options":0}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c7c695","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c7c695","endpointID":1}],"configuredReportings":[],"meta":{"onLevelSupported":false}}},"appVersion":3,"stackVersion":2,"hwVersion":0,"dateCode":"20211229","swBuildId":"2.0.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1698529121041,"defaultSendRequestWhen":"immediate"}

### Comments

Tried to write a custom controller for it to not expose "color_temp" but to just have white as second option to RGB but failed because of lack of knowledge.
The model that I have is "5 pin RGBW", so it doesn't have color temperature (2000-6000K) but only has RGB channels and a single white channel.

### External converter

```shell
const extend = require('zigbee-herdsman-converters/lib/extend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const types = require('zigbee-herdsman-converters/lib/types');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const libColor = require('zigbee-herdsman-converters/lib/color');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    model: 'ZB-RGBW',
    fingerprint: [
        {
            modelID: 'ZB-CL01',
            manufacturerName: 'eWeLight'
        },
    ],
    vendor: 'Lonsonho',
    description: 'ZB RGBW dimmer (converted)',
    extend: tuya.extend.light_onoff_brightness_color(),
    exposes: [
        e.light_brightness_color(false)
         .setAccess('color_xy', ea.STATE_SET)
    ],
    toZigbee: utils.replaceInArray(
        tuya.extend.light_onoff_brightness_color().toZigbee,
        [tz.light_color],
        [{
            key: ['color'],
            convertSet: async (entity, key, value, meta) => {
                const color = libColor.Color.fromConverterArg(value);
                const enableWhite =
                    // Zigbee2MQTT frontend white value
                    (color.isXY() && (color.xy.x === 0.3125 || color.xy.y === 0.32894736842105265)) ||
                    // Home Assistant white color picker value
                    (color.isXY() && (color.xy.x === 0.323 || color.xy.y === 0.329));
    
                if (enableWhite) {
                    await entity.command('lightingColorCtrl', 'tuyaRgbMode', {enable: false});
                    const newState = {color_mode: 'color_temp'};
                    if (color.isXY()) {
                        newState.color = color.xy;
                    } else {
                        newState.color = color.rgb.gammaCorrected().toXY().rounded(4);
                    }
                    return {
                        state: libColor.syncColorState(newState, meta.state, entity, meta.options, meta.logger)
                    };
                } else {
                    return await tz.light_color.convertSet(entity, key, value, meta);
                }
            },
        }]
    ),
    meta: {applyRedFix: true},
};

module.exports = definition;
```

### Supported color modes

color (rgb), white

zagi988, 2023-10-31:
It's more of a problem than new device support as more as I think of it. Had to emulate over Node-Red and template light to get it to have sense.

stonegray, 2023-11-02:
I have the bulbs for testing purposes if needed. The fingerprint is the same but the bulbs have firmware version `1.0.13`/`20201216`. The fingerprint is otherwise identical unfortunately.

zagi988, 2023-11-10:
Can anyone at least help me write a custom controller since the fingerprint is the same?

dotWee, 2024-05-12:
> Can anyone at least help me write a custom controller since the fingerprint is the same?

See my workaround fix here: https://github.com/Koenkk/zigbee2mqtt/issues/21361#issuecomment-2105394183

closed 2025-01-11 by github-actions[bot]