Koenkk/zigbee2mqtt #19470: Unable to add Xiaomi QBKG26LM endpoints (left, center, right) to Group

tuonglequoc, 2023-10-29:
### What happened?

I cannot add any endpoint of [QBKG26LM](https://www.zigbee2mqtt.io/devices/QBKG26LM.html#xiaomi-qbkg26lm) into Group.

The scenario is that I want to turn on or off multiple buttons of the switch at the same time. IF I do this in HA, there is a delay, all the lights didn't turn on/off at the same time. So that I come to zigbee2mqtt Groups feature to help.

### What did you expect to happen?

Be able to add Xiaomi QBKG26LM endpoints to Group

### How to reproduce it (minimal and precise)

1. Create Group
2. Add any QBKG26LM endpoint to that Group

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20221226

### Adapter

Noname CC2652P

### Debug log

Debug 2023-10-28 21:21:27Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/add' with data '{"device":"0x00158d0008024ebb/left","group":"Lights","transaction":"frtlo-4"}'
Info 2023-10-28 21:21:27Adding 'Room Switch 1' to 'Lights'
Debug 2023-10-28 21:21:50Error: Command 0x00158d0008024ebb/1 genGroups.add({"groupid":1,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33650 - 1 - 75 - 4 - 0 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)
Info 2023-10-28 21:21:50MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x00158d0008024ebb/left","group":"Lights"},"error":"Failed to add from group (Command 0x00158d0008024ebb/1 genGroups.add({\"groupid\":1,\"groupname\":\"\"}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 33650 - 1 - 75 - 4 - 0 after 10000ms))","status":"error","transaction":"frtlo-4"}'
Error 2023-10-28 21:21:50Failed to add from group (Command 0x00158d0008024ebb/1 genGroups.add({"groupid":1,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33650 - 1 - 75 - 4 - 0 after 10000ms))

tuonglequoc, 2023-10-29:
Database:
`{
    "id": 4,
    "type": "Router",
    "ieeeAddr": "0x00158d0008024ebb",
    "nwkAddr": 33650,
    "manufId": 4151,
    "manufName": "LUMI",
    "powerSource": "Mains (single phase)",
    "modelId": "lumi.switch.n3acn3",
    "epList": [1, 2, 3, 21, 31, 41, 42, 43, 51, 52, 53, 61],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 256,
            "inClusterList": [0, 4, 3, 6, 5, 2, 10, 64704, 2820],
            "outClusterList": [25, 10, 64704],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "lumi.switch.n3acn3",
                        "manufacturerName": "LUMI",
                        "powerSource": 1,
                        "zclVersion": 1,
                        "appVersion": 90,
                        "stackVersion": 2,
                        "hwVersion": 0,
                        "dateCode": "08-13-2021"
                    }
                },
                "aqaraOpple": {
                    "attributes": {
                        "240": 0,
                        "247": {
                            "type": "Buffer",
                            "data": [
                                3, 40, 41, 5, 33, 3, 0, 8, 33, 90, 1, 9, 33, 1, 4, 15, 35, 0, 0, 144, 0, 12, 32, 0, 100,
                                16, 0, 101, 16, 0, 102, 16, 0, 149, 57, 64, 163, 158, 63, 150, 57, 240, 167, 14, 69,
                                151, 57, 0, 0, 0, 0, 152, 57, 0, 134, 195, 60, 154, 32, 0
                            ]
                        },
                        "512": 1,
                        "513": 1,
                        "515": 0
                    }
                },
                "genOnOff": { "attributes": { "245": 58946048, "onOff": 0 } },
                "genGroups": { "attributes": {} },
                "genIdentify": { "attributes": { "identifyTime": 0 } },
                "genScenes": { "attributes": {} },
                "genOta": { "attributes": {} }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "2": {
            "profId": 260,
            "epId": 2,
            "devId": 256,
            "inClusterList": [0, 4, 5, 6, 64704],
            "outClusterList": [],
            "clusters": {
                "genBasic": { "attributes": { "hwVersion": 0 } },
                "genOnOff": { "attributes": { "245": 58946048, "onOff": 0 } },
                "aqaraOpple": { "attributes": { "512": 1 } }
            },
            "binds": [],
            "configuredReportings": [],
[... 103 lines cut ...]
            "meta": {}
        },
        "61": {
            "profId": 260,
            "epId": 61,
            "devId": 256,
            "inClusterList": [18],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 90,
    "stackVersion": 2,
    "hwVersion": 0,
    "dateCode": "08-13-2021",
    "zclVersion": 1,
    "interviewCompleted": true,
    "meta": { "configured": 2114939527 },
    "lastSeen": 1698551541537,
    "defaultSendRequestWhen": "immediate"
}
`

Koenkk, 2023-10-29:
The device doesn't seem to respond to any of the add groups command. If other commands, like turning on/off work, I expect that this device doesn't support groups.

tuonglequoc, 2023-10-29:
Of course all other funtionalities work as expected. Seem the device doesn't support groups. That sounds sad 😞

closed 2024-05-27 by github-actions[bot]