Koenkk/zigbee2mqtt #19471: Tuya BSD29 power readings not updating after awhile

lishengng, 2023-10-29:
### What happened?

The plug's power reading freezes randomly after awhile and only started happening after I paired more than 1 of this plug. I have 5 of these plugs paired currently and when this issue arises there are no logs for the plugs that stop updating power readings.

The only way to amend the issue is to restart the add-on but it does not fix it permanently.

[image]

[image]

### What did you expect to happen?

The power reading updates.

### How to reproduce it (minimal and precise)

-

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20210625

### Adapter

Sonoff Zigbee Dongle-E

ashimokawa, 2023-10-30:
Reminds me of #17604, only fix for me is downgrade to 1.30.3.

lishengng, 2023-10-31:
> Reminds me of #17604, only fix for me is downgrade to 1.30.3.

How do you downgrade?

ashimokawa, 2023-10-31:
@lishengng 

Depends on how you installed, what I did is that in my local git clone:
git stash
git checkout 1.30.3
git stash pop
# edit install.sh and uncomment "git pull"
./install.sh

Try at your own risk, make backups.

mozdi, 2023-11-08:
I´m facing the same issues with my Tuya TS0201 Temperature&Humidity Sensors.
Also downgraded to 1.30.3 this morning. Let´s see if it sorts out the issue for me as well.

ashimokawa, 2023-11-11:
@mozdi 
Any results?

mozdi, 2023-11-11:
No issues so far since the downgrade.

ashimokawa, 2023-11-11:
This bug in there more than half a year ago, right after there were fixes for 0 values reported for some tuya devices, I suspect that all other similar devices broke that did not need this workaround. Just a hunch.

Koenkk, 2023-11-12:
Could you provide the data/database.db entry of your device @ashimokawa or @mozdi ?

ashimokawa, 2023-11-12:
Sure, here is one of the three I have.
Blitzwolf SHP-13
This one had no problem before the "wrong values" commits but right after 1.30.3 they were no longer usable for me.

```
{"id":8,"type":"Router","ieeeAddr":"0x84ba20fffe77b374","nwkAddr":61329,"manufId":4098,"manufName":"_TZ3000_g5xawfcq","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0121","manufacturerName":"_TZ3000_g5xawfcq","stackVersion":0,"dateCode":""}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,9715]}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1,"rmsVoltage":237,"rmsCurrent":0,"activePower":0}},"genOnOff":{"attributes":{"onOff":0,"moesStartUpOnOff":2,"tuyaBacklightMode":1,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe7f588a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe7f588a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe7f588a","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-814849522},"lastSeen":1699794467491,"defaultSendRequestWhen":"immediate"}

```

Koenkk, 2023-11-12:
@ashimokawa with the following ext converter, can you provide the z2m STDOUT until the power measurements stop updating?

https://gist.github.com/Koenkk/09b460e1c99526fa909079f79eda2fc4

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

aaronntw, 2024-01-19:
Facing issue with Tuya BSD29_1 smart plug in Z2M v1.35.1-1. All entities stopped updating and holds the last recorded value, not able to switch the LED on or off, did not test the power switch but assume it is the same as the LED. 

Tried to restart the Z2M add-on but value is still not updating. Only way to get the value to start updating again is to unplug and reinsert to the wall socket.

ctml91, 2024-06-13:
Similar issue with my 2 sengled plugs, at some point over the last year they stopped having their power attribute updated automatically. These devices haven't had any firmware changes, only z2m updates. If I go into z2m and press refresh manually it successfully reports current power but will not continuously update.

Zigbee Model
E1C-NB7

Zigbee Manufacturer
sengled

I will try and re-add the device to z2m and report back if that fixes it.

ctml91, 2024-06-18:
Couldn't get working again by resetting devices. Same issue with two separate z2m instances in different households with the same plugs. Downgrading as suggested by @ashimokawa to v1.30.3 was the only way to get working again on both z2m instances (1.37.X before).

docker.io/koenkk/zigbee2mqtt:1.30.3

I suppose by pinning to this older z2m version I will lose compatibility for newer devices so eventually may become a bigger issue for me, but for now this works.

brunopiras, 2024-06-20:
> Facing issue with Tuya BSD29_1 smart plug in Z2M v1.35.1-1. All entities stopped updating and holds the last recorded value, not able to switch the LED on or off, did not test the power switch but assume it is the same as the LED.
> 
> Tried to restart the Z2M add-on but value is still not updating. Only way to get the value to start updating again is to ~unplug and reinsert to the wall socket~.
> 
> Edit: Unplug and reinsert the smart plug does not get the entities updated. Had to press and hold the button on the smart plug and let it pair again with Z2M, then click reconfigure and the values started updating again. ![20240120-041704_%pn1](https://private-user-images.githubusercontent.com/17488206/298187782-d943f337-2b0f-49e7-a5ec-6779ad8e0752.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg4Njg3ODYsIm5iZiI6MTcxODg2ODQ4NiwicGF0aCI6Ii8xNzQ4ODIwNi8yOTgxODc3ODItZDk0M2YzMzctMmIwZi00OWU3LWE1ZWMtNjc3OWFkOGUwNzUyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjIwVDA3MjgwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThkYmRhNGY5YjE4NmVhNGE1ODVmZWYwODIzMTFmM2IzMGIzZjBkMTQwNmJhMWY5MmY0MGJmZDVmNWMzNTY3NzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yqtbQIjb_HbZS1YX8OGwEInhlNfJ2oEyLIwIwPy7Zog)

Same issue.. 3 of them...

SimeJah, 2024-06-20:
Im having the same issue here..

ctml91, 2024-06-29:
@Koenkk any idea what may be causing the issue in newer version of Z2M? I've rolled back like others to fix the plugs but now I'm having other issues I think because of being on an older release like the MQTT device/entity naming discussed in this thread.

https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8

Koenkk, 2024-06-30:
Could someone try the suggestion from https://github.com/Koenkk/zigbee2mqtt/issues/19471#issuecomment-1807225061 ?

aaronntw, 2024-06-30:
[image]
[image]
@Koenkk did I follow the correct step and safe to reload my config?

brunopiras, 2024-06-30:
> Could someone try the suggestion from https://github.com/Koenkk/zigbee2mqtt/issues/19471#issuecomment-1807225061 ?

Just now.

Koenkk, 2024-06-30:
@aaronntw make sure to add it to `configuration.yaml` only, not the HA addon config.

ashimokawa, 2024-11-13:
@Koenkk 
The issue seems resolved for me now.

I had it on run 1.30.3 (Dongle FW 7.3.0.0) for a year without any problem and a few days ago I updated everything:
- latest zigbee2mqtt
- latest firmware on SONOFF Dongle-E (8.0.2.0)
- switched to ember protocol implementation

And it is also stable, for days now!

closed 2025-01-20 by github-actions[bot]