Koenkk/zigbee2mqtt #19475: Impossible to appair stripe led Livarno home by LIDL

diazovitch, 2023-10-29:
### Link

https://www.lidl.be/p/fr-BE/livarno-home-ruban-a-led-smart-home-2-m/p100337807

### Database entry

Impossible to pair it

### Comments

Hi,

I have a new stripe led form Lidl Livarno home HG08132 but it's impossible to pair it.
I had already a similar stripe led from LIDL but it is a HG06104A and it work properly.
What is the problem please ?

Sorry I'm not an expert.

I use domoticz with zigbee2mqtt 1.33.0 and plugin  3.2.0 beta.

Thank you.

### Supported color modes

color temperature and RGB color

diazovitch, 2023-10-31:
Here you can find the log :

`info  2023-10-31 20:52:07: Device '0x003c84fffeb13e79' joined
info  2023-10-31 20:52:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x003c84fffeb13e79","ieee_address":"0x003c84fffeb13e79"},"type":"device_joined"}'
info  2023-10-31 20:52:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x003c84fffeb13e79"},"type":"device_connected"}'
info  2023-10-31 20:52:08: Starting interview of '0x003c84fffeb13e79'
info  2023-10-31 20:52:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x003c84fffeb13e79","ieee_address":"0x003c84fffeb13e79","status":"started"},"type":"device_interview"}'
info  2023-10-31 20:52:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x003c84fffeb13e79"},"type":"pairing"}'`

diazovitch, 2023-11-05:
Nobody to help me please ?

Allgoyer, 2024-01-07:
Hi, 

did you find a solution and achieve a pairing? I have the same problem.

regards allgoyer

diazovitch, 2024-01-07:
I have not been successful to date. I installed a new installation of domoticz on another rpi with ZigbeeForDomoticz and it works perfectly. But that doesn't solve my problem because my installation is based on zigbee2mqtt and I don't want to re-pair everything on the other test installation.

Allgoyer, 2024-01-08:
Strangely enough, I can pair and use the LED strip via a Conbee 2 stick with zigbee2mqtt on version 1.30.2. However, this does not work with a Sonoff Zigbee stick on version 1.35.1 (home assistant). Can anyone help here?
Thanks

diazovitch, 2024-01-08:
it's very interesting that you say because my installation with ZigbeeForDomoticz use a CC2531 and it works but on my installation with the sonoff stick and zigbee2mqtt it doesn't work.

Allgoy, 2024-01-09:
Then it seems like it's a problem with the sonoff stick

closed 2024-08-08 by github-actions[bot]

floriaaan, 2025-01-12:
I can pair this device with a Sonoff Zigbee adapter but I got a problem when turning off the device, it don't power off instead it goes to a yellow/warm white.

Can someone add support to the HG08132 device or at least, provide more informations on how-to ? (how to avoid Z2M to identify it as [HG06104A](https://www.zigbee2mqtt.io/devices/HG06104A.html#lidl-hg06104a) ?)

Thanks :)