Koenkk/zigbee2mqtt #19477: [New device support]: LYWSD03MMC using https://github.com/devbis/z03mmc

orrpan, 2023-10-29:
### Link

https://www.aliexpress.com/item/1005005916924715.html?algo_pvid=0ee04745-9791-4204-b98d-297a2a407fb7&algo_exp_id=0ee04745-9791-4204-b98d-297a2a407fb7-0&pdp_npi=4%40dis%21SEK%21239.01%2155.11%21%21%21152.72%21%21%402103849716986010100653262e4978%2112000034841017184%21sea%21SE%210%21AB&curPageLogUid=lWiFVFxobeeX

### Database entry

{"id":22,"type":"EndDevice","ieeeAddr":"0xa4c138e7285c16e8","nwkAddr":44761,"manufId":4417,"manufName":"Xiaomi","powerSource":"Battery","modelId":"LYWSD03MMC","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1,1029,1026],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":14,"dateCode":"20231027","swBuildId":"3001-1004","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1698601079407,"defaultSendRequestWhen":"immediate"}

### Comments

I don't have time to develop right now, but would be awesome if someone did!

Koenkk, 2023-10-29:
Already added and will be supported in the 1 November release.

closed 2023-10-29 by Koenkk

salopette, 2023-12-22:
Can you please include the 1.0.6 FW update in the database!?

https://github.com/devbis/z03mmc/releases/download/1.0.6/ATC_ota_40000.bin