Koenkk/zigbee2mqtt #19478: Aqara WS-EUK04 no actions in decoupled mode

Scoff123, 2023-10-29:
### What happened?

No button press actions are displayed in the state window, or passed to HA when in decoupled mode. Seen various previous reports of this, for this device and similar Aqara light switches and, seemingly, updating coordinator firmware to update zstack seems to resolve. 

### What did you expect to happen?

Button press actions displayed in state viewer, like left button single, right button double etc.

### How to reproduce it (minimal and precise)

Add the WS-EUK04 to Z2M by pairing with zigbee coordinator. Set either or both left or right switch to decoupled mode. Press either button, and there are no button press actions shown in the state view, or passed to HA.

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Sonoff ZB Bridge 

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13198784/log.txt)

Scoff123, 2023-12-05:
@Koenkk - is there any more info you need, or logs I can provide in order to help with this?

Scoff123, 2024-01-13:
Similar issue where zstack firmware update was suggested and reported as fixing the issue - what can be done when the coordinator is EZSP, i.e., Sonoff ZBBridge running Tasmota?

[#17305](https://github.com/Koenkk/zigbee2mqtt/issues/17305)

Koenkk, 2024-01-13:
You should raise this with the firmware maintainers (I'm not maintaining the EZSP fw). Here is a link to the [zstack fix](https://github.com/Koenkk/Z-Stack-firmware/blob/4e888d3310eaf24c19699b00a29e24d512e49e8c/coordinator/Z-Stack_3.x.0/firmware.patch#L675).

closed 2024-01-13 by Koenkk