Koenkk/zigbee2mqtt #19480: [Feature request]: Group Defaults

xcorp, 2023-10-29:
### Is your feature request related to a problem? Please describe

I want to be able to set a default key/value for groups, in this case retain
I found the device_options key but can't find one for groups?

### Describe the solution you'd like

Create a group_options key to be able to set default values for groups.

### Describe alternatives you've considered

NA

### Additional context

NA

EmJay276, 2024-01-04:
Would love to have this feature!

I am using default device options, but they get overwritten by a group. f.e.
```
device_options:
  transition: 1
```
get overwritten / ignored by the group back to `transition: 0`

xcorp, 2024-07-06:
Still interested in this, commenting just to keep the issue from going stale.

closed 2025-05-15 by github-actions[bot]