Koenkk/zigbee2mqtt #19487: IKEA E1524/E1810 missing exposes

tobiasbrage, 2023-10-30:
### What happened?

I'm missing quite a lot of exposes from this remote.

The page on the zigbee2mqtt website states the following:

Action (enum)
Triggered action (e.g. a button click). Value can be found in the published state on the action property. It's not possible to read (/get) or write (/set) this value. The possible values are: arrow_left_click, arrow_left_hold, arrow_left_release, arrow_right_click, arrow_right_hold, arrow_right_release, brightness_down_click, brightness_down_hold, brightness_down_release, brightness_up_click, brightness_up_hold, brightness_up_release, toggle.

I'm not seeing any exposes from either brightness_up og arrow_left.

### What did you expect to happen?

I expected all exposes to be present.

The remote is updated to the latest firmware 24.4.5, as far as I can tell at least.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20231005.0

### Adapter

Home Assistant SkyConnect

larstobi, 2024-04-29:
Not stale.

dokatomi, 2024-05-03:
The arrow_left_hold and arrow_right_hold not working in the 1.37.0-1 version. The previous HA addon version 1.36.1-1 has no issue with this remote.

rdbraber, 2024-05-03:
Having the same issue after upgrading to 1.37.0. 1.36.0 works fine.

der-gordon, 2024-05-03:
> The arrow_left_hold and arrow_right_hold not working in the 1.37.0-1 version. The previous HA addon version 1.36.1-1 has no issue with this remote. Others for example brightness_up_hold and brightness_down_hold working for me.

Having exactly the same problem. Does anyone have suggestions?

dokatomi, 2024-05-03:
Currently I downgraded to the previous version.

doctordarko, 2024-05-03:
> The arrow_left_hold and arrow_right_hold not working in the 1.37

I have tried just about every combo of device reconfigure, remove device, factory reset, battery out, z2m restart to no avail. Everything but left/right hold works.

Xploder, 2024-05-04:
Can confirm on my E1810. Since 1.37 left/right hold doesn't work anymore.

Koenkk, 2024-05-04:
Could you provide the debug log when triggering the left/right hold action?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Xploder, 2024-05-04:
Hey Koenkk, sure thing.

This is with Right Arrow held:
https://pastebin.com/AwuBuWKc

Thank you.

Koenkk, 2024-05-04:
Does it work after setting `legacy: false` for this device? ([docs](https://www.zigbee2mqtt.io/devices/E1524_E1810.html#options))

Xploder, 2024-05-04:
This is with `legacy: false`.
`legacy: true` also does not work.

Koenkk, 2024-05-05:
Pushed a fix, let me know if it works!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

doctordarko, 2024-05-05:
> let me know if it works!

Left/right hold actions work on E1810 with 1.37.0-dev commit: [358e2ee]

rdbraber, 2024-05-06:
I can also confirm the fix works.

Koenkk, 2024-05-06:
Awesome!

closed 2024-05-06 by Koenkk