Koenkk/zigbee2mqtt #19488: [New device support]: QS-Zigbee-ECC02

buldoog, 2023-10-30:
### Link

https://a.aliexpress.com/_EusyN47

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x00124b002471bdbf","nwkAddr":56600,"manufId":0,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"ZB-SW02","epList":[1,2],"endpoints":{"1":{" profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genBasic":{"attributes":{"modelId":"ZB-SW02","manufacturerName": "eWeLink","powerSource":1,"zclVersion":1,"appVersion":8,"hwVersion":1,"dateCode":"20210703","swBuildId":"1.0.7"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001949ff68"," endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genBa sic":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001949ff68","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":8,"hwVersion":1,"dateCod e":"20210703","swBuildId":"1.0.7","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1698168834707,"defaultSendRequestWhen":"immediate"}

### Comments

The device is detected as ZB-SW02 but not working, Not reporting LQI.

buldoog, 2023-11-02:
After reset is detected as [QS-Zigbee-C03](https://www.zigbee2mqtt.io/devices/QS-Zigbee-C03.html#lonsonho-qs-zigbee-c03) and working fine.

rubentest, 2024-02-16:
Hello @buldoog I have the same, detected as QS-Zigbee-C03 but I can't calibrate time.

You can ?