Koenkk/zigbee2mqtt #19489: Xiaomi WSDCGQ11LM temperature calibration only possible in whole integers

habitoti, 2023-10-30:
### What happened?

The temperature calibration allows only whole integers (-1, 1, ...) in the calibration dialog. This is usually too much and makes not much sense for such a device.

### What did you expect to happen?

Temperature calibration should accept a float for calibration. At least +/- 0.1 steps should be supported by the UI.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20230507

### Adapter

Sonoff 3.0

lux73, 2023-11-01:
in my opinion this should be good for these Devices - it has only an accuracy of  ±3%

habitoti, 2023-11-01:
But that deviation is rather static per device. And even then, 3% on 20 Degrees is 0.6 degrees, not 1.
Once I found the actual deviation (which can as well also be just 1%...), I'd like to adjust it accordingly, but 1 degree steps is far too much. At least .5 would do, but why not allow for .1 steps (like most other temperature sensors allow)?

lux73, 2023-11-01:
keep in Mind that are not highly accurate Sensors - but if you want to calibrate these, why not implement this via your preferred Home Automation Software?

i doubt that the Dev's will implement 0.1 steps

habitoti, 2023-11-01:
I have an additional HA template sensor defined that adjusts by the .3 I need. However it's just strange I even have to do that if there is already an adjustment available in the settings. Shouldn't be a major effort to allow .1 (or .5) steps instead of just 1.  

From my experience, comparing the readings of the sensors with calibrated thermometers they are pretty accurate over time. If you'd properly adjust the offset once, they are absolutely fit for purpose in home settings.

habitoti, 2023-11-02:
So what I just found in addition: after using the calibration from the UI at all, the log continuously shows: "Exception while calling fromZigbee converter: 'temperature_calibration' is not a number, got string ()}", so it seems generally sth. wrong with it. Maybe it is in the end even supposed to allow proper (lower) calibration values (like most other climate entities...)?

lux73, 2023-11-02:
i can't second this - for me calibration works as expected

if using plain z2m check your config:

`temperature_calibration: -1`

no other character allowed!

habitoti, 2023-11-02:
How do you know that it's expected that floats are not allowed. Is that documented anywhere?

lux73, 2023-11-02:
for sure:

https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html#xiaomi-wsdcgq11lm

https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html#options

if you put a float inside it will throw the Error you can see ^^ atm not possible other than a Number.

habitoti, 2023-11-02:
AFAIK there is no specific "float". A number can be float or integer.

lux73, 2023-11-02:
afaik you can only insert integer - for me its working properly

good luck by fixing your "problem"

habitoti, 2023-11-02:
Why are you like that? Just because you don't have the need doesn't mean nobody else does not either. Why are you even coming to this thread then? Isn't this to report issues? I fell like this is one. If you don't: lucky you!

user34756361233, 2023-12-07:
Add the compensation in configuration.yaml of zigbee2mqtt:

`  '0x00158d000xxxxyyyy':
    friendly_name: hal achter xaomi
    temperature_precision: 1
    temperature_calibration: -1.8`

habitoti, 2023-12-08:
Thanks. This even more so stresses my point that this should also be properly reflected in the Z2M UI accordingly and allow for .1 steps. Would be bard to properly and consistently maintain that over time else.

closed 2024-07-06 by github-actions[bot]