Koenkk/zigbee2mqtt #19500: One paired device has desapear from panel

horttorrell32, 2023-10-31:
### What happened?

I had a device paired and it had been working for months, it was also included in a group used for automations. I don't know when, but it disappeared and doesn't show up on the dashboard and I can't trade with it since, but it's still working with the group's request.

### What did you expect to happen?

Some way to show the device back on the dashboard and I can operate with it again. Since it looks paired because the group request works, I need some way to do it without re-pairing, since the device is in a hard-access place.

### How to reproduce it (minimal and precise)

I don't know

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20221226

### Adapter

ZBDongle-P (zStack3x0)

maumi, 2023-10-31:
I cannot really help you, but I have the same behaviour. 
A minimum of 2 devices are disappeared sice some days. 
All from the same device type. 
I re-paired one and it re-appeared and works again as expected. 
but another one is harder to reach so it would be nice to have an alternative way of repairing.

horttorrell32, 2023-11-05:
Can I modify the database in someway to allow the device come back to Dashboard?
It seems paired because It works as per gruop request.

qq547176052, 2023-11-18:
我也有同样的问题

jack9603301, 2023-11-18:
> 我也有同样的问题

不要再国际社区使用中文

peter1rhodes, 2023-11-22:
I can't be sure this is your issue, but I've recently found a similar thing, and I suspect it was because the devices were part of a Group that had the option selected to then hide the individual member devices from HA. In my case, the devices had been removed and the Group deleted, so I can't even fix it that way! Have a look in your `configuration.yaml` for z2m, and see whether it says something like `home_assistant: null` (apologies, I've now deleted these lines so can't remember what it was actually called! But all my missing devices had an entry with a `null` value, and none of the others did)

horttorrell32, 2023-12-25:
Hello all,
I have finally resolved the issue.
I found an old backup of z2m and compared. Then I saw that device was removed from database.db and devices.yaml, but It was included in the group.yaml list and this is why launch group the device works.
Then I stop z2m, I did a very acuratte merge in the database.db and devices.yaml only with the device removed. I knew which one was because in the list of group I found the ieeeadress.
I start the z2m and the device reappear but disconected, after some manual actions, z2m correct the nwkadress and the device come back!!
Thanks and I Hope this can help.
Merry Christmas.

closed 2023-12-28 by horttorrell32