Koenkk/zigbee2mqtt #19506: Woox Smart Smoke Alarm R7049 not detected

crudooper, 2023-10-31:
### What happened?

Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_bxdyeaa9'
Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### What did you expect to happen?

As the device to be recognized as it is supported :
https://www.zigbee2mqtt.io/devices/R7049.html

### How to reproduce it (minimal and precise)

Activate pairing mode
Press 5 seconds on smart smoke alarm button
Zigbee2mqtt detect the device, but is not able to recognized it

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20210708

### Adapter

ZBDongle-P

crudooper, 2023-11-16:
I tried to add it with external_connector :
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const newWoox = {
	R7049_status: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp; // First we get the data point ID
                const value = getDataValue(dpValue); // This function will take care of converting the data to proper JS type
                switch (dp) {
                case 1:
                    result.smoke = Boolean(!value);
                    break;
                case 8:
                    result.test_alarm = value;
                    break;
                case 9: {
                    const testAlarmResult = {0: 'checking', 1: 'check_success', 2: 'check_failure', 3: 'others'};
                    result.test_alarm_result = testAlarmResult[value];
                    break;
                }
                case 11:
                    result.fault_alarm = Boolean(value);
                    break;
                case 14: {
                    const batteryLevels = {0: 'low', 1: 'middle', 2: 'high'};
                    result.battery_level = batteryLevels[value];
                    result.battery_low = value === 0;
                    break;
                }
                case 16:
                    result.silence_siren = value;
                    break;
                case 20: {
                    const alarm = {0: true, 1: false};
                    result.alarm = alarm[value];
                    break;
                }
                default:
                    meta.logger.debug(`zigbee-herdsman-converters:Woox Smoke Detector: NOT RECOGNIZED DP #${
                        dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
	},
    R7049_testAlarm: {
        key: ['test_alarm'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, 8, value);
        },
    },
    R7049_alarm: {
        key: ['alarm'],
        convertSet: async (entity, key, value, meta) => {
            const linkAlarm = {true: 0, false: 1};
            await tuya.sendDataPointEnum(entity, 20, linkAlarm[value]);
        },
    },
	R7049_silenceSiren: {
        key: ['silence_siren'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, 16, value);
        },
	},
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_bxdyeaa9'}],
        model: 'R7049',
        vendor: 'Woox',
        description: 'Smart smoke alarm',
        meta: {timeout: 30000, disableDefaultResponse: true},
        fromZigbee: [newWoox.R7049_status, fz.ignore_tuya_set_time, fz.ignore_time_read],
        toZigbee: [newWoox.R7049_silenceSiren, newWoox.R7049_testAlarm, newWoox.R7049_alarm],
        exposes: [e.battery_low(),
            e.binary('smoke', ea.STATE, true, false).withDescription('Smoke alarm status'),
            e.binary('test_alarm', ea.STATE_SET, true, false).withDescription('Test alarm'),
            e.enum('test_alarm_result', ea.STATE, ['checking', 'check_success', 'check_failure', 'others'])
                .withDescription('Test alarm result'),
            e.enum('battery_level', ea.STATE, ['low', 'middle', 'high']).withDescription('Battery level state'),
            e.binary('alarm', ea.STATE_SET, true, false).withDescription('Alarm enable'),
            e.binary('fault_alarm', ea.STATE, true, false).withDescription('Fault alarm status'),
            e.binary('silence_siren', ea.STATE_SET, true, false).withDescription('Silence siren')],
    },
];`

but I got an error when getting values :
`Exception while calling fromZigbee converter: getDataValue is not defined}`

Any idea to help me

crudooper, 2023-11-16:
I tried to add it with external_connectors with :
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const newWoox = {
	R7049_status: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp; // First we get the data point ID
                const value = getDataValue(dpValue); // This function will take care of converting the data to proper JS type
                switch (dp) {
                case 1:
                    result.smoke = Boolean(!value);
                    break;
                case 8:
                    result.test_alarm = value;
                    break;
                case 9: {
                    const testAlarmResult = {0: 'checking', 1: 'check_success', 2: 'check_failure', 3: 'others'};
                    result.test_alarm_result = testAlarmResult[value];
                    break;
                }
                case 11:
                    result.fault_alarm = Boolean(value);
                    break;
                case 14: {
                    const batteryLevels = {0: 'low', 1: 'middle', 2: 'high'};
                    result.battery_level = batteryLevels[value];
                    result.battery_low = value === 0;
                    break;
                }
                case 16:
                    result.silence_siren = value;
                    break;
                case 20: {
                    const alarm = {0: true, 1: false};
                    result.alarm = alarm[value];
                    break;
                }
                default:
                    meta.logger.debug(`zigbee-herdsman-converters:Woox Smoke Detector: NOT RECOGNIZED DP #${
                        dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
	},
    R7049_testAlarm: {
        key: ['test_alarm'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, 8, value);
        },
    },
    R7049_alarm: {
        key: ['alarm'],
        convertSet: async (entity, key, value, meta) => {
            const linkAlarm = {true: 0, false: 1};
            await tuya.sendDataPointEnum(entity, 20, linkAlarm[value]);
        },
    },
	R7049_silenceSiren: {
        key: ['silence_siren'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, 16, value);
        },
	},
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_bxdyeaa9'}],
        model: 'R7049',
        vendor: 'Woox',
        description: 'Smart smoke alarm',
        meta: {timeout: 30000, disableDefaultResponse: true},
        fromZigbee: [newWoox.R7049_status, fz.ignore_tuya_set_time, fz.ignore_time_read],
        toZigbee: [newWoox.R7049_silenceSiren, newWoox.R7049_testAlarm, newWoox.R7049_alarm],
        exposes: [e.battery_low(),
            e.binary('smoke', ea.STATE, true, false).withDescription('Smoke alarm status'),
            e.binary('test_alarm', ea.STATE_SET, true, false).withDescription('Test alarm'),
            e.enum('test_alarm_result', ea.STATE, ['checking', 'check_success', 'check_failure', 'others'])
                .withDescription('Test alarm result'),
            e.enum('battery_level', ea.STATE, ['low', 'middle', 'high']).withDescription('Battery level state'),
            e.binary('alarm', ea.STATE_SET, true, false).withDescription('Alarm enable'),
            e.binary('fault_alarm', ea.STATE, true, false).withDescription('Fault alarm status'),
            e.binary('silence_siren', ea.STATE_SET, true, false).withDescription('Silence siren')],
    },
];
```

But I got : 
```
Exception while calling fromZigbee converter: getDataValue is not defined}
Exception while calling fromZigbee converter: getDataValue is not defined}
Exception while calling fromZigbee converter: getDataValue is not defined}
Exception while calling fromZigbee converter: getDataValue is not defined}
```

Could you give me a tip, please ?

JakubGoliasz, 2023-11-17:
That's what I have and it works for me. I combined the solutions I found here. It's important to use "lib/legacy" instead of "lib/tuya". I hope I helped.

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/legacy');

const woox = {
    fz: {
        status: {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'commandDataReport'],
            convert: (model, msg, publish, options, meta) => {
                const result = {};
                for (const dpValue of msg.data.dpValues) {
                    const dp = dpValue.dp; // First we get the data point ID
                    const value = tuya.getDataValue(dpValue); // This function will take care of converting the data to proper JS type
                    switch (dp) {
                    case 1:
                        result.smoke = Boolean(!value);
                        break;
                    case 8:
                        result.test_alarm = value;
                        break;
                    case 9:
                        const test_alarm_result = {0: 'checking', 1: 'check_success', 2: 'check_failure', 3: 'others'};
                        result.test_alarm_result = test_alarm_result[value];
                        break;
                    case 11:
                        result.fault_alarm = Boolean(value);
                        break;
                    case 14:
                        const battery_levels = {0: 'low', 1: 'middle', 2: 'high'};
                        result.battery_level_state = battery_levels[value];
                        break;
                    case 16:
                        result.silence_siren = value;
                        break;
                    case 20:
                        const alarm = {0: true, 1: false};
                        result.alarm = alarm[value];
                        break;
                    default:
                        meta.logger.warn(`zigbee-herdsman-converters:Woox Smoke Detector: NOT RECOGNIZED DP #${
                            dp} with data ${JSON.stringify(dpValue)}`); // This will cause zigbee2mqtt to print similar data to what is dumped in tuya.dump.txt
                    }
                }
                return result;
            },
        },
    },

    tz: {
        silence_siren: {
            key: ['silence_siren'],
            convertSet: async (entity, key, value, meta) => {
                await tuya.sendDataPointBool(entity, 16, value); // tuya.sendDataPoint* functions take care of converting the data to proper format
            },
        },
        test_alarm: {
            key: ['test_alarm'],
            convertSet: async (entity, key, value, meta) => {
                await tuya.sendDataPointBool(entity, 8, value); // tuya.sendDataPoint* functions take care of converting the data to proper format
            },
        },
        alarm: {
            key: ['alarm'],
            convertSet: async (entity, key, value, meta) => {
                const link_alarm = {true: 0, false: 1}
                await tuya.sendDataPointEnum(entity, 20, link_alarm[value]); // tuya.sendDataPoint* functions take care of converting the data to proper format
            },
        },
    },
};

module.exports = [{
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_bxdyeaa9'}],
    model: 'Zigbee R7049 Smoke Alarm',
    vendor: 'Woox',
    description: 'Woox Zigbee R7049 Smoke Alarm',
    meta: {configureKey: 1, timeout: 30000, disableDefaultResponse: true},
    
    fromZigbee: [woox.fz.status, fz.ignore_tuya_set_time, fz.ignore_time_read],
    toZigbee: [woox.tz.silence_siren, woox.tz.test_alarm, woox.tz.alarm],
    
    exposes: [e.battery_low(),
        exposes.binary('smoke', ea.STATE, true, false).withDescription('Smoke alarm status'),
        exposes.binary('test_alarm', ea.STATE_SET, true, false).withDescription('Test alarm'),
        exposes.enum('test_alarm_result', ea.STATE, []).withDescription('Test alarm result'),
        exposes.enum('battery_level_state', ea.STATE, []).withDescription('Battery level state'),
        exposes.binary('alarm', ea.STATE_SET, true, false).withDescription('Alarm enable'),
        exposes.binary('fault_alarm', ea.STATE, true, false).withDescription('Fault alarm status'),
        exposes.binary('silence_siren', ea.STATE_SET, true, false).withDescription('Silence siren')],
}];
```

Koenkk, 2023-11-17:
@JakubGoliasz could you make a pr for this?

crudooper, 2023-11-19:
Many thanks @JakubGoliasz , it's working like a charm !

xeroski, 2023-12-19:
Hi. Which version of Z2M will this change be implemented in?

synga-nl, 2023-12-26:
@Koenkk @JakubGoliasz Is there any progression for this issue?

JakubGoliasz, 2023-12-26:
Sorry, I haven't had the opportunity to be here longer lately.  If anyone has some time, please add this converter to the official solution.

dmit2k, 2024-01-28:
Hello, 

any progress so far in going official?

gednet, 2024-02-15:
> Hello,
> 
> any progress so far in going official?

I will glad that someone repair that, because official integration not working property, but in devices is marked as supported.
https://www.zigbee2mqtt.io/devices/R7049.html

Koenkk, 2024-02-18:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-18 by Koenkk