Koenkk/zigbee2mqtt #19509: Bticino K4027C cover position not working

syntesys87, 2023-11-01:
### What happened?

I added to Zigbee2Mqtt my 6 covers Bticino K4027C; the problem is that the setting of cover position does not work; works only 0-100 for open or close.

This is the status JSON:
{
    "linkquality": 128,
    "position": 100,
    "state": "CLOSE",
    "tilt": 0,
    "update": {
        "installed_version": 3556351,
        "latest_version": 3556351,
        "state": "idle"
    },
    "action": null,
    "led_in_dark": null,
    "update_available": null
}

And I have not activated "No position support" because in this JSON I see that I have the latest version.

### What did you expect to happen?

The cover sets the right position.

### How to reproduce it (minimal and precise)

Add a new BTicino device and try to set the position.

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12 SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

syntesys87, 2023-11-02:
Also, I found that in the new release of ZigBee2MQTT there is this commit: [https://github.com/Koenkk/zigbee-herdsman-converters/pull/6333](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6333)... Could this be ported also for Bticino K4027C?

Thank you.

syntesys87, 2023-11-03:
@FabianMangold I opened yesterday this issue after seeing your [pull request](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6333) and I would integrate it into K4027C, because the cover position and the calibration is not working (I don't know il tilt works, I have covers without tilt)... How did you approach the debug of NLLV firmware and the tries you did for make this working? I would debug how it works in the app and replicate it in Zigbee2MQTT.

I see from logs of Netatmo plugin in Home Assistant this: 2023-11-03 11:47:33.293 DEBUG (MainThread) [pyatmo.home] Setting state for home (62134d6bd95995623b03a8ba) according to {'modules': [{'id': '---', 'target_position': 49, 'bridge': ---'}]} (I removed MAC addresses from component and bridge).

syntesys87, 2023-11-03:
Thank you for the support! 
I'm using 1.33.2 with HomeAssistant 2023.11.0, last version of everything. Everything is connected to ITEAD Sonoff ZBDongle-E (ZigBee 3.0 USB Dongle Plus v2) with last firmware.
The device is recognized (after last update of Z2M) as 067776A. 
The device works only on up-down-stop, automatic calibration in classic_nllv is not working, works on specific_nllv (but useless, because the position after calibration doesn't work), not working in manual with temporal and venetian.

FabianMangold, 2023-11-03:
My pleasure, thanks for the update. It makes more sense now.
I've updated the [comment](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6333#issuecomment-1790548665) regarding the calibration modes in the PR#6333.
Basically, classic_nllv seems to be identical to up_down_stop, no calibration required. specific_nllv is the mode that features the automatic calibration.
You may need to calibrate the blind several time before achieving good results, but i found the temporal mode to be more precise as well!
Z2M doesn't do anything here, it merely sets the switch in the right mode, the "rest" is done by the switch firmware itself. We can only hope that newer firmware versions may improve those calibration methods.
When you say the manual temporal / venetian calibration does not work, do you mean the "calibration procedure" or the results afterwards ?

syntesys87, 2023-11-03:
I mean the calibration procedure, I tested temporal yesterday, following this:
"Once the blind is in the up position, start the calibration by pressing and holding the down button, until the blind reaches to bottom stop. Once there, the relay should click. "
At the end, the relay does not click.

FabianMangold, 2023-11-03:
Okay, that might be somewhat different on the Bticino. Just skip that part, and continue.
Do you have an original gateway and the corresponding app ? If so, you could try to calibrate the device using those "tools" and update our procedures.
Again, those procedures were determined OK for the Legrand 067776 / 067776A "only".

syntesys87, 2023-11-03:
It seems that when I press the reset button for run the association, it loses the calibration; I already tested readding to original gateway, calibrating and readding to zigbee2mqtt, but this does not work.

FabianMangold, 2023-11-03:
The calibration is "stored" in the switch, and re-associating indeed seems to clear those values.
Nevertheless, can you try the "same calibration" procedure once the device is associate to Z2M ?
Is the procedure any different from the one that we had described in the PR ticket ?

syntesys87, 2023-11-03:
The procedure is the same, on original app it runs an automatic calibration and then it works. With Z2M I set the shutter in specific_nllv mode and it runs automatic calibration... The procedure seems the same! 
In the process there are two things different: the red led does not turn off after the association with Z2M and if I hold up+down button for ten seconds, it doesn't run automatic calibration

FabianMangold, 2023-11-03:
I was actually tallking about the manual calibration process, not the automatic one.
"What" does work when your use the original app ? The position report ?

Regarding the LED, there is another thread / user reporting the same issue, which i cannot reproduce on my own devices. (See https://github.com/Koenkk/zigbee2mqtt/issues/19108#).
Why would you press Up+Down? Switching modes via Z2M is sufficient to bring the switch into the calibration mode, there is no need for physically pressing the buttons.

syntesys87, 2023-11-03:
Yes, but tried also original method for seeing the result.
Regarding manual mode, from BTicino it works in this mode: press down button until it closes, press up button until it opens, press down button until the blind touches the window surface, press up button to reopen... But setting temporal mode, after I press the up button, it exit from calibration procedure.

FabianMangold, 2023-11-03:
OK for the calibration, to be investigated.
> Yes, but tried also original method for seeing the result. 
What does "Yes" refer to?
- Does the position report work after the automatic calibration ? Does it as reliably in Z2M as it does in the original app
- Just to confirm: Tilt doesn't work in that mode right ? Neither in Z2M nor with the original GW

syntesys87, 2023-11-03:
Yes was an ok for the up+down press for calibration 😁
After automatic calibration, it continues to report 0-100-50 (open, close, other positions), not, for example, 27 or 69.
I don't have blinds with tilt, only up-down.

FabianMangold, 2023-11-03:
Sorry for the stupid questions, just making sure we are talking about the same things.

Okay, that is really odd, the automatic calibration or temporal mode should actually enable proper position reporting.
Is this also the case with the original GW or only Z2M ?

syntesys87, 2023-11-03:
With Z2M seems that is not working, with original GW everything works as expeced; also, the position reporting works well with the original Netatmo integration in Home Assistant.

FabianMangold, 2023-11-03:
Can you provide you database.db here please ?

syntesys87, 2023-11-03:
No problem, do you need it with a shutter configured? Because, at the moment, they are all connected with original GW.

FabianMangold, 2023-11-03:
Yes please, whenever you have time, no hurry.

syntesys87, 2023-11-03:
[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/13253062/database.db.zip)

Here attached the file. Now i try temporal mode :)

syntesys87, 2023-11-03:
[This](https://photos.app.goo.gl/yzbzXbHR5bT6va9YA) is the link of the temporal try. When it starts, it goes on red led and then the switch sets into up_down_stop.

FabianMangold, 2023-11-03:
Can you please leave the system running for a while and re-upload the database.db ? Unfortunately it doesn't contain yet any attributes, which makes it kind of worthless for now.
Thanks a lot.

FabianMangold, 2023-11-03:
Thanks for the video, that helps :)
Can you switch it to temporal, and really wait for the blind to go to the upper stop. Once it is there, give it a couple of seconds, and then press the switch. Just don't touch the switch until the blind is in a steady position.

syntesys87, 2023-11-03:
Yes, no problem, I will send the new file tomorrow, so that attributes could be populated.
In the video, I start from the upper position, but when I click the down button, happens red led and exit from calibration.

philippk75, 2023-11-03:
hello, with my bticino N4027C the poistion control works, i calibrated it with specific calibration automatic mode, also different percentsges of positions work.

syntesys87, 2023-11-04:
[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/13255997/database.db.zip)
Here attached new database.db. 
@philippk75 what firmware are you running in your N4027C?

philippk75, 2023-11-04:
> [database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/13255997/database.db.zip) Here attached new database.db. @philippk75 what firmware are you running in your N4027C?

Firmware-Date
20230406
Firmware-Version
0036

syntesys87, 2023-11-04:
The version is the same, mine switches has also 0036 firmware. Do you have blinds with tilt or lift only? Could the problem be related to that?

philippk75, 2023-11-04:
> The version is the same, mine switches has also 0036 firmware. Do you have blinds with tilt or lift only? Could the problem be related to that?

i have both, tilt and non tilt

FabianMangold, 2023-11-05:
Hi @philippk75 and @syntesys87,

@philippk75:
I've just submitted [PR#6405](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6405) to fix the device detection. Once merged, your switches will be reported as BTicino devices again, still keeping the new feature set.
Please let me know if this worked and if that's okay.

@syntesys87:
Thanks for sharing your database.db, this helped to confirm the HW ID of Bticino switches.
Also, the database shows that your switch is calibrated in NLLV_classic, meaning Up/Down/Stop, which doesn't report any position.
Could you please retry the automatic calibration (NLLV_Specific), and make sure that the "No Position Reporting" option located in the "Settings (Specific)" tabsheet is set to OFF ?

Thanks
Th

philippk75, 2023-11-05:
> Hi @philippk75 and @syntesys87,
> 
> @philippk75: I've just submitted [PR#6405](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6405) to fix the device detection. Once merged, your switches will be reported as BTicino devices again, still keeping the new feature set. Please let me know if this worked and if that's okay.
> 
> @syntesys87: Thanks for sharing your database.db, this helped to confirm the HW ID of Bticino switches. Also, the database shows that your switch is calibrated in NLLV_classic, meaning Up/Down/Stop, which doesn't report any position. Could you please retry the automatic calibration (NLLV_Specific), and make sure that the "No Position Reporting" option located in the "Settings (Specific)" tabsheet is set to False ?
> 
> Thanks

hello, thanks, this means that we have to wait the next zibee2mqtt update? or can we start somethings manuel?

syntesys87, 2023-11-05:
Hello all and thanks for the support! 
@FabianMangold no problem for trying, it's very strange that in the frontend I see that calibration mode is set to specific_nllv and in the database is set to classic_nllv... Are you looking to the switch with friendly name 0x0004740000cb2a3b, right?
How can I install last update? Is there a guide to follow for test latest PR?

Thank you!

FabianMangold, 2023-11-05:
Hello both,
The fix has just been merged, so it should be available in the DEV branch shortly.

Regarding the usage of the DEV branch, you may want to checkout this page and follow the instructions depending on the way you are using Z2M:
https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

And lastly, yes @syntesys87, i was looking at exactly this switch, the entry having ID:7. If you check the reporings in the "closuresWindowCovering" cluster, you will see the calibrationMode and tuyaMotorReversal attributes reading 1, which is classic_nllv. I am quite sure that refreshing the value from the frontend (use the little refresh icon next to the calibration mode drop down) should update the value.
Let me know.

syntesys87, 2023-11-06:
It's very strange, with the stable branch I can't find the calibrationMode attribute but only tuyaMotorReversal... Also, if I refresh the value on frontend, I can see "specific_nllv". These are the screens:
[image]
[image]
[image]

Today I'll try dev branch.

FabianMangold, 2023-11-06:
Hi @syntesys87,
OK, now i understand the issue. You have to use the DEV branch, otherwise the attribute names are not mapped correctly. This is due to the fact, the calibrationMode is a Legrand specific attribute. This was fixed in Zigbee-Herdsman [PR#787](https://github.com/Koenkk/zigbee-herdsman/pull/787), which is not yet available in the stable branch.
Please use the DEV branch until the next stable (next month), so that we are all on the same page and working from the same code base.
Let us know the outcome and have a nice day

philippk75, 2023-11-06:
> Hi @syntesys87, OK, now i understand the issue. You have to use the DEV branch, otherwise the attribute names are not mapped correctly. This is due to the fact, the calibrationMode is a Legrand [...]

to swicth to the DEV version in home assiatnt do i have do deinstall the stable version and install the dev version? do i have to make some backups?

syntesys87, 2023-11-06:
Hello @philippk75, standing to the guide, install ZigBee2Mqtt edge from repositories if you have HAOS installation, copy the addon configuration, stop the stable addon and run edge addon.

syntesys87, 2023-11-06:
@FabianMangold resolved for one of 6 covers! ID 7 is now working well, position works!

[image]

Is there a strange behavior on model of the newly added covers: they are recognized as Legrand instead BTicino... I'll send a new database.db in a couple of hours, waiting for values to populate.

[image]

FYI: ID 7 had an old firmware that I updated some minutes before I opened this issue, the other 5 had latest firmware installed before I configured all Zigbee2Mqtt instance.

FabianMangold, 2023-11-06:
Hi
I‘ve submitted a patch for that yesterday evening. Not sure if your Edge plugin is already up to date. Can you check if you are running the zigbee-herdsman-converters 15.18.0 ?
All devices with HW Version 13 should be reported as BTicino devices now.

syntesys87, 2023-11-06:
[image]

philippk75, 2023-11-06:
> @FabianMangold installed edge addon, the problem is resolved for one of 6 covers! ID 7 is now working well, including led change and position!
> 
> [image]
> 
> Is there a strange behavior on newly added covers: they are recognized as Legrand instead BTicino... I'll send a new database.db in a couple of hours, waiting for values to populate.
> 
> [image]
> 
> FYI: ID 7 had an old firmware that I updated some minutes before I opened this issue, the other 5 had latest firmware installed before I configured all Zigbee2Mqtt instance.

hello, 

for me now it's OK

[image]

the tilt calibration now is OK or do we have to wait for further developments?

I notice that also standard rollo shutter without venetians have the tilt option which makes less sense? o can i set it as rolloshutter without tilt and this options doesn't appear?

Best regards

Philipp

dadino, 2023-11-06:
I'm still on the stable channel, as I just read this thread. This morning I tried to calibrate with "temporal" on my BTicino K4027C, I followed the steps by @FabianMangold and left the house while the switch was still in the orange LED step (calibrating). I'm checking it now and the "calibration_mode" field is null, with no position reported (well, it reports 50, which is not correct).

```
{
    "last_seen": "2023-11-06T09:06:47.442Z",
    "led_in_dark": "ON",
    "linkquality": 0,
    "position": 50,
    "state": "OPEN",
    "tilt": 100,
    "update": {
        "installed_version": 3556351,
        "latest_version": 3556351,
        "state": "idle"
    },
    "action": null,
    "calibration_mode": null,
    "update_available": null
}
```

I guess I'll have to wait for the fixes to get to the stable channel and try again, when the switch is correctly recognized as a BTicino cover switch.

syntesys87, 2023-11-06:
[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/13265061/database.db.zip)

@FabianMangold updated database.db with all the other 5 covers recognized as legrand instead bticino.

FabianMangold, 2023-11-06:
Good evening @philippk75,
Glad to see that the devices are reported correctly again.
I don't understand your question. There is no further development regarding tilt, as already mentioned previously, all Z2M does is tell the switch how to behave (which mode it should run in), and that's basically it.
Is there anything else you are missing, or is it all OK for you now ?

Regarding the "tilt" angle, you've mentioned this before. I am trying to figure out how to do this best, because we are all affected by this "issue". That's just the way Z2M works, and any improvement on that front is a "nice to have". Would you mind opening another "issue" for this, and pin me in there ? I would like to avoid mixing up topics in issues, it's getting complicated to adress everyone's issue in a single thread.
Thanks a lot

FabianMangold, 2023-11-06:
Hi @syntesys87,
Thanks again for provide the database.db. Looking at it closely, it looks like you have other shutter switch having hardware ID 9. I'll submit a fix shortly, to have this corrected as well.
I am not sure that using the Hardware ID is the way to go really, but only "Legrand" could tell us how to properly differentiate BTicino switches from Legrand switches. It's the only difference i can spot really :/

Please always use the DEV branch when experimenting with new features, otherwise we may spend a lot of time tracking down issues that may already have been fixed. You will indeed need the DEV branch to have the attributes reported correctly (i.e. using the correct name). In our case, calibrationMode is equivalent to tuyaMotorReversal.

Please switch to DEV and let me know the outcome.
In case you are eager to test it, you can probably patch your local legrand.ts (or js) and replace the current fingerprint line in the 067776A device definition with the following:

```
        whiteLabel: [
            {
                model: 'K4027C/L4027C/N4027C/NT4027C', vendor: 'BTicino', description: 'Shutter SW with level control',
                fingerprint: [{hardwareVersion: 9}, {hardwareVersion: 13}],
            },
        ],
```
Happy testing!

FabianMangold, 2023-11-06:
Hi @dadino,

From your status, the calibrationMode hasn't been set correctly (it's reporting NULL).
Could you please switch to the DEV branch and give it a try ? This is likely due to the fact the attribute names do not get mapped correctly, which has been fixed (or rather improved) in the DEV branch a couple of days ago.
Let me know how that goes.

philippk75, 2023-11-07:
> Good evening @philippk75, Glad to see that the devices are reported correctly again. I don't understand your question. There is no further development regarding tilt, as already mentioned previously, all Z2M does is tell the switch how to behave (which mode it should run in), and that's basically it. Is there anything else you are missing, or is it all OK for you now ?
> 
> Regarding the "tilt" angle being displayed in all modes, this is somewhat annoying indeed, I agree. You've mentioned this before, and I am trying to figure out how to do this best, because we are all affected by this "issue". That's just the way Z2M works, and any improvement on that front is a "nice to have". Would you mind opening another "issue" for this, and pin me in there?
> 
> I would like to avoid mixing up topics in issues, it's getting complicated to adress everyone's issues in a single thread. Thanks a lot

Hello,

thanks for your effort. maybe you can control the hiding of the options for the inclination of the blinds by taking into account the type in the device itself? Hide for roller shutters, show for Venetian blinds?

Thanks

[image]

syntesys87, 2023-11-07:
> Hi @syntesys87, Thanks again for provide the database.db. Looking at it closely, it looks like you have other shutter switch having hardware ID 9. I'll submit a fix shortly, to have this corrected as well. I am not sure that using the Hardware ID is the way to go really, but only "Legrand" could tell us how to properly differentiate BTicino switches from Legrand switches. It's the only difference i can spot really :/
> 
> Please always use the DEV branch when experimenting with new features, otherwise we may spend a lot of time tracking down issues that may already have been fixed. You will indeed need the DEV branch to have the attributes reported correctly (i.e. using the correct name). In our case, calibrationMode is equivalent to tuyaMotorReversal.
> 
> Please switch to DEV and let me know the outcome. In case you are eager to test it, you can probably patch your local legrand.ts (or js) and replace the current fingerprint line in the 067776A device definition with the following:
> 
> ```
>         whiteLabel: [
>             {
>                 model: 'K4027C/L4027C/N4027C/NT4027C', vendor: 'BTicino', description: 'Shutter SW with level control',
>                 fingerprint: [{hardwareVersion: 9}, {hardwareVersion: 13}],
>             },
>         ],
> ```
> 
> Happy testing!
> 
> Edit: [PR#6423](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6423) for the fix

Thank you for the fix! I'll wait the merge with next release, no problem.

FabianMangold, 2023-11-07:
HI @syntesys87,
The fix has been merged, let me know if that fixes the remaining issues!
Thanks a lot

syntesys87, 2023-11-08:
Hi @FabianMangold, all fixed, now I see the new hardware as BTicino instead Legrand!
Last question, do you know if there are open bugs regarding the led that remains green blinking after pairing?

This bug could be closed.

philippk75, 2023-11-08:
@FabianMangold i opendes this post: https://github.com/Koenkk/zigbee2mqtt/issues/19621

syntesys87, 2023-11-08:
@FabianMangold yes, this issue could be closed! 
I have not reproduced led problem on my hardware 13 switch, but in all the other hardware 9 switches.
Thank you for the support, you are great!

FabianMangold, 2023-11-08:
My pleasure @syntesys87, happy to help if i can.
It's odd, if seems that the issue is somehow linked to the hardware ID. Users in the thread i mentioned above report exactly this issue. It might be worth reaching out to Legrand via their forum and ask for some guidance there.
Can you please mark this issue as solved, since you are the one that has opened it ?
Cheers

syntesys87, 2023-11-08:
@FabianMangold yes, it's very strange... As it's strange that there are multiple hardware IDs for the same component... There isn't any parameter or code that is useful to identify che switch. Here https://github.com/Koenkk/zigbee2mqtt/issues/16948#issuecomment-1680592070 you can find the procedure to workaround the blinking green led problem.

Thank you for your effort!

closed 2023-11-08 by syntesys87

r0bb10, 2023-11-14:
I finally managed to get my 7x K4027C working (firmware 36) using the method posted [here](https://github.com/Koenkk/zigbee2mqtt/issues/16948#issuecomment-1680592070) and get rid of the stupid K4510C Gateway.

It was really a pain to fix the green flashing light because in this mode it's not fully paired and the calibration procedure cannot be started.. the first 2 K4027C paired fine but the others was fail&repeat a lot.. eventually fixed and got the purple light on all of them (fully paired) and then the calibration procedure can be started (up+down button for 5 sec).

r0bb10, 2023-11-19:
> > I finally managed to get my 7x K4027C working (firmware 36) using the method posted [here](https://github.com/Koenkk/zigbee2mqtt/issues/16948#issuecomment-1680592070) and get rid of the stupid K4510C Gateway.
> 
> @r0bb10 How did you update the bTicino devices? I keep getting the error `Device didn't respond to OTA request` I was able to update a couple of K4027C and those I was able to fully pair, but I can't fully pair the other with older firmware.

i have a K4510C gateway and use it only to update the devices, never managed to ota update from z2m or zha.

andreapx, 2023-11-19:
> i have a K4510C gateway and use it only to update the devices, never managed to ota update from z2m or zha.

You've never tried with Z2M or you've tried and failed?

r0bb10, 2023-11-19:
> > i have a K4510C gateway and use it only to update the devices, never managed to ota update from z2m or zha.
> 
> You've never tried with Z2M or you've tried and failed?

never tried.

andreapx, 2023-11-19:
To Update the devices with the gateway you have to pair them all to it and then pair them back to Z2M?

r0bb10, 2023-11-19:
> To update the devices with the gateway do you have to pair them all to it and then pair them back to Z2M?

exactly, i had everything paired with the bticino gateway, some with new firmware some with older one, i left them all connected and they slowly updated to the latest available and then moved them to z2m (and then to zha).

andreapx, 2023-11-19:
> exactly, i had everything paired with the bticino gateway, some with new firmware some with older one, i left them all connected and they slowly updated to the latest available and then moved them to z2m (and then to zha).

Damn! I have about 70 devices, it's unthinkable!
Thanks for the feedback!

[18 more replies]