Koenkk/zigbee2mqtt #19510: Stopped working suddenly

tcheronneau, 2023-11-01:
### What happened?

I'm using zigbee2mqtt for a while now and most of the time it works super well. 
But today all of the sudden it stopped working properly with a lot of : 
`Zigbee2MQTT:error 2023-11-01 17:09:51: Publish 'set' 'state' to '0x001788010c6556c3' failed: 'Error: Command 0x001788010c6556c3/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
Zigbee2MQTT:info  2023-11-01 17:09:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish
 'set' 'state' to '0x001788010c6556c3' failed: 'Error: Command 0x001788010c6556c3/11 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'undefined' (25))'","meta":{"friendly_name":"0x001788010c6556c3"},"type":"zigbee_publish_error"}'`

From time to time it's still working but it's not viable. 
I've changed nothing in my setup recently. 

This kind of problem already happened from time to time but normally I reset the adapter by flashing the firmware again but here it's not working. 
I got a big network (60 devices) but even those close to the adapters are not working. 
I was using the 20230507 firmware but I saw an issue saying to try the 20230923 so I did with no luck. 

### What did you expect to happen?

My lights to turn on directly. 

### How to reproduce it (minimal and precise)

Trying to stop / start lights. 

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20230923

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

info  2023-11-01 17:09:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x001788010c6556c3' failed: 'Error: Command 0x001788010c6556c3/11 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'undefined' (25))'","meta":{"friendly_name":"0x001788010c6556c3"},"type":"zigbee_publish_error"}'
error 2023-11-01 17:09:47: Publish 'set' 'state' to '0x001788010c6556c3' failed: 'Error: Command 0x001788010c6556c3/11 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
info  2023-11-01 17:09:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x001788010c6556c3' failed: 'Error: Command 0x001788010c6556c3/11 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'undefined' (25))'","meta":{"friendly_name":"0x001788010c6556c3"},"type":"zigbee_publish_error"}'

mshamash, 2023-11-02:
I noticed the same today on my Home Assistant Yellow, using the built-in zigbee system. Suddenly stopped working for all devices, and can't reach them or do LQI queries to make a map. Restarting the Yellow didn't seem to fix anything, and everything is up to date – haven't made any changes for a few weeks...

awudarowicz, 2023-11-02:
same here, almost everything stopped working  😱

diegoweb, 2023-11-02:
[image]
I set z2m to debug mode and this is what I get when I try to refresh the state of any switch.

merlinpimpim, 2023-12-13:
Hi,
It's strange, I now have the same problem (with the Skyconnect coordinator) and more particularly certain devices (Nodon module, Ledvance socket...).
And the strangest thing is that the problems are minimized when I run Z2M in debug mode (but in that case, watch out for the disk space consumed!).
In the logs themselves, nothing extraordinary though (cf error message already posted above).
I don't understand what exactly is going on!

Z2M=1.34.0

(I haven't yet tried the suggestion of booting HA without the USB key, then plugging it in and starting Z2M manually. Even if it worked, it wouldn't be very normal !).

Koenkk, 2023-12-13:
@tcheronneau try the [`20221226`](https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin) firmware, both 20230507 and 20230923 might be unstable.

tcheronneau, 2023-12-13:
Hey @Koenkk I also tried this one I think without success. In the end it came back after a day (without any action) so maybe it was linked to something in the environment (some interference or something).

merlinpimpim, 2023-12-13:
Hot plugging the Skyconnect and then manually turning on ZM afterwards doesn't change the problem. 😒

For the record, here's what I see regularly:

`Zigbee2MQTT:error 2023-13-12 20:59:09: Failed to configure 'ALLÉE - Portail - Module Nodon', attempt 1 (Error: Read 0x9035eafffed8d440/1 genOnOff(["nodonTransitionTime"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:22)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/nodon.ts:55:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))`

lieblinger, 2024-03-16:
Facing the same issue in one of my Z2M installations. Could you ever fix this issue?

My adapter is ZBDongle-P and already tried different firmware versions.

merlinpimpim, 2024-03-17:
According to the documentation, this command only works with Texas Instrument-based adapters (CC2652/CC1352).

closed 2024-10-15 by github-actions[bot]