Koenkk/zigbee2mqtt #19513: Problem with _TZE200_fjjbhx9d handling brightness adjustment from Home Assistant

mick5423, 2023-11-02:
### What happened?

Brightness changes initiated from Home Assistant are not handled as expected, also blocking future brightness changes directly from Zigbee2MQTT until the light switch is power cycled.

From a fresh power cycle, Zigbee2MQTT and dimming from the physical switch play nicely until any attempt is made from Home Assistant which also sends a state=ON alongside brightness requests.

### What did you expect to happen?

Light should change brightness to set value.

### How to reproduce it (minimal and precise)

Power on switch
Adjusting brightness from Zigbee2MQTT Exposes page or physical device works correctly across multiple changes at this point.
Adjusting brightness from Home Assistant doesn't result in any change, but do see a little flicker after time delay typical of the soft-on transition time.
Any future attempts at changing brightness from Zigbee2MQTT are also ignored until power is cycled.

My interpretation:
Can see Home Assistant sends state=ON in parallel with brightness adjustment. 

It seems as though despite the {skip: tuya.skip.stateOnAndBrightnessPresent} modifier in the datapoint definitions, the state is still being sent and handled, but the brightness is ignored on that request, as are subsequent brightness changes from Zigbee2MQTT. The light state needs to be cycled off/on before brightness changes are handled again.

Is it perhaps something to do with skip.stateOnAndBrightnessPresent not being invoked due to Home Assistant invoking /l1/set "brightness" as opposed to how Zigbee2MQTT seems to invoke /set "brightness_l1"? Sorry I don't really understand how the code functions!

From lib/tuya.ts:
export const skip = {
    // Prevent state from being published when already ON and brightness is also published.
    // This prevents 100% -> X% brightness jumps when the switch is already on
    // https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1263592783
    stateOnAndBrightnessPresent: (meta: Tz.Meta) => meta.message.hasOwnProperty('brightness') && meta.state.state === 'ON',
};

### Zigbee2MQTT version

1.33.2-1

### Adapter firmware version

7.3.1.0 build 0

### Adapter

Home Assistant SkyConnect (running RCP firmware)

### Debug log

Sequential logs below with comments/interpretation:
Correctly handled brightness change after power on, triggered from Zigbee2MQTT. Receive 2x responses against DP2.
```
Debug 2023-11-02 12:04:08Received MQTT message on 'zigbee2mqtt/MBR Ceiling Lights/set' with data '{"brightness_l1":59}'
Debug 2023-11-02 12:04:08Publishing 'set' 'brightness' to 'MBR Ceiling Lights'
Debug 2023-11-02 12:04:08Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":2560}' from endpoint 1 with groupID 0
Debug 2023-11-02 12:04:11Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,230],"type":"Buffer"},"datatype":2,"dp":2}],"seq":2816}' from endpoint 1 with groupID 0
```

Attempted brightness change from Home Assistant. Single response from DP1 instead of 2x responses against DP2 as above. Also get a little flicker on the lightbulb after a delay typical of the soft-on transition time.
```
Debug 2023-11-02 12:04:17Received MQTT message on 'zigbee2mqtt/MBR Ceiling Lights/l1/set' with data '{"state":"ON","brightness":190}'
Debug 2023-11-02 12:04:17Publishing 'set' 'brightness' to 'MBR Ceiling Lights'
Debug 2023-11-02 12:04:17Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":3072}' from endpoint 1 with groupID 0
```

Attempted brightness change from Zigbee2MQTT but no actual change in brightness. Receive 1x response from DP2 with a [0-1000] brightness of 177 and a 3 in the 3rd value???
```
Debug 2023-11-02 12:04:33Received MQTT message on 'zigbee2mqtt/MBR Ceiling Lights/set' with data '{"brightness_l1":240}'
Debug 2023-11-02 12:04:33Publishing 'set' 'brightness' to 'MBR Ceiling Lights'
Debug 2023-11-02 12:04:33Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,177],"type":"Buffer"},"datatype":2,"dp":2}],"seq":3328}' from endpoint 1 with groupID 0
```

Power cycle
```
Debug 2023-11-02 12:04:45Received MQTT message on 'zigbee2mqtt/MBR Ceiling Lights/set' with data '{"state_l1":"OFF"}'
Debug 2023-11-02 12:04:45Publishing 'set' 'state' to 'MBR Ceiling Lights'
Debug 2023-11-02 12:04:45Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":3584}' from endpoint 1 with groupID 0
Debug 2023-11-02 12:04:45Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":3840}' from endpoint 1 with groupID 0
Debug 2023-11-02 12:04:50Received MQTT message on 'zigbee2mqtt/MBR Ceiling Lights/set' with data '{"state_l1":"ON"}'
Debug 2023-11-02 12:04:50Publishing 'set' 'state' to 'MBR Ceiling Lights'
Debug 2023-11-02 12:04:50Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4096}' from endpoint 1 with groupID 0
Debug 2023-11-02 12:04:50Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4352}' from endpoint 1 with groupID 0
```

Set brightness via Zigbee2MQTT. Note the returned [0-1000] brightness of 232 - this was the value previously set in first step, and was the brightness powered back up as
```
Debug 2023-11-02 12:04:57Received MQTT message on 'zigbee2mqtt/MBR Ceiling Lights/set' with data '{"brightness_l1":254}'
Debug 2023-11-02 12:04:57Publishing 'set' 'brightness' to 'MBR Ceiling Lights'
Debug 2023-11-02 12:04:57Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":4608}' from endpoint 1 with groupID 0
Debug 2023-11-02 12:05:02Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":4864}' from endpoint 1 with groupID 0
```

Power off / demo complete
```
Debug 2023-11-02 12:05:09Received MQTT message on 'zigbee2mqtt/MBR Ceiling Lights/set' with data '{"state_l1":"OFF"}'
Debug 2023-11-02 12:05:09Publishing 'set' 'state' to 'MBR Ceiling Lights'
Debug 2023-11-02 12:05:09Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":5120}' from endpoint 1 with groupID 0
Debug 2023-11-02 12:05:09Received Zigbee message from 'MBR Ceiling Lights', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":5376}' from endpoint 1 with groupID 0
```

mick5423, 2023-11-02:
Additional testing: looks like a bigger bug with this device. Any repeat state=ON message while already on puts it into this condition whereby brightness can't be adjusted until that switch is power cycled. 

So might potentially need another "skip" variant to block any state=ON messages if already ON?

As a workaround of the built-in brightness adjustment in Home Assistant which always sends an ON signal, I found I can directly publish a brightness payload to MQTT through a Template Light. Not ideal, but gets the device to behave if options for skipping are limited.

```
light:
  - platform: template
    lights:
      mbr_l1:
        friendly_name: "Ceiling Fan Light"
        level_template: "{{ state_attr('light.mbr_ceiling_lights_l1', 'brightness') }}"
        value_template: "{{ states('light.mbr_ceiling_lights_l1') }}"
        turn_on:
          - if:
              condition: template
              value_template: "{{ is_state('light.mbr_ceiling_lights_l1', 'off') }}"
            then:
              service: light.turn_on
              target: 
                entity_id: light.mbr_ceiling_lights_l1
        turn_off:
          service: light.turn_off
          target: 
            entity_id: light.mbr_ceiling_lights_l1
        set_level:
          service: mqtt.publish
          data:
            qos: 0
            retain: false
            topic: zigbee2mqtt/MBR Ceiling Lights/l1/set
            payload: "{\"brightness\":{{ brightness }}}"
        supports_transition_template: "{{ true }}"
      mbr_l2:
        friendly_name: "Wall Lights"
        level_template: "{{ state_attr('light.mbr_ceiling_lights_l2', 'brightness') }}"
        value_template: "{{ states('light.mbr_ceiling_lights_l2') }}"
        turn_on:
          - if:
              condition: template
              value_template: "{{ is_state('light.mbr_ceiling_lights_l2', 'off') }}"
            then:
              service: light.turn_on
              target: 
                entity_id: light.mbr_ceiling_lights_l2
        turn_off:
          service: light.turn_off
          target: 
            entity_id: light.mbr_ceiling_lights_l2
        set_level:
          service: mqtt.publish
          data:
            qos: 0
            retain: false
            topic: zigbee2mqtt/MBR Ceiling Lights/l2/set
            payload: "{\"brightness\":{{ brightness }}}"
        supports_transition_template: "{{ true }}"
```

thebigspin, 2023-12-19:
I have the same issue (caused much headscratching!) - @mick5423 your provided workaround works great for me, thank you!

CobraDunn, 2024-01-13:
> Additional testing: looks like a bigger bug with this device. Any repeat state=ON message while already on puts it into this condition whereby brightness can't be adjusted until that switch is power [...]

Thank you very much for this workaround. I've been struggling with this for months. There aren't many options for single-gang double and triple dimmers, and It's looking like a solution isn't on any update roadmaps I've seen.

closed 2024-08-11 by github-actions[bot]

eatoff8, 2024-09-14:
> Additional testing: looks like a bigger bug with this device. Any repeat state=ON message while already on puts it into this condition whereby brightness can't be adjusted until that switch is power [...]

So this is still a thing, and this is still a working solution. How do we get this sorted? Raise a new issue?