Koenkk/zigbee2mqtt #19516: Update Fail for NodON SIN 4-2-20

Pocky0013, 2023-11-02:
### What happened?

Impossible to upgrade to OTA that Z2M offers. I simply receive a timeOUT error, with no specific explanation. 
Look like the same for #18724 (the error does not seem to have resolved since August)

### What did you expect to happen?

OTA uprade good

### Zigbee2MQTT version

 1.33.2

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

Pocky0013, 2023-11-06:
[image]
Now with HA 2023.11, I have a notification that an update is in progress, with a spinning wheel. While I never run this. And in Z2M no OTA bet is in progress.

moshiko2312, 2024-02-25:
hi me to
what need to do?
on original Nodon app no have update

[image]

dguihal, 2024-08-07:
As of today, I have been able to run an update through z2m on both SIN-4-RS-20 and SIN-4-2-20

I would suggest you to retry because probably issue is fixed now

moshiko2312, 2024-08-07:
I will try you ar on last ver?

moshiko2312, 2024-08-07:
Not working  for me 
￼

nanostra, 2024-10-17:
ONE year later...and a lot of crappy or incomplete firmware updates!!! Buy Nodon?

[image]

[image]

moshiko2312, 2024-10-17:
Me to i don't know what to do ...

dguihal, 2024-10-17:
[image]

Did you try to do a bluetooth (mobile app) update ?

moshiko2312, 2024-10-17:
> image.png (view on web)
>
> Did you try to do a bluetooth (mobile app) update ?

Yes but no update all on release

Gw3n4, 2025-01-03:
Hi,
I meet the same issue. I get the messsage 'no new update available' when trying to update my SIN-4-20 device's firmware.
What I see on the mqtt topic: 
  "update": {
    "installed_version": 65794,
    "latest_version": 66816,
    "state": "available"
  },
  "update_available": null
}
Installed version is not the latest, but no update available. Weird.

closed 2025-05-15 by github-actions[bot]