Koenkk/zigbee2mqtt #19523: IKEA VINDSTYRKA, no data after pairing. 1.33.2-dev

sanyafifa, 2023-11-02:
### What happened?

IKEA VINDSTYRKA, after the interview I don't get any data except LQI
[image]

### Zigbee2MQTT version

1.33.2-dev

### Adapter firmware version

20230507

### Adapter

 zStack3x0

### Debug log

ikea  = `0x287681fffe6f2bb1`
debug:
[z2m_ikea_debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/13241560/z2m_ikea_debug.txt)

sanyafifa, 2023-12-01:
after updating to 1.34.0-dev - no changes. no data has been received! 
help me :)

sanyafifa, 2024-01-07:
`Error 2024-01-07 20:05:55Failed to configure '0x287681fffe6f2bb1', attempt 3 (Error: ConfigureReporting 0x287681fffe6f2bb1/1 pm25Measurement([{"attribute":"measuredValueIkea","minimumReportInterval":60,"maximumReportInterval":120,"reportableChange":2}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4476,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:674:22) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/ikea.ts:1160:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:167:17))`

z2m ver: 1.35.1

zikson, 2024-01-28:
Same issue happening for me. Other device paired long time ago working fine.

sanyafifa, 2024-01-28:
> Reverted to main and now pairing is working fine.

Excuse me, can I tell you more about how you fixed it?

sjorge, 2024-01-28:
This should hopefully be fixed in the february release (https://github.com/Koenkk/zigbee-herdsman-converters/pull/6972)

hexonMD, 2024-03-09:
> > Reverted to main and now pairing is working fine.
> 
> Excuse me, can I tell you more about how you fixed it?

Could use this info as well.

closed 2024-10-06 by github-actions[bot]