Koenkk/zigbee2mqtt #19524: [Feature request]: Starting without MQTT Server

TRuHa83, 2023-11-02:
### Is your feature request related to a problem? Please describe

Due to technical limitations, I maintain two servers in my office. One of these servers, in addition to hosting multiple services, houses the MQTT Server, while on another PC, I run Zigbee2MQTT. Up to now, both systems have been functioning correctly as expected. However, I recently encountered an unexpected issue: the main server hosting the MQTT Server experienced a failure, resulting in the inability to start the Zigbee2MQTT service, as the latter depends on the former to operate properly.

This limitation caught me by surprise, as it is ironic to attempt to start Zigbee2MQTT when it cannot find the MQTT server. The irony lies in the fact that despite my intention to turn on the lights controlled by Zigbee2MQTT, I was prevented from doing so due to the service not starting.

### Describe the solution you'd like

I believe the server should start even in the absence of the MQTT Server. It could even be an option to configure from the frontend.

### Describe alternatives you've considered

Let go of the stories and don't rely so much on technology.

### Additional context

I might be rambling on the topic. I've never had issues at home, but due to the problem I encountered at the office, where we don't have switches, we rely solely on Zigbee2MQTT to turn on the lights. If at least the Frontend could start, we could turn on the lights temporarily until we resolve the connection issue. Nevertheless, I can always wire an emergency on/off switch. It was just an idea. Best regards.

redpixy, 2023-11-03:
TBH, without MQTT available non of your switches will even work. As MQTT is the transport layer between your devices and the application for sending out the commands. Without this transport layer nothing is going to work.

In regards to your issue, you could always use device bind to your lights to a switch. See - https://www.zigbee2mqtt.io/guide/usage/binding.html

redpixy, 2023-11-03:
about the MQTT server going down on the server. Was that server running the MQTT experience high load? As I previously experience issues when hosting my MQTT on my production server where the traffic load pretty high and the entire Z2M pretty much became unresponsive

emandtf, 2023-11-04:
> In regards to your issue, you could always use device bind to your lights to a switch.

I have 3 types of different dimmers, 4 of buttons, 2 types of plugs/switches, 2 types of motion, 1 bulb and 1 knob type (for a total of 43 zigbee devices) but no anyone of them allowing Binding to work...
I.e: I tried to bind a button with/to a dimmer, but no any Action is selectable other than "Gen Basic", "OTA" or "Identify". No any "OnOff", "LControl", etc... 
Same trying to bind knobs to dimmers, switches to bulbs, motion sensors to dimmers or bulbs, etc...
Binding maybe works for less than 1% of supported devices, so it cannot be a solution.

Morover if there are some "smartness procedure" using HA Automations (i.e: turn on corridor light only when lux sensor detects dark and motions sensor detect movement) it goes in contrast with Binding (if someone is lucky enough to have compatible devices for binding), so again this could be the way.

closed 2023-11-11 by TRuHa83