Koenkk/zigbee2mqtt #19527: Xiaomi WXKG15LM - Both buttons trigger all zigbee Plugs

MartijnDisse, 2023-11-02:
### What happened?

I added the Xiaomi WXKG15LM to Z2M. After pairing I can use the switch. However, without having any automation or script running in HA related to that new switch, ALL zigbee plugs connect toggle their status. This happens whatever button I press (left or right).

### What did you expect to happen?

I expect other Plug not to respond.

### How to reproduce it (minimal and precise)

Have one or more Plugs (any brand) paired.
Pair Xiaomi WXKG15LM.
Press any button on Xiaomi WXKG15LM

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20210708

### Adapter

Sonoff USB 3.0 Dongle-P Zigbee

### Debug log

19:52:34Received Zigbee message from 'Test', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:34Received Zigbee message from 'Test', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:34Received Zigbee message from 'Test', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:34Received Zigbee message from 'Test', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:34Received Zigbee message from 'Test', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:34Received Zigbee message from 'Test', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:34Received Zigbee message from 'Test', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:34Received Zigbee message from 'Test', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:34Received Zigbee message from 'Test', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:34Received Zigbee message from 'Blitzwolf Plug - I', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":69}' from endpoint 1 with groupID 0
Debug 2023-11-02 19:52:35Received Zigbee message from 'Blitzwolf Plug - I', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":69}' from endpoint 1 with groupID 0

remkolems, 2024-01-04:
This happen to me too just yet. 

All my Xiaomi smart sockets went offline and on again. Empty string "" returned after pressing either the left or the right button or both. It even happens when not paired with zigbee2mqtt (force removed) and even after I restarted the HA zigbee2mqqt docker and broker.

Zigbee2MQTT [v1.35.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.0)

NicolasQuenon, 2024-01-13:
Hi,
I have exactly the same problem. I try to unlink the swith from my HA with zigbee2mqtt box and I continue to have the same problem : at each click all zigbee devices are triggered.
For exemple, light inverse their states and swith or plug too.
Do your have a solution or an idea to help us?
Thanks

Wersal-kp, 2024-04-19:
Hi,
I have the same problem. Pressing the button would trigger most other zigbee devices in the network.
Any idea how to solve this?

4s1, 2024-04-21:
I ran into the same issue today, involuntarily.
In my case this happens, when I joined the switch with "Permit Join (All)". 
So I removed the switch from network and joined it again with "Permit join (Coordinator)".

While doing the latter, it won't toggle any of my plugs.

It looks like the switch binds to all actors supporting On/Off event when joining is permitted trough all devices?

Can anyone confirm my guess?

Wersal-kp, 2024-04-23:
It did not work this way for me. Permit join (Coordinator) also caused toggling some devices ;/

Wersal-kp, 2024-04-23:
Update: I must say that it finally did work the way you said. I turned on permit join (Coordinator) and switch joined the network correctly without toggling devices. But it took several attempts, either Zigbee2MQTT GUI is buggy and did not use coordinator mode each time or maybe there is other reason behind it.

closed 2024-11-20 by github-actions[bot]