Koenkk/zigbee2mqtt #19530: Can't turn off ZB-RGBCW after setting transition

stonegray, 2023-11-02:
### What happened?

Can't turn off [eWeLight/Lonsonho ZB-RGBCW](https://www.zigbee2mqtt.io/devices/ZB-RGBCW.html#lonsonho-zb-rgbcw) after setting transition.

- Neither the on/off switch or setting brightness zero will turn off the bulb.
- The bulb will turn off if set to color mode and brightness zero. 
- Transitions are functional
- Not resolved by setting transition back to zero; needs to re-pair device to resolve.

Related:  @teewhey in #13632

### What did you expect to happen?

Bulb should work normally after transition is set.

### How to reproduce it (minimal and precise)

- Pair device
- Set transition to 1s
- Turn on bulb and then turn off bulb

### Zigbee2MQTT version

1.33.2-dev commit: 1e90559

### Adapter firmware version

zStack3x0 20220219

### Adapter

Sonoff 3.0

stonegray, 2023-11-02:
Found a temporary solution by adding an external converter:

https://github.com/Koenkk/zigbee2mqtt/issues/11109#issuecomment-1432256179

zagi988, 2023-11-05:
It was working before the last update. Something changed.

GeorgyErm, 2023-11-06:
> Found a temporary solution by adding an external converter:
> 
> [#11109 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/11109#issuecomment-1432256179)

Can you please specify what you have changed for make it working? I tried to use custom converter from the #11109 comment, but it didn't help to turn bulbs off. Actually nothing changed with that custom converter

zagi988, 2023-11-06:
> Found a temporary solution by adding an external converter:
> 
> [#11109 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/11109#issuecomment-1432256179)

Works for me too!

stonegray, 2023-11-06:
> Can you please specify what you have changed for make it working? 

You need to manually specify it in the external converters tab in z2m settings or by manually editing your config file; it takes a path relative to the same directory your `configuration.yaml` is in.

GeorgyErm, 2023-11-06:
> > Can you please specify what you have changed for make it working?
> 
> You need to manually specify it in the external converters tab in z2m settings or by manually editing your config file; it takes a path relative to the same directory your `configuration.yaml` is in.
> 
> If it's in the same directory as your config just add this to your `configuration.yaml`:
> 
> ```yaml
> external_converters:
>   - lonsonho.js
> ```

I've done it. Actually I have several other external converters which are working, so it seems like the problem not in converters config. What bulbs firmware have you got? I received my bulbs last week, so probably they have another firmware with some new bugs. 
[image]

zagi988, 2023-11-09:
I've got the same firmware and even tho external converter was working, it's not anymore. With transition being set to 0 or anything else. Can't turn it off over Z2M.

KyleStilkey, 2024-01-16:
Same issue here with the latest push. My ZB-RGBCW from lonsonho refuse to turn off, using the work around linked above fixed this for me, placed the js in the zibee2mqtt folder, added the file name under the settings external converters, restarted and the bulb started reporting the correct off states instead of unknown.

Was working perfectly fine before the latest stable push for me.

closed 2024-08-15 by github-actions[bot]