Koenkk/zigbee2mqtt #19531: Progressing OTA update flooding MQTT when last_seen enabled

PaJaSoft, 2023-11-02:
### What happened?

I've paired a new device identified as model: "TS011F" and manufacturer: "_TZ3000_gjnozsaz" to my established ZB network, interviewing was smooth and everything is working as expected onward (only `manuSpecificBosch` and `manuSpecificTuya_3` aren't covered by converters, but it isn't a big deal and isn't necessary).

After verifying that I'm able to control all features of this mains device, tried to check firmware updates. A new update was found (including a new picture of the device), so I triggered OTA update. Update is progressing fine however I see one implementation detail which needs developers attention.

I've enabled `last_seen` feature for all my devices previously (I guess it is just global parameter, can't be set just for a particular device) and it introduces a problem - a new MQTT message is generated *too often* - in my case 2-4 times *per second*!

I understand that it would make sense because the device was really seen 'right now' (even no other reported valued was changed) but in this case I would prefer to take into the account at least configured `Debounce` parameter for such a device.

Better solution would be to generate MQTT msg where only `last_seen` attribute was changed during an OTA update progress with a lower cadence - one per second at max, probably even lower...

### What did you expect to happen?

When only last_seen reported attribute is changed, lower cadence of a new MQTT message - couple times per second is just flooding the MQTT server.

At least configured Debounce value should be taken into the account.

### How to reproduce it (minimal and precise)

Enable last_seen feature in z2m (disabled by-default).
Pair a device which needs an update and is covered by z2m OTA update.
Set Debounce for this device to couple of seconds - optional
Trigger OTA firmware update for this device

You should be able to see multiple MQTT messaged generated every second during the OTA update process.

### Zigbee2MQTT version

1.33.1 (commit #7e63039)

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle E

### Debug log

> Zigbee2MQTT:info  2023-11-02 21:56:19: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:19.757Z","linkquality":240,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:info  2023-11-02 21:56:20: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:20.029Z","linkquality":228,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:info  2023-11-02 21:56:20: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:20.302Z","linkquality":236,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:info  2023-11-02 21:56:20: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:20.579Z","linkquality":228,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:debug 2023-11-02 21:56:20: Received Zigbee message from '0xa4c138955b3c79ca', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":77}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-11-02 21:56:20: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:20.645Z","linkquality":228,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:info  2023-11-02 21:56:20: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:20.854Z","linkquality":228,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:info  2023-11-02 21:56:21: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:21.126Z","linkquality":228,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:info  2023-11-02 21:56:21: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:21.405Z","linkquality":232,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:info  2023-11-02 21:56:21: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:21.671Z","linkquality":228,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:info  2023-11-02 21:56:21: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:21.944Z","linkquality":244,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'
Zigbee2MQTT:info  2023-11-02 21:56:22: MQTT publish: topic 'zigbee2mqtt/0xa4c138955b3c79ca', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-11-02T20:56:22.217Z","linkquality":236,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":77,"latest_version":192,"progress":59.01,"remaining":696,"state":"updating"},"update_available":null,"voltage":239}'

PaJaSoft, 2024-05-01:
/keep

PaJaSoft, 2024-05-01:
Issue is still valid...

Note: I'm not allowed to remove 'stale' label:-(

mundschenk-at, 2024-05-29:
Yes, this is definitely still a problem with `dev`.

closed 2025-01-11 by github-actions[bot]