Koenkk/zigbee2mqtt #19536: Ikea Trådfri RGB Spotlight effects not working

Estrobeda, 2023-11-03:
### What happened?

Brand: Ikea
Device: RGB Spotlight (TRADFRI bulb GU10 CWS 345lm)
Model: LED1923R5
Firmware: 1.0.021 (2019062886LEL08N)

Issue:
The following effects does not work on this device:

    blink
    breathe
    okay
    channel_change
    finish_effect
    stop_effect

According to the logs, the MQTT message is sent and picked up by Zigbee2MQTT and Zigbee2MQTT says it publishes the changes but the actual result is "no change".
Additionally, if you change the effect in the UI and then go to any other tab and then go back, the effect is null (---)

Code of context:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/a60d4f6e3b709160a045e836200910a1ab8660dc/src/devices/ikea.ts#L1015

### Zigbee2MQTT version

 1.33.2-1

### Adapter firmware version

 0x26720700

### Adapter

 ConBee2/RaspBee2

martonbtoth, 2023-12-16:
I'm encountering the same issue on 1.34.0 with a "TRADFRI bulb E27 CWS 806lm" model. Effects work fine on my "TRADFRI bulb E27 CWS opal 600lm" bulbs.

cmprmsd, 2024-02-14:
Same problem here. Same bulb and firmware :(

closed 2024-09-12 by github-actions[bot]

ZeFish, 2024-10-06:
Same here with 12 brand new gu10