Koenkk/zigbee2mqtt #19538: Z2M failure to load after upload to v.1.33.2

PaoloVittori, 2023-11-03:
### What happened?

Updating from v1.33.1-1 to v1.33.2-1 causes error and failure to load
I Know this is related to the external convertes.
I use external converters for many devices to add features that the default converters don't have but which the device still supports.

### What did you expect to happen?

Z2M to load after version update.

### How to reproduce it (minimal and precise)

Install newer version.

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

20230507

### Adapter

 SONOFF ZB Dongle-P Zigbee3.0 USB Dongle

### Debug log

[12:20:42] INFO: Preparing to start...
[12:20:43] INFO: Socat not enabled
[12:20:45] INFO: Starting Zigbee2MQTT...

/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

PaoloVittori, 2023-12-01:
After having fixed the external converters to solve the error, with the new update (version 1.34.0.1), I have to work on the external converters once again.
Zigbee2mqtt proves to be a terrible and inadequate system, Deconz is much better.

Koenkk, 2023-12-02:
External converters are not meant for long term usage, it's meant for developing. Once the converter is working, please contribute it back to zigbee-herdsman-converters.