Koenkk/zigbee2mqtt #19539: ZM25RX-08/30 directions are reversed after 1.33.2 upgrade

burnsie-la, 2023-11-03:
### What happened?

Since upgrading to 1.33.2 my Zemismart tubular blind closes when I send the OPEN command and vice versa.  Changing the motor direction or using the Invert Cover options within Zigbee2MQTT had no effect.  Setting the position of the cover using the 0-100 value works as expected.

Device type: EndDevice
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_7eue9vhc
Description: Tubular motor

It seems to have started happening after the introduction of this PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6341/files

### What did you expect to happen?

The Close command should close the cover, the open command should open the cover.

### How to reproduce it (minimal and precise)

Pair a _TZE200_7eue9vhc with Z2M.
Press Close in the Exposes tab of the device in the UI
The blind opens, instead of closes.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Home Assistant SkyConnect

### Debug log

Topic: zigbee2mqtt/Top Floor Hallway Blind
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13249392/log.txt)

burnsie-la, 2023-11-03:
This also seems to affect the _TZE200_bv1jcqqu version of the Zemismart ZM25RX-08/30

Koenkk, 2023-11-03:
@donbobka  could you check this?

donbobka, 2023-11-04:
@burnsie-la I've set state according to physical buttons+motor direction option. Can you send the photo of your blinds? I think maybe it depends on direction of blinds rolled onto tube 🤔 

P.S.: There's expose option to invert control that can reverse it.

Koenkk, 2023-11-04:
@donbobka could you make a PR for that?

burnsie-la, 2023-11-04:
> @burnsie-la I've set state according to physical buttons (buttons at left side)+motor direction option=normal. Can you send the photo of your blinds? It's strange that internal invert option doesn't work
> 
> Open/close actually depends on multiple factors 🤔: left/right side, direction of blinds rolled onto tube (in my case is left + under) [image]
> 
> @Koenkk, I don't think there is right way of installation and the best solution will change it back to keep it backwards compatible. Also maybe we should keep it as standard for tubular motor installation to keep open/close state consistent across all motors: like left side + rolled over (and then we have inverted option to invert it)

[image]
[image]

I have blinds that go in both directions, and they previously worked fine with Z2M. I originally set the motor direction and set points with the remote control, and the remote control still functions perfectly making all the blinds go up and down using the physical buttons as expected.

KablammoNick, 2023-11-13:
Just to add to this - I have a mix of left/right roll-over blinds now doing the same. All are set to Open=100 in z2m, with the inverse set on the motor itself.
I have HA feed the blinds into both Google Home and HomeKit too - strangely HomeKit Open/Close works fine, however HA and GH do not.
Setting a % works correctly (80% = 80% open) in all three.
Using the RF remote works fine.

Zigbee2MQTT version
1.33.2

Adapter firmware version
20210708

Adapter
Sonoff ZBDongle-P

TRusselo, 2023-11-13:
I have had the problem since the past few updates.

cover position is correct,  0% is fully closed, 100% is fully open
but open/close is reversed.

when the cover is closed 0%, I get one option, Open, clicking open button does nothing.
If I open the cover to 10% clicking Open will Close the blind.

TRusselo, 2023-11-13:
oh.. and there is no working battery state for this motor, never worked.

donbobka, 2023-11-17:
I'm back home, let me create PR

donbobka, 2023-11-17:
I'm back and tested with reverted state values, but it makes open/close action inverted for me 🤔 (different hardware/firmware revision?)
```
[1, 'state', tuya.valueConverterBasic.lookup({ 'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2) })],
```

@TRusselo @KablammoNick for the following setup (left side) [image]
If you set `Motor direction` to `normal` and click button "down", is direction of rotation the same as in my video? 

https://github.com/Koenkk/zigbee2mqtt/assets/589057/66ce9130-ba88-4496-97ce-387fbfb7b322

KablammoNick, 2023-11-17:
I'm away for a few days but will have a suss when I get home.

donbobka, 2023-11-17:
Also my motor with default options `normal` and `not inverted`: position 0 = open, position 100 = close

TRusselo, 2023-11-17:
changing the "motor direction" does not fix anything in home assistant for me.  cant see **any** change when changing "motor direction" option

donbobka, 2023-11-18:
@TRusselo Does motor give feedback when you change options: small movement up and down? 

https://github.com/Koenkk/zigbee2mqtt/assets/589057/979a3ac2-29d9-4943-9365-51abd62e8df1

donbobka, 2023-11-18:
Some photos of internals
[image]
[image]
[image]
[image]
[image]
[image]

TRusselo, 2023-11-18:
> @TRusselo Does motor give feedback when you change options: small movement up and down? Is your motor `_TZE200_bv1jcqqu` or `_TZE200_7eue9vhc` in Zigbee Manufacturer in zigbee2mqtt?
> 

[image]

_TZE200_7eue9vhc
no reaction from motor when changing motor direction option. not like your video. nothing.

notes:
the up/down was correct couple months ago (or less)
battery % has never worked.

donbobka, 2023-11-18:
I'll revert changes for _TZE200_7eue9vhc, seems it has different protocol

donbobka, 2023-11-18:
I'll revert `_TZE200_7eue9vhc` seems it has different protocol

burnsie-la, 2023-11-18:
> @TRusselo Does motor give feedback when you change options: small movement up and down? Is your motor `_TZE200_bv1jcqqu` or `_TZE200_7eue9vhc` in Zigbee Manufacturer in zigbee2mqtt?
> IMG_9805.mov

I also get no feedback from the motor at all when changing the settings in Z2M.  But I do if I reverse the motor direction using the RF remote that came with the device

donbobka, 2023-11-18:
@burnsie-la 🤔  and your device is `_TZE200_bv1jcqqu`, isn't it. Did Z2M motor direction option worked before `1.33.1`?

burnsie-la, 2023-11-18:
> @burnsie-la 🤔 and your device is `_TZE200_bv1jcqqu`, isn't it. Did Z2M motor direction option worked before `1.33.1`?

I have both _TZE200_bv1jcqqu and _TZE200_7eue9vhc, motor direction and battery never worked via Z2M for either. They seem to function the same.

donbobka, 2023-11-18:
Seems i have and updated revision (just firmware?) that supports zigbee commands for motor direction/set limits. 

[PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6492) with fix

KablammoNick, 2023-11-22:
> I'm back and tested with reverted state values, but it makes open/close action inverted for me 🤔 (different hardware/firmware revision? i see some visual differences: like on-off button).
> 
> ```
> [1, 'state', tuya.valueConverterBasic.lookup({ 'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2) })],
> ```
> 
> If i set `Motor direction` to `reverse` to fix open/close actions, then buttons are inverted
> 
> @TRusselo @KablammoNick for the following setup (left side) [image] If you set `Motor direction` to `normal` and click physical buttons, are directions of rotation same as in my video?

No feedback from motor with settings changes... and changing motor_direction in Z2M makes no difference to the physical buttons on my blinds.

All blinds are TS0601 & _TZE200_7eue9vhc.
All blinds are roll-over, with motor side specified... however I can't remember which ones were inverted on the device at installation. I've made the assumption based off your video and flipping it to match roll-over, the ones with the motor on the right side are "correct" with physical up/down buttons.

```
Name			Model	Manufacturer		Motor Side	Z2M Invert Cover	Invert on Device	Z2M motor_direction
blind_backdoor_left	TS0601	_TZE200_7eue9vhc	Left		False			True			blank?
blind_backdoor_right	TS0601	_TZE200_7eue9vhc	Right		False			False			normal
blind_dining		TS0601	_TZE200_7eue9vhc	Left		False			True			blank?
blind_kitchen		TS0601	_TZE200_7eue9vhc	Right		False			False			normal
blind_sidedoor_left	TS0601	_TZE200_7eue9vhc	Left		False			True			blank?
blind_sidedoor_right	TS0601	_TZE200_7eue9vhc	Right		False			False			normal
```

TRusselo, 2023-11-22:
Im confused why physical buttons have entered the conversation.

[image]

donbobka, 2023-11-22:
@TRusselo logical buttons and physical buttons are connected (at least on my motor)
> I'm back and tested with reverted state values, but it makes open/close action inverted for me 🤔

Hopefully this [video](https://www.youtube.com/watch?v=sdjC6S45dmw) from another PR explains it (still waiting for a few more motors to test it completely).

[PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6503) that reverts logical open/close action has been already merged into master

closed 2023-11-28 by Koenkk