Koenkk/zigbee2mqtt #19540: Sonoff ZB bridge makes Zigbee2mqtt crash

acca84, 2023-11-03:
### What happened?

I'm using Zigbee2mqtt-Koenkk container image on my Qnap nas connected to my Sonoff Zb bridge (flashed in tasmota) over my LAN. I followed this guide : https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_sonoff_zbbridge.html

Everything works fine excepted if i turn my Sonoff RF bridge off then on

The Zigbee2mqtt log is attached below.

It seams Zigbee2mqqt can't connect Sonoff rf bridge whereas i can connect to its webpage on port 80
Sometimes, restarting my docker solves the problem, sometimes not !

### What did you expect to happen?

Zigbee2mqtt reconnects 

### How to reproduce it (minimal and precise)

Connect Zigbee2mqtt and Sonof rf bridge. Turn Sonoff RF bridge off then on

### Zigbee2MQTT version

1.33.1 

### Adapter firmware version

Tasmota Version 13.2.0(zbbridge)-2_7_4_9(2023-10-19T09:01:58)

### Adapter

Sonoff RF bridge tasmota

### Debug log

`info  2023-11-03 13:21:22: Logging to console and directory: '/app/data/log/2023-11-03.13-21-22' filename: log.txt
info  2023-11-03 13:21:23: Starting Zigbee2MQTT version 1.33.1 (commit #7e63039)
info  2023-11-03 13:21:23: Starting zigbee-herdsman (0.19.0)
error 2023-11-03 13:21:43: Error while starting zigbee-herdsman
error 2023-11-03 13:21:43: Failed to start zigbee
error 2023-11-03 13:21:43: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-11-03 13:21:43: Exiting...
error 2023-11-03 13:21:43: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)`

acca84, 2023-11-03:
I have found the solution, the following rule, which starts the TCP server for Zigbee2mqtt was disabled on tasmota, i don't know why 
`backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"Sonoff ZHABridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0`

after issuying 'rule1' in the console, it works
But i don't know why it was disabled

cropse, 2023-12-21:
I have the exact same issue here, and I think it crash after couple day, I have to keep typing this command to console.

Zigbee2MQTT version
1.34.0-1
Adapter firmware version
Tasmota Version 13.3.0(zbbridge)-2_7_4_9(2023-10-19T09:01:58)
Adapter
Sonoff RF bridge tasmota

dinhchinh82, 2024-06-02:
> I have found the solution, the following rule, which starts the TCP server for Zigbee2mqtt was disabled on tasmota, i don't know why `backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; [...]

I got exactly same your issue even I already set the rule1 start automatically after sonoff bridge restart (TCPStart 8888). But if sonoff bridghe is restart slowly then z2m is unable to connect to that TCP coordinator again and hang forever except restarting manually.

closed 2025-01-03 by github-actions[bot]