Koenkk/zigbee2mqtt #19550: GL-B-003P joins/leaves the zigbee network every minute

Zi0nHD, 2023-11-03:
### What happened?

After factory resetting four GL-B-003P bulbs, I cannot connect them to my network properly. They join and leave every 1 minute.
They have been working fine for over a year now.
I use HA/zigbee2mqtt on Raspi 4 with an SSD and a USB extension cable for my SONOFF Zigbee 3.0 USB Dongle Plus-P.
I saw something similar in #11202 but now I don't know if downgrading to another Zigbee Stack version can help in this case. I'm not that deep into this topic.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Debug log

info 2023-11-03 21:26:30Zigbee: allowing new devices to join.
info 2023-11-03 21:26:30MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"520zd-1"}'
info 2023-11-03 21:27:53Device '0xa4c138b8ce8d1f20' joined
info 2023-11-03 21:27:53MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b8ce8d1f20","ieee_address":"0xa4c138b8ce8d1f20"},"type":"device_joined"}'
info 2023-11-03 21:27:53MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138b8ce8d1f20"},"type":"device_connected"}'
info 2023-11-03 21:27:53Starting interview of '0xa4c138b8ce8d1f20'
info 2023-11-03 21:27:54MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b8ce8d1f20","ieee_address":"0xa4c138b8ce8d1f20","status":"started"},"type":"device_interview"}'
info 2023-11-03 21:27:54MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138b8ce8d1f20"},"type":"pairing"}'
info 2023-11-03 21:27:54MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b8ce8d1f20","ieee_address":"0xa4c138b8ce8d1f20"},"type":"device_announce"}'
info 2023-11-03 21:27:54MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c138b8ce8d1f20"},"type":"device_announced"}'
info 2023-11-03 21:27:55Successfully interviewed '0xa4c138b8ce8d1f20', device has successfully been paired
info 2023-11-03 21:27:55Device '0xa4c138b8ce8d1f20' is supported, identified as: Gledopto Zigbee 7W E26/E27 Bulb RGB+CCT (pro) (GL-B-003P)
info 2023-11-03 21:27:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee 7W E26/E27 Bulb RGB+CCT (pro)","exposes":[{"features":[{"access":7,"description":"On/off state of this light","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","label":"Brightness","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","label":"Color temp","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":155},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":495}],"property":"color_temp","type":"numeric","unit":"mired","value_max":495,"value_min":155},{"access":7,"description":"Color temperature after cold power on of this light","label":"Color temp startup","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":155},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":495},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup","type":"numeric","unit":"mired","value_max":495,"value_min":155}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","label":"Effect","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"GL-B-003P","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","label":"Transition","name":"transition","property":"transition","type":"numeric","value_min":0},{"access":2,"description":"When enabled colors will be synced, e.g. if the light supports both color x/y and color temperature a conversion from color x/y to color temperature will be done when setting the x/y color (default true).","label":"Color sync","name":"color_sync","property":"color_sync","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Gledopto"},"friendly_name":"0xa4c138b8ce8d1f20","ieee_address":"0xa4c138b8ce8d1f20","status":"successful","supported":true},"type":"device_interview"}'
info 2023-11-03 21:27:55Configuring '0xa4c138b8ce8d1f20'
info 2023-11-03 21:27:55MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee 7W E26/E27 Bulb RGB+CCT (pro)","friendly_name":"0xa4c138b8ce8d1f20","model":"GL-B-003P","supported":true,"vendor":"Gledopto"},"type":"pairing"}'
info 2023-11-03 21:27:55MQTT publish: topic 'homeassistant/light/0xa4c138b8ce8d1f20/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/0xa4c138b8ce8d1f20/set","device":{"identifiers":["zigbee2mqtt_0xa4c138b8ce8d1f20"],"manufacturer":"Gledopto","model":"Zigbee 7W E26/E27 Bulb RGB+CCT (pro) (GL-B-003P)","name":"0xa4c138b8ce8d1f20","sw_version":"10051203"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"json_attributes_topic":"zigbee2mqtt/0xa4c138b8ce8d1f20","max_mireds":495,"min_mireds":155,"name":null,"object_id":"0xa4c138b8ce8d1f20","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"schema":"json","state_topic":"zigbee2mqtt/0xa4c138b8ce8d1f20","supported_color_modes":["color_temp"],"unique_id":"0xa4c138b8ce8d1f20_light_zigbee2mqtt"}'
info 2023-11-03 21:27:55MQTT publish: topic 'homeassistant/select/0xa4c138b8ce8d1f20/effect/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0xa4c138b8ce8d1f20/set/effect","device":{"identifiers":["zigbee2mqtt_0xa4c138b8ce8d1f20"],"manufacturer":"Gledopto","model":"Zigbee 7W E26/E27 Bulb RGB+CCT (pro) (GL-B-003P)","name":"0xa4c138b8ce8d1f20","sw_version":"10051203"},"enabled_by_default":false,"icon":"mdi:palette","json_attributes_topic":"zigbee2mqtt/0xa4c138b8ce8d1f20","name":"Effect","object_id":"0xa4c138b8ce8d1f20_effect","options":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"unique_id":"0xa4c138b8ce8d1f20_effect_zigbee2mqtt"}'
info 2023-11-03 21:27:55MQTT publish: topic 'homeassistant/sensor/0xa4c138b8ce8d1f20/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138b8ce8d1f20"],"manufacturer":"Gledopto","model":"Zigbee 7W E26/E27 Bulb RGB+CCT (pro) (GL-B-003P)","name":"0xa4c138b8ce8d1f20","sw_version":"10051203"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xa4c138b8ce8d1f20","name":"Linkquality","object_id":"0xa4c138b8ce8d1f20_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigb [... cut ...]
info 2023-11-03 21:37:22MQTT publish: topic 'zigbee2mqtt/Thermostat-SZ', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":19,"frost_detection":"ON","heating":null,"linkquality":80,"local_temperature":17.7,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"off","window_detection":"OFF"}'
info 2023-11-03 21:37:33Finished Touchlink scan
info 2023-11-03 21:37:33MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{"found":[]},"status":"ok","transaction":"520zd-3"}'

M4ch4d0C0l1d4r, 2023-11-04:
I also have this adapter, I have EZSP v12 Coordinator revision 7.3.1.0 build 176. I solved the problem by downgrading Z2M, version 1.33.1-1 works correctly for me, 1.33.2-1 doesn't, it It's problematic, the siren goes off on its own, devices become unavailable from time to time, I've been downgrading for two days, no more problems.

JMoratelli, 2023-11-04:
I also have this adapter, I have EZSP v12 Coordinator revision 7.3.1.0 build 176. I solved the problem by downgrading Z2M, version 1.33.1-1 works correctly for me, 1.33.2-1 doesn't, it It's problematic, the siren goes off on its own, devices become unavailable from time to time, I've been downgrading for two days, no more problems.

Zi0nHD, 2023-11-04:
@JMoratelli okay i will try that.
is there a way to downgrade the z2m addon for HA ?

JMoratelli, 2023-11-04:
http://192.168.x.xxx:8123/hassio/backups 
Just go to this path (change to your IP) and click on the z2m backup and click on "restore"

Zi0nHD, 2023-11-04:
I did a fresh installation (while trying to fix everything) . so unfortunately i got no backups of older versions.Is it Possible to add the repository of z2m at the 1.33.1-1 branch ?

JMoratelli, 2023-11-04:
Hmm, I just don't know how, I've never needed to do that. but it is possible

Zi0nHD, 2023-11-17:
I changed the ZigBee coordinator to the sky connect and switched to zha. No Problems for now. I also tried the conbee 2 with z2m and was able to connect the bulbs again. And after that they staid in the network.

So I think that the firmware on the so sonoff stick was brocken.

closed 2023-11-17 by Zi0nHD