Koenkk/zigbee2mqtt #19553: Error: Connection not initialized

jsprenkle, 2023-11-04:
### What happened?

Deployed zigbee2mqtt as per instructions here. Sonoff zbdongle-e crashes with

error 2023-11-03 07:12:06: Error: Connection not initialized
    at Ezsp.execCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at Driver.startup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:140:25)
    at EZSPAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:172:16)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:60:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)

### What did you expect to happen?

A helpful error message about what's wrong.

### How to reproduce it (minimal and precise)

Get dongle out of box. Plug into working rpi3b+. Configure to connect to work mosquitto mqtt server.
Deploy as per instructions.

### Zigbee2MQTT version

front end wont start, how to find this?

### Adapter firmware version

front end wont start, how to find this?

### Adapter

Sonoff ZBDongle-E

### Debug log

Since I have a crash dump if the code is even somewhat readable I may be able to gain a clue what's wrong. Where is the source code in Github? The paths in the crash dump don't exist in Git or on the machine as scripts.

debug log:

`Nov  3 20:03:00 raspberrypi systemd[1]: Started zigbee2mqtt.
Nov  3 20:03:02 raspberrypi npm[657672]: > zigbee2mqtt@1.33.2 start
Nov  3 20:03:02 raspberrypi npm[657672]: > node index.js
Nov  3 20:03:09 raspberrypi npm[657684]: Zigbee2MQTT:debug 2023-11-03 20:03:09: Loaded state from file /opt/zigbee2mqtt/data/state.json
Nov  3 20:03:09 raspberrypi npm[657684]: Zigbee2MQTT:info  2023-11-03 20:03:09: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-11-03.20-03-09' filename: log.txt
Nov  3 20:03:09 raspberrypi npm[657684]: Zigbee2MQTT:debug 2023-11-03 20:03:09: Removing old log directory '/opt/zigbee2mqtt/data/log/2023-11-03.19-58-55'
Nov  3 20:03:09 raspberrypi npm[657684]: Zigbee2MQTT:info  2023-11-03 20:03:09: Starting Zigbee2MQTT version 1.33.2 (commit #9996c93)
Nov  3 20:03:09 raspberrypi npm[657684]: Zigbee2MQTT:info  2023-11-03 20:03:09: Starting zigbee-herdsman (0.21.0)
Nov  3 20:03:09 raspberrypi npm[657684]: Zigbee2MQTT:debug 2023-11-03 20:03:09: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[67,92,47,128,103,170,212,171],"networkKey":"HIDDEN","panID":45919},"serialPort":{"adapter":"auto","path":"/dev/ttyACM0","rtscts":true}}'
Nov  3 20:03:31 raspberrypi npm[657684]: Zigbee2MQTT:error 2023-11-03 20:03:31: Error while starting zigbee-herdsman
Nov  3 20:03:31 raspberrypi npm[657684]: Zigbee2MQTT:error 2023-11-03 20:03:31: Failed to start zigbee
Nov  3 20:03:31 raspberrypi npm[657684]: Zigbee2MQTT:error 2023-11-03 20:03:31: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Nov  3 20:03:31 raspberrypi npm[657684]: Zigbee2MQTT:error 2023-11-03 20:03:31: Exiting...
Nov  3 20:03:32 raspberrypi npm[657684]: Zigbee2MQTT:error 2023-11-03 20:03:32: Error: Connection not initialized
Nov  3 20:03:32 raspberrypi npm[657684]:     at Ezsp.execCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
Nov  3 20:03:32 raspberrypi npm[657684]:     at Ezsp.version (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
Nov  3 20:03:32 raspberrypi npm[657684]:     at Driver.startup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:140:25)
Nov  3 20:03:32 raspberrypi npm[657684]:     at EZSPAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:172:16)
Nov  3 20:03:32 raspberrypi npm[657684]:     at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
Nov  3 20:03:32 raspberrypi npm[657684]:     at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:60:27)
Nov  3 20:03:32 raspberrypi npm[657684]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
Nov  3 20:03:32 raspberrypi npm[657684]:     at start (/opt/zigbee2mqtt/index.js:107:5)
Nov  3 20:03:32 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Nov  3 20:03:32 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Nov  3 20:03:32 raspberrypi systemd[1]: zigbee2mqtt.service: Consumed 15.702s CPU time.
`

Interloping, 2023-11-13:
I had this problem just start out-of-the-blue this morning after previously having this dongle working for about 2 days only. A few line numbers different in the error, but mostly looks the same:
```
2023-11-14 09:22:10: Error: Connection not initialized
    at Ezsp.execCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at Driver.startup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:59:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)
```
My zigbee2mqtt version is  1.32.2. I tried plugging the dongle into another computer and running and got the same issue. Tried re-flashing the firmware, which reported completing successfully, but I'm still getting this issue.

rsilvestre, 2023-12-04:
I have the same issue with a fresh new Zonoff dongle E

johntdyer, 2023-12-10:
Same issue w/ SLZB-06M running 2.0.11-dev firmware

Kuechenfenster, 2024-04-23:
any fix yet?

aubichon, 2024-04-23:
Nope

Thanks,

Rod Aubichon
250-899-3257
***@***.***

closed 2024-11-23 by github-actions[bot]