Koenkk/zigbee2mqtt #19554: Bseed ts0601 dimmer

leeharper40, 2023-11-04:
### What happened?

Hi, I have had this working for a while now but all of a sudden it has had a few issures with the following errors
Front end:
(Error: Read 0xa4c138fb5720f535/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 52383 - 1 - 3 - 0 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)) 
Addon Log:
Zigbee2MQTT:error 2023-11-04 06:33:28: Failed to configure 'Stairs Dimmer', attempt 1 (Error: Read 0xa4c138fb5720f535/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 52383 - 1 - 3 - 0 - 1 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7))
The problem is i can no longer control the device from the front end.. turn on or off, brightness etc. But if I turn the device on or change brightness from the switch it shows the actioins in the front end straight away.. I have tried re pairing the device and power cycling but results are the same and all other sensors and switches work great 

Thanks
lLee

### Zigbee2MQTT version

 1.33.2 

### Adapter firmware version

20221226

### Adapter

usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus

timothylui, 2023-11-04:
I'm also having this same issue

leeharper40, 2023-11-07:
Hi,
now Having exactly the same issue with a Bseed TS0012 been working great fot months . Swapped out the dimmer with exactly the same type as I never got around to swapping an old wifi one and for thats working great for the time being. I suppose my question is , are these just shit switches in general and just bin them off ???

closed 2024-06-05 by github-actions[bot]