Koenkk/zigbee2mqtt #19555: Aqara Smart Curtain Motor C3

zonzlee, 2023-11-04:
### Link

https://www.aqara.cn/curtain-c3_overview

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x54ef4410009630e0","nwkAddr":59606,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.curtain.acn04","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":515,"inClusterList":[5,4,3,0,258,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.curtain.acn04","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":9,"stackVersion":27,"hwVersion":1,"dateCode":"20230407"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":9,"stackVersion":27,"hwVersion":1,"dateCode":"20230407","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1699081757490,"defaultSendRequestWhen":"immediate"}

### Comments

Hello, Aqara has recently released the smart curtain motor C3, can this device be supported in the new version.

i-93, 2024-01-04:
+1 Please add the support for it

i-93, 2024-01-05:
Tried to follow [the guide](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html). Created the file ZNCLDJ01LM.js based on the closest Aqara C2 code.

`
const {} = require('zigbee-herdsman-converters/lib/modernExtend');

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.curtain.acn04'],
    model: 'ZNCLDJ01LM',
    vendor: 'Xiaomi',
    description: 'Aqara C3 curtain motor',
    fromZigbee: [xiaomi.fromZigbee.xiaomi_basic, fz.xiaomi_curtain_position, fz.xiaomi_curtain_position_tilt, fz.xiaomi_curtain_hagl07_status],
    toZigbee: [tz.xiaomi_curtain_position_state, tz.xiaomi_curtain_options],
    onEvent: async (type, data, device) => {
        // The position (genAnalogOutput.presentValue) reported via an attribute contains an invalid value
        // however when reading it will provide the correct value.
        if (data.type === 'attributeReport' && data.cluster === 'genAnalogOutput') {
            await device.endpoints[0].read('genAnalogOutput', ['presentValue']);
        }
    },
    exposes: [e.cover_position().setAccess('state', ea.ALL),
        e.binary('running', ea.STATE, true, false)
            .withDescription('Whether the motor is moving or not'),
        e.enum('motor_state', ea.STATE, ['closing', 'opening', 'stop'])
            .withDescription('The current state of the motor.'), e.power_outage_count()],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
`

Device is recognized, but all the attempts to access it produce errors, like:
`
zigbee2mqtt  | Zigbee2MQTT:error 2024-01-05 07:03:58: Publish 'get' 'state' to '0x54ef441000a483f8' failed: 'Error: Read 0x54ef441000a483f8/1 genAnalogOutput([85], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
zigbee2mqtt  | Zigbee2MQTT:debug 2024-01-05 07:03:58: Error: Read 0x54ef441000a483f8/1 genAnalogOutput([85], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-05 07:03:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'state' to '0x54ef441000a483f8' failed: 'Error: Read 0x54ef441000a483f8/1 genAnalogOutput([85], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"0x54ef441000a483f8"},"type":"zigbee_publish_error"}'
`

Pleaser advice.
Thanks!

zonzlee, 2024-01-10:
> +1
> 
> 请添加对它的支持。如果需要，我随时准备帮助调试。

First of all, thank you for your reply.

 I used the aqara C2 file before, yes, all access will get an error.  
Then I tried to use it through ZHA, which correctly recognized the device, but not the curtain position, and could only control its opening and closing through the curtain service of home assistant, and apparently I was able to use it, although it was not perfect.

What I want to tell you is that all my ZigBee devices are in ZHA, even if ZigBee2mqtt supports it, I can't use it, obviously I can't migrate all my devices to ZigBee2mqtt.

If you just want to help me use it, thanks, I can only use it in ZHA.

If, you want to add a changer for aqara c3, or you want to use it in the future, I'm willing to debug it, please give me time, I need to add ZigBee2mqtt to home assistant.

Thank you!

zonzlee, 2024-01-10:
> +1
> 
> 请添加对它的支持。如果需要，我随时准备帮助调试。

自动翻译了，才看到你回复的是中文，我在bbs.hassbian.com中有发布对aqara C3 的支持请求，我们可以在这个论坛中沟通。
这个是链接 https://bbs.hassbian.com/thread-23338-1-1.html
谢谢。

latel, 2024-01-11:
> First of all, thank you for your reply.
>
> I used the aqara C2 file before, yes, all access will get an error. Then I tried to use it through ZHA, which correctly recognized the device, but not the [...]

so chinglish, 😄

zonzlee, 2024-01-15:
> > > +1
> > > 请添加对它的支持。如果需要，我随时准备帮助调试。
> > 
> > 
> > 首先，感谢您的回复。
> > 我之前使用过 aqara C2 文件，是的，所有访问都会出错。然后我尝试通过ZHA使用它，它正确识别了设备，但不能识别窗帘位置，只能通过家庭助理的窗帘服务来控制它的开合，显然我能够使用它，尽管它并不完美。
> > 我想告诉你的是，我所有的 ZigBee 设备都在 ZHA 中，即使 ZigBee2mqtt 支持它，我也不能使用它，显然我不能将我所有的设备迁移到 ZigBee2mqtt。
> > 如果你只是想帮我使用它，谢谢，我只能在ZHA使用它。
> > 如果，你想给aqara c3添加一个转换器，或者你想在将来使用它，我愿意调试它，请给我时间，我需要将ZigBee2mqtt添加到家庭助手中。
> > 谢谢！
> 
> 所以 Chinglish， 😄

微软翻译  😄

marcuscps, 2024-04-22:
+1

Any updates here?
I have a C3 motor and I'm running Zigbee2Mqtt.

I can help with any debugging if needed.
I'm also willing to contribute, even though it would be my first contribution to Zigbee2Mqtt.

Thanks a lot!

leonshadow15, 2024-04-22:
I'm interested in buying and using c3 with HA too.
Has anyone worked on it so far?

marthubner, 2024-07-26:
Here is an external converter for this device for some basic functions that I somehow put together.
The device also has other features, but to try add them I will need to sniff Zigbee traffic, I don't have a sniffer yet.

```
const { identify } = require('zigbee-herdsman-converters/lib/modernExtend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const lumi = require('zigbee-herdsman-converters/lib/lumi');
const { lumiZigbeeOTA } = require('zigbee-herdsman-converters/lib/lumi');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const { isString, getOptions, assertNumber, getFromLookup } = require('zigbee-herdsman-converters/lib/utils');

// Define lumi_curtain_position here
const lumi_curtain_position_state = {
    key: ['state', 'position'],
    options: [exposes.options.invert_cover()],
    convertSet: async (entity, key, value, meta) => {
        if (Array.isArray(meta.mapped)) throw new Error('Not supported for groups');
        if (key === 'state' && typeof value === 'string' && value.toLowerCase() === 'stop') {
            if (['ZNJLBL01LM', 'ZNCLDJ14LM'].includes(meta.mapped.model)) {
                const payload = { 'presentValue': 2 };
                await entity.write('genMultistateOutput', payload);
            } else {
                await entity.command('closuresWindowCovering', 'stop', {}, getOptions(meta.mapped, entity));
            }

            if (!['ZNCLDJ11LM', 'ZNJLBL01LM', 'ZNCLBL01LM', 'ZNCLDJ01LM'].includes(meta.mapped.model)) {
                // Request position update on stop for certain models
                await entity.read('genAnalogOutput', [0x0055]);
            }

            return { state: { state: 'STOP' } };
        } else {
            const lookup = { 'open': 100, 'close': 0, 'on': 100, 'off': 0 };

            value = typeof value === 'string' ? value.toLowerCase() : value;
            if (isString(value)) {
                value = getFromLookup(value, lookup);
            }
            assertNumber(value);
            value = meta.options.invert_cover ? 100 - value : value;

            if (['ZNCLBL01LM', 'ZNCLDJ01LM'].includes(meta.mapped.model)) {
                await entity.command('closuresWindowCovering', 'goToLiftPercentage', { percentageliftvalue: value }, getOptions(meta.mapped, entity));
            } else {
                const payload = { 'presentValue': value };
                await entity.write('genAnalogOutput', payload);
            }
        }
    },
    convertGet: async (entity, key, meta) => {
        if (!Array.isArray(meta.mapped) && ['ZNCLBL01LM', 'ZNCLDJ01LM'].includes(meta.mapped.model)) {
            await entity.read('closuresWindowCovering', ['currentPositionLiftPercentage']);
        } else {
            await entity.read('genAnalogOutput', [0x0055]);
        }
    },
};

const lumidef = {manufacturerCode: 0x115f, disableDefaultResponse: true};

const tz_curtain_hand_open = {
    key: ['hand_open'],
    convertSet: async (entity, key, value, meta) => {
        await entity.write('manuSpecificLumi', {'curtainHandOpen': value}, lumidef);
    },
    convertGet: async (entity, key, meta) => {
        await entity.read('manuSpecificLumi', ['curtainHandOpen'], lumidef);
    },
};

const fz_curtain_hand_open = {
    cluster: 'manuSpecificLumi',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};
        if (msg.data.hasOwnProperty('curtainHandOpen')) {
            result.hand_open = msg.data['curtainHandOpen'] === 1;
        }
        return result;
    },
};

const tz_curtain_reverse = {
    key: ['reverse_direction'],
    convertSet: async (entity, key, value, meta) => {
        await entity.write('closuresWindowCovering', {'windowCoveringMode': value});
    },
    convertGet: async (entity, key, meta) => {
        await entity.read('closuresWindowCovering', ['windowCoveringMode']);
    },
};

const fz_curtain_reverse =  {
    cluster: 'closuresWindowCovering',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};
        if (msg.data.hasOwnProperty('windowCoveringMode')) {
            result.reverse_direction = msg.data['windowCoveringMode'] === 1;
        }
        return result;
    },
};

const definition = {
    zigbeeModel: ['lumi.curtain.acn04'],
    model: 'ZNCLDJ01LM',
    vendor: 'Aqara',
    description: 'Curtain controller C3',
    fromZigbee: [
        lumi.fromZigbee.lumi_basic,
        lumi.fromZigbee.lumi_curtain_position_tilt,
        lumi.fromZigbee.lumi_specific,
        fz_curtain_hand_open,
        fz_curtain_reverse,
    ],
    toZigbee: [
        lumi_curtain_position_state,
        lumi.toZigbee.lumi_curtain_limits_calibration,
        tz_curtain_hand_open,
        tz_curtain_reverse,
    ],
    exposes: [
        e.cover_position().setAccess('state', ea.ALL),
        e.enum('limits_calibration', ea.SET, ['start', 'end', 'reset']).withDescription('Calibrate the position limits'),
        e.binary('hand_open', ea.ALL, true, false).withDescription('Pulling curtains by hand starts the motor'),
        e.binary('reverse_direction', ea.ALL, true, false).withDescription('Whether the curtain direction is inverted'),
    ],
    extend: [identify()],
    meta: {},
};

module.exports = definition;

```

marcuscps, 2024-08-02:
> Here is an external converter for this device for some basic functions that I somehow put together. The device also has other features, but to try add them I will need to sniff Zigbee traffic, I [...]

Worked like a charm!! :) 
Thanks a lot!

leonshadow15, 2024-08-02:
> Here is an external converter for this device for some basic functions that I somehow put together. The device also has other features, but to try add them I will need to sniff Zigbee traffic, I [...]

thank you! are you planning to get a sniffer soon by any chance?

leonshadow15, 2024-08-02:
> > thank you! are you planning to get a sniffer soon by any chance?
> 
> Sniffer already arrived and I tried to follow [this](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#usage) guide but it turned out I was unable to sniff the zigbee traffic between the E1 Aqara Hub and the C3 curtain motor. Either I did something wrong or Aqara is using some non standard Trust Center link key as I could not get the messages decrypted.
> 
> So I chose trial and error approach and this is what I currently have. I will upload updated converter soon. ![image](https://private-user-images.githubusercontent.com/86839034/354591076-fdea7107-a4f3-4efc-ac98-032a7270f688.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI2MDE0OTYsIm5iZiI6MTcyMjYwMTE5NiwicGF0aCI6Ii84NjgzOTAzNC8zNTQ1OTEwNzYtZmRlYTcxMDctYTRmMy00ZWZjLWFjOTgtMDMyYTcyNzBmNjg4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODAyVDEyMTk1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEzZjgwMzk0ZWFkYmRmODcxYTQ4MmI1NWY3ODVmOTQzZjc5MWMzMDZkMjcyZDRmZWMzODE5MGEwN2NhMmQ1NTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.BtcRvcwLTXl2Zb_7dcODILsR12ge9CU3swBMG-wUl-E)

wow excellent job! thanks for contributing. it looks that the most important features are already there! keep us posted :)

Mozgoklui, 2024-08-02:
Can someone explain to a newbie how and where to insert the code so that C3 somehow works in HA? Thank you very much!

leonshadow15, 2024-08-02:
> Can someone explain to a newbie how and where to insert the code so that C3 somehow works in HA? Thank you very much!

correct me if I'm wrong but probably we should wait for him to add it as a new z2m supported device first

marthubner, 2024-08-02:
> Can someone explain to a newbie how and where to insert the code so that C3 somehow works in HA? Thank you very much!

In File Editor in HA navigate to /homeassistant/zigbee2mqtt and create a new file there called for example C3.js and put the code in that file.
Then in /homeassistant/zigbee2mqtt/configuration.yaml add the bellow lines and restart Zigbee2MQTT addon.
```
external_converters:
    - C3.js
```

> correct me if I'm wrong but probably we should wait for him to add it as a new z2m supported device first

I will try to add it to z2m so it is supported, but it will take some time. You can use the external converter in the meantime.

Mozgoklui, 2024-08-02:
> In File Editor in HA navigate to /homeassistant/zigbee2mqtt and create a new file there called for example C3.js and put the code in that file. Then in /homeassistant/zigbee2mqtt/configuration.yaml [...]

Thanks. But I allredy tried to do it, but dev.console while conf checking send me this:
"Integration error: external_converters - Integration 'external_converters' not found."

Mozgoklui, 2024-08-02:
> In File Editor in HA navigate to /homeassistant/zigbee2mqtt and create a new file there called for example C3.js and put the code in that file. Then in /homeassistant/zigbee2mqtt/configuration.yaml add the bellow lines and restart Zigbee2MQTT addon.
> ```
> external_converters:
>     - C3.js
> ```
It works great!!!

marthubner, 2024-08-02:
Updated external converter here: [ZNCLDJ01LM.zip](https://github.com/user-attachments/files/16473355/ZNCLDJ01LM.zip)
Let me know if anything is not working as expected.

```
const { identify, quirkAddEndpointCluster } = require('zigbee-herdsman-converters/lib/modernExtend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const lumi = require('zigbee-herdsman-converters/lib/lumi');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const { isString, getOptions, assertNumber, getFromLookup } = require('zigbee-herdsman-converters/lib/utils');

const lumiZigbeeOTA = () => {
    // Many Lumi devices miss OTA on endpoint 1 even while supporting it.
    // https://github.com/Koenkk/zigbee2mqtt/issues/10660
    const result = quirkAddEndpointCluster({
        endpointID: 1,
        outputClusters: ['genOta'],
    });
    result.ota = ota.zigbeeOTA;
    return result;
};

// Define lumi_curtain_position here
const tz_curtain_position_state = {
    key: ['state', 'position'],
    options: [exposes.options.invert_cover()],
    convertSet: async (entity, key, value, meta) => {
        if (Array.isArray(meta.mapped)) throw new Error('Not supported for groups');
        if (key === 'state' && typeof value === 'string' && value.toLowerCase() === 'stop') {
            if (['ZNJLBL01LM', 'ZNCLDJ14LM'].includes(meta.mapped.model)) {
                const payload = { 'presentValue': 2 };
                await entity.write('genMultistateOutput', payload);
            } else {
                await entity.command('closuresWindowCovering', 'stop', {}, getOptions(meta.mapped, entity));
            }

            if (!['ZNCLDJ11LM', 'ZNJLBL01LM', 'ZNCLBL01LM', 'ZNCLDJ01LM'].includes(meta.mapped.model)) {
                // Request position update on stop for certain models
                await entity.read('genAnalogOutput', [0x0055]);
            }

            return { state: { state: 'STOP' } };
        } else {
            const lookup = { 'open': 100, 'close': 0, 'on': 100, 'off': 0 };

            value = typeof value === 'string' ? value.toLowerCase() : value;
            if (isString(value)) {
                value = getFromLookup(value, lookup);
            }
            assertNumber(value);
            value = meta.options.invert_cover ? 100 - value : value;

            if (['ZNCLBL01LM', 'ZNCLDJ01LM'].includes(meta.mapped.model)) {
                await entity.command('closuresWindowCovering', 'goToLiftPercentage', { percentageliftvalue: value }, getOptions(meta.mapped, entity));
            } else {
                const payload = { 'presentValue': value };
                await entity.write('genAnalogOutput', payload);
            }
        }
    },
    convertGet: async (entity, key, meta) => {
        if (!Array.isArray(meta.mapped) && ['ZNCLBL01LM', 'ZNCLDJ01LM'].includes(meta.mapped.model)) {
            await entity.read('closuresWindowCovering', ['currentPositionLiftPercentage']);
        } else {
[... 255 lines cut ...]
        tz_curtain_manual_stop,
        tz_curtain_beep,
    ],
    exposes: [
        e.cover_position().setAccess('state', ea.ALL),
        e.numeric('curtain_speed', ea.ALL).withValueMin(1).withValueMax(100).withDescription('Speed of curtain movement').withUnit('%'),
        e.binary('hand_open_close', ea.ALL, 'ON', 'OFF').withDescription('Gently pull to open/close the curtain automatically'),
        e.binary('adaptive_pulling_speed', ea.ALL, 'ON', 'OFF').withDescription('The faster/slower the curtain is pulled manually, the faster/slower the curtain will move'),
        e.binary('manual_stop', ea.ALL, 'ON', 'OFF').withDescription('Manually pulling the curtain during operation stops the motor'),
        e.binary('reverse_direction', ea.ALL, true, false).withDescription('Whether the curtain direction is inverted'),
        e.enum('curtain_state', ea.STATE, ['Opening', 'Closing', 'Stopped', 'Blocked']).withDescription('Current state of the curtain (Opening, Closing, Stopped, Blocked)'),
        e.enum('last_manual_operation', ea.STATE, ['Open', 'Close', 'Stop']).withDescription('Last triggered manual operation (Open, Close, Stop'),
        e.numeric('curtain_position', ea.STATE).withValueMin(1).withValueMax(100).withUnit('%'),
        e.numeric('traverse_time', ea.STATE).withUnit('sec').withDescription('Time in seconds to get from one end to another'),
        e.enum('limits_calibration', ea.SET, ['Start', 'End', 'Reset']).withDescription('Calibrate the position limits'),
        e.enum('calibration_status', ea.STATE, ['Not calibrated', 'Half calibrated', 'Fully calibrated'])
            .withDescription('Calibration status of the curtain: Not calibrated, Half calibrated, Fully calibrated'),
        e.calibrated(),
        e.enum('beep', ea.SET, [0,1,2]).withDescription('Device will beep for chosen amount of seconds'),
    ],
    extend: [identify(), lumiZigbeeOTA()],
};

module.exports = definition;
```

Mozgoklui, 2024-08-03:
[marthubner](https://github.com/marthubner), I don't know much about GitHub. Is there any way to donate here?

marthubner, 2024-08-03:
> [marthubner](https://github.com/marthubner), I don't know much about GitHub. Is there any way to donate here?

I've just activated the option to accept donations on my github account so you can do it there if you feel to do so, I appreciate it, but there's really no need for it, I made this mainly for my own use.

Mozgoklui, 2024-08-03:
> I've just activated the option to accept donations on my github account so you can do it there if you feel to do so, I appreciate it, but there's really no need for it, I made this mainly for my own [...]

Bro, I have a 4-meter curtain in my bedroom, I have long been missing the ability to control it remotely. Now it will work even from the alarm clock. Thanks!

marcuscps, 2024-08-05:
@marthubner This is fantastic work! Can't thank you enough.

marthubner, 2024-08-05:
Another updated version: 
[ZNCLDJ01LM.zip](https://github.com/user-attachments/files/16496234/ZNCLDJ01LM.zip)

Added a toggle button because the motor also supports it.
Added an automatic calibration process that mimics the way the official Aqara app does it. (I couldn't test it properly because my curtain rail hasn't arrived yet, so try at your own risk).

```
const {identify, quirkAddEndpointCluster} = require('zigbee-herdsman-converters/lib/modernExtend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const lumi = require('zigbee-herdsman-converters/lib/lumi');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const {isString, getOptions, getFromLookup} = require('zigbee-herdsman-converters/lib/utils');

const lumiZigbeeOTA = () => {
    // Many Lumi devices miss OTA on endpoint 1 even while supporting it.
    // https://github.com/Koenkk/zigbee2mqtt/issues/10660
    const result = quirkAddEndpointCluster({
        endpointID: 1,
        outputClusters: ['genOta'],
    });
    result.ota = ota.zigbeeOTA;
    return result;
};

const tz_curtain_toggle = {
    key: ['toggle'],
    convertSet: async (entity, key, value, meta) => {
        if (key === 'toggle' && typeof value === 'string' && value.toLowerCase() === 'toggle') {
            await entity.write('manuSpecificLumi', {0x0420: {value: 3, type: 0x20}}, lumidef);
        }
    },
    convertGet: async (entity, key, meta) => {
        await entity.read('closuresWindowCovering', ['currentPositionLiftPercentage']);
    },
};

const tz_curtain_position_state = {
    key: ['state', 'position'],
    convertSet: async (entity, key, value, meta) => {
        if (key === 'state' && typeof value === 'string' && value.toLowerCase() === 'stop') {
            await entity.command('closuresWindowCovering', 'stop', {}, getOptions(meta.mapped, entity));
            return { state: { state: 'STOP' } };
        } else {
            const lookup = { 'open': 100, 'close': 0 };
            value = typeof value === 'string' ? value.toLowerCase() : value;
            if (isString(value)) {
                value = getFromLookup(value, lookup);
            }
            await entity.command('closuresWindowCovering', 'goToLiftPercentage', { percentageliftvalue: value }, getOptions(meta.mapped, entity));
        }
    },
    convertGet: async (entity, key, meta) => {
        await entity.read('closuresWindowCovering', ['currentPositionLiftPercentage']);
    },
};

const fz_curtain_position_state = {
    cluster: 'closuresWindowCovering',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};
        if (msg.data.hasOwnProperty('currentPositionLiftPercentage') && msg.data['currentPositionLiftPercentage'] <= 100) {
            const position = msg.data['currentPositionLiftPercentage'];
[... 347 lines cut ...]
        e.cover_position().setAccess('state', ea.ALL),
        e.numeric('curtain_speed', ea.ALL).withValueMin(1).withValueMax(100).withDescription('Speed of curtain movement').withUnit('%'),
        e.binary('hand_open_close', ea.ALL, 'ON', 'OFF').withDescription('Gently pull to open/close the curtain automatically'),
        e.binary('adaptive_pulling_speed', ea.ALL, 'ON', 'OFF').withDescription('The faster/slower the curtain is pulled manually, the faster/slower the curtain will move'),
        e.binary('manual_stop', ea.ALL, 'ON', 'OFF').withDescription('Manually pulling the curtain during operation stops the motor'),
        e.binary('reverse_direction', ea.ALL, true, false).withDescription('Whether the curtain direction is inverted'),
        e.enum('curtain_state', ea.STATE, ['Opening', 'Closing', 'Stopped', 'Blocked']).withDescription('Current state of the curtain (Opening, Closing, Stopped, Blocked)'),
        e.enum('last_manual_operation', ea.STATE, ['Open', 'Close', 'Stop']).withDescription('Last triggered manual operation (Open, Close, Stop)'),
        e.numeric('curtain_position', ea.STATE).withValueMin(1).withValueMax(100).withUnit('%'),
        e.numeric('traverse_time', ea.STATE).withUnit('sec').withDescription('Time in seconds to get from one end to another'),
        e.enum('manual_calibration', ea.SET, ['Start', 'End', 'Reset']).withDescription('Calibrate the position limits'),
        e.enum('automatic_calibration', ea.SET, ['Calibrate'])
            .withDescription('Performs an automatic calibration process similar to Aqara’s method to set curtain limits.'),
        e.enum('calibration_status', ea.STATE, ['Not calibrated', 'Half calibrated', 'Fully calibrated'])
            .withDescription('Calibration status of the curtain (Not calibrated, Half calibrated, Fully calibrated)'),
        e.calibrated(),
        e.enum('beep', ea.SET, [0,1,2]).withDescription('Device will beep for chosen amount of seconds'),
    ],
    extend: [identify(), lumiZigbeeOTA()],
    meta: {},
};

module.exports = definition;

```

latel, 2024-08-06:
> Sniffer already arrived and I tried to follow [this](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#usage) guide but it turned out I was unable to sniff the zigbee traffic [...]

many thanks, how to adjust the travel range of curtains, whether it’s single-opening or double-opening, sort of?

marthubner, 2024-08-08:
> many thanks, how to adjust the travel range of curtains, whether it’s single-opening or double-opening, sort of?

There are also other attributes that I haven't found out what they do or if they actually do anything. I haven't found that changing them has changed the behaviour of the motor in any way.
One of these attributes (namely 0x0408) has states 0,1,2 and can be read and written to. I think it might represent a left-open/right-open/double-open kind of setting but as I mentioned, I haven't observed any effect on device behavior.
I think maybe Aqara uses it to display a proper curtain animation in their App which is not case of Zigbee2MQTT.
It does not really matter if your curtain is single-open or double-open. You just set your end positions and use "Reverse direction" setting if needed.

latel, 2024-08-12:
> There are also other attributes that I haven't found out what they do or if they actually do anything. I haven't found that changing them has changed the behaviour of the motor in any way. One of [...]

Thank you for your detailed response. I have another question: how can I modify the default control buttons for the curtains in Home Assistant? My curtains open from left to right, but the default control buttons in Home Assistant show them as opening up and down.

[image]

hzonz, 2024-10-31:
> There are also other attributes that I haven't found out what they do or if they actually do anything. I haven't found that changing them has changed the behaviour of the motor in any way. One of [...]

Thank! Because of your involvement, we are able to use the aqara C3 device correctly in HA.

Because my zigbee devices are all used in ZHA, currently this device can only use position percentage control in ZHA, and no other functions can be used.

Do you have any plans to write quirks for aqara C3 in the future?

latel, 2024-10-31:
Does this merged already into zigbee2mqtt new release now?

rvalitov, 2025-01-03:
@latel I see only basic support in Z2M, but not the settings shown on screenshot by @marthubner so I guess this issue is still not resolved...

marthubner, 2025-01-05:
> Because my zigbee devices are all used in ZHA, currently this device can only use position percentage control in ZHA, and no other functions can be used.
> 
> Do you have any plans to write quirks for aqara C3 in the future?

Hi, I don´t use ZHA so I don't plan to add support for this device.

> Does this merged already into zigbee2mqtt new release now?

> @latel I see only basic support in Z2M, but not the settings shown on screenshot by @marthubner so I guess this issue is still not resolved...

This has been merged since release [1.40.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.0#:~:text=%237848%20ZNCLDJ01LM%20Aqara%20Curtain%20controller%20C3%20(%40marthubner))

closed 2025-03-15 by github-actions[bot]