Koenkk/zigbee2mqtt #19556: Aqara Smart Radiator Thermostat E1 SRTS-A01

chrineuh, 2023-11-04:
### What happened?

There is no way to use the device with an external sensor. 
How can I solve this in Iobroker? 

[image]

### What did you expect to happen?

I want to set the temperature of the device to an external temperature sensor, but there is no option to set the value to this device. All values are like: 
zigbee.1.54ef44100073b6e3.device_temperature
zigbee.1.54ef44100073b6e3.local_temperature
are read only. 

### Zigbee2MQTT version

1.8.24

### Adapter firmware version

1.8.24

### Adapter

ioBroker.zigbee

Mwirenfeldt, 2023-11-09:
it is posible to send a Mqtt to set the local_temp from a external
i use this code in my automation:
`      - conditions:
          - condition: trigger
            id: tempChange
        sequence:
          - if:
              - condition: template
                value_template: "{{ not is_state(aqara_trv_select_entity, 'external')}}"
            then:
              - service: select.select_option
                target:
                  entity_id: "{{aqara_trv_select_entity}}"
                data:
                  option: external
              - delay:
                  hours: 0
                  minutes: 0
                  seconds: 10
                  milliseconds: 0
          - service: mqtt.publish
            data:
              topic: "zigbee2mqtt/{{aqara_trv_device_name}}/set/sensor_temp"
              payload_template: >-
                {{ (states(external_temp_sensor)|float(0)|round(1)) }}`
                
     i found this Home Assistant Blueprint:
     https://community.home-assistant.io/t/z2m-aqara-trv-e1-link-external-temperature-sensor/609689/18

chrineuh, 2023-11-09:
Hello, 

thank you, but how i can do that in iobroker. 
The data state for that value is not visible...

Koenkk, 2023-11-09:
Please create an issue in the iobroker issue tracker, this is the z2m one.

closed 2023-11-09 by Koenkk