Koenkk/zigbee2mqtt #19557: [New device support]: Tuya TS0601 TZE204_81yrt3lo

Aposei, 2023-11-04:
### Link

https://es.aliexpress.com/item/1005005994003685.html?spm=a2g0o.order_list.order_list_main.5.4d10194dz2u28M&gatewayAdapt=glo2esp

### Database entry

{"id":14,"type":"EndDevice","ieeeAddr":"0xa4c13862871e3f84","nwkAddr":43949,"manufId":4417,"manufName":"_TZE204_vawy74yh","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001f\u0000\u0000\u0000e \u0000\u0000\u0000\u0012","65506":56,"65508":1,"appVersion":73,"modelId":"TS0601","manufacturerName":"_TZE204_vawy74yh","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1699095641358,"defaultSendRequestWhen":"immediate"}

### Comments

I don't have enough knowledge to do it myself, can you help to create a converter? Thanks.

frederick9372, 2023-11-04:
Same probleme with same device even update 1.33.2
https://fr.aliexpress.com/item/1005005997839493.html?spm=a2g0o.order_list.order_list_main.17.53075e5bRXG1fg&gatewayAdapt=glo2fra

Aposei, 2023-11-06:
Hi. I'm trying to map the tuyaDatapoints but in the log I only get this

[image]

Iroamstaff, 2023-11-25:
I also have the same smoke detector, no progress for support?

Aposei, 2023-11-26:
> I also have the same smoke detector, no progress for support?

I can't find the tuyaDatapoints

closed 2023-12-05 by Aposei