Koenkk/zigbee2mqtt #19558: Hue Motion sensors won't pair anymore

vogler, 2023-11-04:
### What happened?

Two out of my three Hue motion sensors are no longer connected and also can't be paired anymore. The log just repeats `device_announced` and `Retrieving state ... after reconnect`.
It's the old motion sensors: https://www.zigbee2mqtt.io/devices/9290012607.html
I tried pairing them to my Hue Bridge (channel 20) and then back to z2m (channel 11) as recommended there, but that also didn't work. Also tried >10 times with a USB extension cable.

### What did you expect to happen?

Motion sensors should pair as they did before.

### Zigbee2MQTT version

1.33.2-dev commit: 366e29d4

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (CC2652P CP2102N)

### Debug log

Zigbee2MQTT:info  2023-11-04 15:01:44: Zigbee2MQTT started!

  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,174,32,0,12] +12s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,174,32,0,12] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [174,32,0] - 12 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":8366,"relaycount":0,"relaylist":[]} +12s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,159,174,32,132,8,84,137] +24ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,159,174,32,132,8,84,137] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 159 - [174,32,132,8,84] - 137 +0ms
  zigbee-herdsman:adapter:zStack:znp:error Error while parsing to ZpiObject 'Error: CommandID '159' from subsystem '5' not found
  zigbee-herdsman:adapter:zStack:znp:error     at Function.fromUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:60:19)
  zigbee-herdsman:adapter:zStack:znp:error     at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:103:38)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.emit (node:events:514:28)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:19:14)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.Transform._write (node:internal/streams/transform:175:8)
  zigbee-herdsman:adapter:zStack:znp:error     at writeOrBuffer (node:internal/streams/writable:392:12)
  zigbee-herdsman:adapter:zStack:znp:error     at _write (node:internal/streams/writable:333:10)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.Writable.write (node:internal/streams/writable:337:10)
  zigbee-herdsman:adapter:zStack:znp:error     at SerialPort.ondata (node:internal/streams/readable:766:22)' +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +176ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,117,225,117,225,84,53,40,3,1,136,23,0,128,221] +24s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,117,225,117,225,84,53,40,3,1,136,23,0,128,221] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [117,225,117,225,84,53,40,3,1,136,23,0,128] - 221 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":57717,"nwkaddr":57717,"ieeeaddr":"0x0017880103283554","capabilities":128} +24s
  zigbee-herdsman:adapter:zStack:adapter Discovering route to 57717 +40s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":57717,"options":0,"radius":30} +36s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,117,225,0,30,238] +36s
  zigbee-herdsman:controller:log Device announce '0x0017880103283554' +37s
Zigbee2MQTT:debug 2023-11-04 15:02:19: Device 'Hue_Motion_Living' announced itself
Zigbee2MQTT:info  2023-11-04 15:02:20: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hue_Motion_Living","ieee_address":"0x0017880103283554"},"type":"device_announce"}'
Zigbee2MQTT:info  2023-11-04 15:02:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Hue_Motion_Living"},"type":"device_announced"}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +227ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,117,225,84,53,40,3,1,136,23,0,0,0,195,254,1,101,69,0,33] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,117,225,84,53,40,3,1,136,23,0,0,0,195,254,1,101,69,0,33] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [117,225,84,53,40,3,1,136,23,0,0,0] - 195 +2ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":57717,"extaddr":"0x0017880103283554","parentaddr":0} +238ms
  zigbee-herdsman:controller:log Device '0x0017880103283554' joined +231ms
Zigbee2MQTT:info  2023-11-04 15:02:20: Accepting joining not in blocklist device '0x0017880103283554'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0} +37s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x0017880103283554' accepted by handler +10ms
  zigbee-herdsman:controller:log Not interviewing '0x0017880103283554', completed 'true', in progress 'false' +1ms
[... 36 lines cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +233ms
Zigbee2MQTT:debug 2023-11-04 15:02:48: Retrieving state of 'Hue_Motion_Kitchen' after reconnect
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,3,69,182,0,0,0,240] +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":2,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,3,2,11,254,0]}} +20ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,2,0,30,6,0,25,3,2,11,254,0,234] +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +18ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,2,55] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,2,55] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,2] - 55 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":2} +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:debug 2023-11-04 15:06:37: Saving state to file /opt/zigbee2mqtt/data/state.json

vogler, 2023-11-04:
Ok, after first removing the sensors, pairing worked again.
Shouldn't this be mentioned somewhere in the docs? (Or even better, do it automatically if it's possible to detect this failure state.)

closed 2023-11-04 by vogler

avanc, 2024-09-15:
I'm stuck with the same problem.
@vogler What do you mean with removing the sensor? Just remove the lines from  configuration.yaml or is there more to be done?

vogler, 2024-09-15:
@avanc Removing it from configuration.yaml wasn't enough. Had to do this: https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-device-remove

avanc, 2024-09-16:
@vogler Thanks for the quick reply. This solved it also on my side.

mircsicz, 2024-10-29:
Worked for me as well...