Koenkk/zigbee2mqtt #19559: Aqara T1 door & window contact sensor is only responding N/A

Mwirenfeldt, 2023-11-04:
### What happened?

I was adding a Aqara T1 door & window contact sensor to my network. and after the interview was done, the device is responing N/A in Contact, Battery and Voltage. 
i have tried removing and adding 5 times, and it is not responing at all. 

FirmwareBuild: 20230109
FirmwareVersion: 2019www.
Model: MCCGQ12LM

### What did you expect to happen?

i expected at least a True False respond. 

### Zigbee2MQTT version

1.33.2 commit: unknown

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Skyconnect

### Debug log

Log from Adding Device to network:
2023-11-04 17:49:13Received Zigbee message from '0x54ef4410009258b0', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0xe0798dfffe110b42","zoneState":1}' from endpoint 1 with groupID 0
Debug 2023-11-04 17:49:13No converter available for 'MCCGQ12LM' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0xe0798dfffe110b42","zoneState":1}'
Info 2023-11-04 17:49:13Successfully interviewed '0x54ef4410009258b0', device has successfully been paired
Info 2023-11-04 17:49:13Device '0x54ef4410009258b0' is supported, identified as: Xiaomi Aqara T1 door & window contact sensor (MCCGQ12LM)
Info 2023-11-04 17:49:13MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara T1 door & window contact sensor","exposes":[{"access":1,"description":"Indicates if the contact is closed (= true) or open (= false)","label":"Contact","name":"contact","property":"contact","type":"binary","value_off":true,"value_on":false},{"access":1,"description":"Remaining battery in %, can take up to 24 hours before reported.","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Voltage of the battery in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"MCCGQ12LM","options":[],"supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"0x54ef4410009258b0","ieee_address":"0x54ef4410009258b0","status":"successful","supported":true},"type":"device_interview"}'

Open and closing the contact:
Debug 2023-11-04 18:00:30Received Zigbee message from '0x54ef4410009258b0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":33}' from endpoint 1 with groupID 0
Debug 2023-11-04 18:00:30No converter available for 'MCCGQ12LM' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":33}'
Debug 2023-11-04 18:00:31Received Zigbee message from '0x54ef4410009258b0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Debug 2023-11-04 18:00:31No converter available for 'MCCGQ12LM' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":32}'

bodek85, 2023-11-05:
exactly the same for me

Klapperman, 2023-11-11:
I've been mucking about trying to create a custom converter and custom device . The way the syntax looks for the default converter is that it's expecting a completely different message structure. When scanning the converters already created the only ones which are similar are a gas and smoke converter. For example:
`byun_smoke_true: {
        cluster: 'ssIasZone',
        type: ['commandStatusChangeNotification'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.endpoint.ID == 1 && msg.data['zonestatus'] == 33) {
                return {smoke: true};
            }
        },
    } as Fz.Converter,`

I'm hoping @Koenkk can help because my attempts to create a custom converter based on this and the available documentation are ending up with a non-working zigbee2mqtt instance which will only start again after removing my external converter :(.

Koenkk, 2023-11-11:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Klapperman, 2023-11-11:
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea
>
>     * save this as file next to `configuration.yaml` as [...]

Hi @Koenkk, this fixes it. After restarting the contact shows OPEN and CLOSED status correctly. Battery and Battery Voltage still are represented with a NULL though, but that's not too big of an issue.

Koenkk, 2023-11-11:
Great, regarding battery please try with https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea and wait 24 hours to see if battery is reported.

Klapperman, 2023-11-13:
> Great, regarding battery please try with https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea and wait 24 hours to see if battery is reported.

Battery is shown as a icon on the devices page of zigbee2mqtt but when you open de exposes page of the sensor it show:
`Contact
Indicates if the contact is closed (= true) or open (= false)                   Closed
Battery
Remaining battery in %, can take up to 24 hours before reported.      N/A%
Voltage
Voltage of the battery in millivolts                                                        N/AmV
Linkquality
Link quality (signal strength)                                                                 51lqi`

Koenkk, 2023-11-13:
Updated https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea , with this, make sure the configure succeeds by manually triggering it from the z2m frontend -> device -> yellow refresh icon, if it fails, wakeup the device right before doing so. Then wait 24 hours and see if battery % is reported now.

dusan-ivanco, 2023-11-14:
Now I have a different problem. After the "Availability Timeout" (in my case, it's 720), all MCCGQ12LM devices are in an offline state, but only those connected through the router. Two devices directly connected to the coordinator report normally. It doesn't matter which router they are connected to (I have Hornbach bulbs, Philips Hue, and Sonoff ZBDongle-E with router firmware in the network). The MCCGQ12LM status reports as okay even when it is already offline. So, if I open the door, it immediately switches to online and shows the correct "open" status. Similarly, when the door is open and after the availability timeout, when the MCCGQ12LM goes offline, if I close the door, it correctly shows the status change to "closed," and of course, switches back to online. MCCGQ12LM worked fine before firmware 20230109, so the issue is probably not with the routers. Thank you very much for your help.

Koenkk, 2023-11-14:
- Does the battery reporting work now?
- I expect 720 is too low, for end devices I recommend to use 1500 which means that the device should at least send one message every 25 hours.

Klapperman, 2023-11-15:
> * Does the battery reporting work now?

After applying the update, restarting zigbee2mqtt and then waking up the device and reconfiguring I get:

```
info  2023-11-15 06:07:23: Contact Sensor Brievenbus Buiten (0x54ef44100084ff12): MCCGQ12LM - Xiaomi Aqara T1 door & window contact sensor (EndDevice)
info  2023-11-15 06:07:23: Contact Sensor Brievenbus Binnen (0x54ef441000850010): MCCGQ12LM - Xiaomi Aqara T1 door & window contact sensor (EndDevice)
info  2023-11-15 06:07:25: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Buiten/availability', payload 'online'
info  2023-11-15 06:07:25: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen/availability', payload 'online'
info  2023-11-15 06:07:28: MQTT publish: topic 'homeassistant/binary_sensor/0x54ef44100084ff12/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x54ef44100084ff12"],"manufacturer":"Xiaomi","model":"Aqara T1 door & window contact sensor (MCCGQ12LM)","name":"Contact Sensor Brievenbus Buiten","sw_version":"2019\u0000www."},"device_class":"door","object_id":"contact_sensor_brievenbus_buiten_contact","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten","unique_id":"0x54ef44100084ff12_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
info  2023-11-15 06:07:28: MQTT publish: topic 'homeassistant/sensor/0x54ef44100084ff12/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x54ef44100084ff12"],"manufacturer":"Xiaomi","model":"Aqara T1 door & window contact sensor (MCCGQ12LM)","name":"Contact Sensor Brievenbus Buiten","sw_version":"2019\u0000www."},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"contact_sensor_brievenbus_buiten_battery","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten","unique_id":"0x54ef44100084ff12_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2023-11-15 06:07:28: MQTT publish: topic 'homeassistant/sensor/0x54ef44100084ff12/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x54ef44100084ff12"],"manufacturer":"Xiaomi","model":"Aqara T1 door & window contact sensor (MCCGQ12LM)","name":"Contact Sensor Brievenbus Buiten","sw_version":"2019\u0000www."},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","object_id":"contact_sensor_brievenbus_buiten_voltage","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten","unique_id":"0x54ef44100084ff12_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
info  2023-11-15 06:07:28: MQTT publish: topic 'homeassistant/sensor/0x54ef44100084ff12/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x54ef44100084ff12"],"manufacturer":"Xiaomi","model":"Aqara T1 door & window contact sensor (MCCGQ12LM)","name":"Contact Sensor Brievenbus Buiten","sw_version":"2019\u0000www."},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"contact_sensor_brievenbus_buiten_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten","unique_id":"0x54ef44100084ff12_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2023-11-15 06:07:28: MQTT publish: topic 'homeassistant/sensor/0x54ef44100084ff12/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x54ef44100084ff12"],"manufacturer":"Xiaomi","model":"Aqara T1 door & window contact sensor (MCCGQ12LM)","name":"Contact Sensor Brievenbus Buiten","sw_version":"2019\u0000www."},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"Update state","object_id":"contact_sensor_brievenbus_buiten_update_state","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten","unique_id":"0x54ef44100084ff12_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
info  2023-11-15 06:07:28: MQTT publish: topic 'homeassistant/binary_sensor/0x54ef44100084ff12/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Contact Sensor Brievenbus Buiten/availability"}],"availability_mode [... cut ...] ll}'
error 2023-11-15 06:11:54: Failed to configure 'Contact Sensor Brievenbus Binnen', attempt 3 (Error: Bind 0x54ef441000850010/1 genPowerCfg from '0x00124b00216003e1/1' failed (AREQ - ZDO - bindRsp after 10000ms)
```

The error `error 2023-11-15 06:11:54: Failed to configure 'Contact Sensor Brievenbus Binnen', attempt 3 (Error: Bind 0x54ef441000850010/1 genPowerCfg from '0x00124b00216003e1/1' failed (AREQ - ZDO - bindRsp after 10000ms)`  is now given each time one of the sensors is triggered.

Koenkk, 2023-11-15:
Thanks, so we can conclude that the device doesn't support `battery`, only `battery_low` seems to be supported which I've added to the exposes now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

dusan-ivanco, 2023-11-16:
> Thanks, so we can conclude that the device doesn't support `battery`, only `battery_low` seems to be supported which I've added to the exposes now.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

[image]

dusan-ivanco, 2023-11-16:
> * Does the battery reporting work now?
> * I expect 720 is too low, for end devices I recommend to use 1500 which means that the device should at least send one message every 25 hours.

"Okno: komora" is a device directly connected to the coordinator, thus correctly testing its "online" status. The "Okno: spálňa 1/2" devices are connected via a router and were opened 13 hours ago, so they are in an online state. Others that haven't been opened recently are in an offline state despite the passive timeout of 1500.
[image]

Klapperman, 2023-11-16:
Weird  @dusan-ivanco, maybe different HW revisions? I received mine 1 week ago? When I get back home I'll check serial numbers etc.. Maybe we can figure out what is different?

In the meantime, my devices behave the same with respect to the online/offline part. Without active usage they do not generate any status traffic like battery status updates by themselves within a 25 hour period. I've lengtened it to 48 hours and am waiting if that will change anything. But as I'm using mine for checking a mailbox they are not triggered daily or even sometimes weekly so the status could be useless with regard to low traffic availability usage.

dusan-ivanco, 2023-11-16:
> Weird  @dusan-ivanco, maybe different HW revisions? I received mine 1 week ago? When I get back home I'll check serial numbers etc.. Maybe we can figure out what is different?
> 
> 
> 
> In the meantime, my devices behave the same with respect to the online/offline part. Without active usage they do not generate any status traffic like battery status updates by themselves within a 25 hour period. I've lengtened it to 48 hours and am waiting if that will change anything. But as I'm using mine for checking a mailbox they are not triggered daily or even sometimes weekly so the availability status could be useless in situations with low or irregular usage. 

It's strange that devices directly connected to the coordinator show their online status correctly. According to my observation, they update their status every 5 hours even without activation (opening). Only devices connected through the router don't update their status without activation.

Klapperman, 2023-11-17:
@dusan-ivanco my device is a:
Aqara Door and Window Sensor T1
Sku: AS070GLW01
Serial nr: A00721/YH1CPH02443
Production Date: 07/2023

Klapperman, 2023-11-18:
@Koenkk my fourth sensor is now correctly reporting it's battery status. I hadn't installed this one, and after installing it this morning, and adding it to Zigbee2MQTT it's reporting it's status correctly. Device "0x54ef4410008501f0" is the newly added sensor. Although I did receive errors when adding it with the external converter it is showing all battery details. So maybe I need to first remove the already paired sensors and re-add them? A reconfigure still just gives errors and doesn't update the battery stats.

```
info  2023-11-18 10:08:38: Zigbee: allowing new devices to join.
info  2023-11-18 10:08:39: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"ccek4-1"}'
info  2023-11-18 10:08:53: Device '0x54ef4410008501f0' joined
info  2023-11-18 10:08:53: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410008501f0","ieee_address":"0x54ef4410008501f0"},"type":"device_joined"}'
info  2023-11-18 10:08:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x54ef4410008501f0"},"type":"device_connected"}'
info  2023-11-18 10:08:53: MQTT publish: topic 'zigbee2mqtt/0x54ef4410008501f0/availability', payload 'online'
info  2023-11-18 10:08:53: Starting interview of '0x54ef4410008501f0'
info  2023-11-18 10:08:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410008501f0","ieee_address":"0x54ef4410008501f0","status":"started"},"type":"device_interview"}'
info  2023-11-18 10:08:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x54ef4410008501f0"},"type":"pairing"}'
info  2023-11-18 10:08:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410008501f0","ieee_address":"0x54ef4410008501f0"},"type":"device_announce"}'
info  2023-11-18 10:08:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef4410008501f0"},"type":"device_announced"}'
info  2023-11-18 10:09:11: Successfully interviewed '0x54ef4410008501f0', device has successfully been paired
info  2023-11-18 10:09:11: Device '0x54ef4410008501f0' is supported, identified as: Xiaomi Aqara T1 door & window contact sensor (MCCGQ12LM)
info  2023-11-18 10:09:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara T1 door & window contact sensor","exposes":[{"access":1,"description":"Indicates if the contact is closed (= true) or open (= false)","label":"Contact","name":"contact","property":"contact","type":"binary","value_off":true,"value_on":false},{"access":1,"description":"Remaining battery in %, can take up to 24 hours before reported.","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Voltage of the battery in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"MCCGQ12LM","options":[],"supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"0x54ef4410008501f0","ieee_address":"0x54ef4410008501f0","status":"successful","supported":true},"type":"device_interview"}'
info  2023-11-18 10:09:11: Configuring '0x54ef4410008501f0'
info  2023-11-18 10:09:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Aqara T1 door & window contact sensor","friendly_name":"0x54ef4410008501f0","model":"MCCGQ12LM","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'
info  2023-11-18 10:09:11: MQTT publish: topic 'homeassistant/binary_sensor/0x54ef4410008501f0/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x54ef4410008501f0/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x54ef4410008501f0"],"manufacturer":"Xiaomi","model":"Aqara T1 door & window contact sensor (MCCGQ12LM)","name":"0x54ef4410008501f0","sw_version":"2019\u0000www."},"device_class":"door","object_id":"0x54ef4410008501f0_contact","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/0x54ef4410008501f0","unique_id":"0x54ef4410008501f0_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
info  2023-11-18 10:09:11: MQTT publish: topic 'homeassistant/sensor/0x54ef4410008501f0/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x54ef4410008501f0/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x54ef4410008501f0"],"manufacturer":"Xiaomi","model":"Aqara T1 door & window contact sensor (MCCGQ12LM)","name":"0x54ef4410008501f0","sw_version":"2019\u0000www."},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0x54ef4410008501f0_battery","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef4410008501f0","unique_id":"0x54ef4410008501f0_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
[... cut ...]
info  2023-11-18 10:14:59: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Buiten', payload '{"battery":null,"battery_low":false,"contact":true,"linkquality":72,"tamper":false,"update":{"installed_version":30,"latest_version":30,"state":"idle"},"update_available":false,"voltage":null}'
info  2023-11-18 10:15:00: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen', payload '{"battery":null,"battery_low":false,"contact":true,"linkquality":24,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
```

Koenkk, 2023-11-18:
@Klapperman it's only reporting `battery_low`, right?

Klapperman, 2023-11-18:
Nee de nieuwe toont alle batterij statistieken.

Alleen de eerder gekoppelde apparaten met de oudere externe converter
script actief doen het nog steeds niet. Wil die morgen verwijderen en
opnieuw toevoegen.

Koenkk, 2023-11-19:
Can you try with https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea + the re-pairing?

Klapperman, 2023-11-19:
> Can you try with https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea + the re-pairing?

Reconfigure still does not work, although the error is simpler now, it's just "cannot be configured"

```
debug 2023-11-19 08:25:13Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Contact Sensor Brievenbus Buiten","transaction":"kgl36-1"}'
info 2023-11-19 08:25:13MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Contact Sensor Brievenbus Buiten"},"error":"Device 'Contact Sensor Brievenbus Buiten' cannot be configured","status":"error","transaction":"kgl36-1"}'
debug 2023-11-19 08:25:21Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Contact Sensor Brievenbus Binnen","transaction":"kgl36-2"}'
info 2023-11-19 08:25:21MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Contact Sensor Brievenbus Binnen"},"error":"Device 'Contact Sensor Brievenbus Binnen' cannot be configured","status":"error","transaction":"kgl36-2"}'
```

Then tried repairing it, first with a soft remove but that gave an error. Then with a forced remove. Device then re-added itself but battery still doesn't work, maybe need to wait 24 hours.

```
debug 2023-11-19 08:28:09: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":false,"id":"Contact Sensor Brievenbus Binnen","transaction":"kgl36-3"}'
info  2023-11-19 08:28:09: Removing device 'Contact Sensor Brievenbus Binnen' (block: false, force: false)
error 2023-11-19 08:28:19: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'Contact Sensor Brievenbus Binnen' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)'
debug 2023-11-19 08:28:19: Error: Failed to remove device 'Contact Sensor Brievenbus Binnen' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
    at Bridge.removeEntity (/app/lib/extension/bridge.ts:567:19)
    at Bridge.onMQTTMessage (/app/lib/extension/bridge.ts:127:34)
info  2023-11-19 08:28:19: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{},"error":"Failed to remove device 'Contact Sensor Brievenbus Binnen' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)","status":"error","transaction":"kgl36-3"}'
debug 2023-11-19 08:28:26: Received Zigbee message from 'Contact Sensor Brievenbus Binnen', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":33}' from endpoint 1 with groupID 0
info  2023-11-19 08:28:26: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen', payload '{"battery":null,"battery_low":false,"contact":false,"linkquality":12,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
debug 2023-11-19 08:28:28: Received Zigbee message from 'Contact Sensor Brievenbus Binnen', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
info  2023-11-19 08:28:28: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen', payload '{"battery":null,"battery_low":false,"contact":true,"linkquality":12,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
debug 2023-11-19 08:28:56: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":false,"id":"Contact Sensor Brievenbus Binnen","transaction":"kgl36-4"}'
info  2023-11-19 08:28:56: Removing device 'Contact Sensor Brievenbus Binnen' (block: false, force: false)
error 2023-11-19 08:29:06: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'Contact Sensor Brievenbus Binnen' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)'
debug 2023-11-19 08:29:06: Error: Failed to remove device 'Contact Sensor Brievenbus Binnen' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
    at Bridge.removeEntity (/app/lib/extension/bridge.ts:567:19)
    at Bridge.onMQTTMessage (/app/lib/extension/bridge.ts:127:34)
info  2023-11-19 08:29:06: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{},"error":"Failed to remove device 'Contact Sensor Brievenbus Binnen' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)","status":"error","transaction":"kgl36-4"}'
debug 2023-11-19 08:33:01: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Contact Sensor Brievenbus Binnen","transaction":"5vf0c-4"}'
info  2023-11-19 08:33:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Contact Sensor Brievenbus Binnen"},"error":"Device 'Contact Sensor Brievenbus Binnen' cannot be configured","status":"error","transaction":"5vf0c-4"}'
debug 2023-11-19 08:34:06: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":true,"id":"Contact Sensor Brievenbus Binnen","transaction":"5vf0c-5"}'
info  2023-11-19 08:34:06: Removing device 'Contact Sensor Brievenbus Binnen' (block: false, force: true)
debug 2023-11-19 08:34:06: Clearing Home Assistant discovery topic for 'Contact Sensor Brievenbus Binnen'
[... cut ...]
info  2023-11-19 08:38:35: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen', payload '{"battery":null,"battery_low":false,"contact":false,"linkquality":81,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
info  2023-11-19 08:38:36: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen', payload '{"battery":null,"battery_low":false,"contact":true,"linkquality":84,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
```

Koenkk, 2023-11-19:
Configure is not needed for no anymore (that's why it gives an error). Let me know if the battery appears in 24 hours from now.

Klapperman, 2023-11-20:
> Configure is not needed for no anymore (that's why it gives an error). Let me know if the battery appears in 24 hours from now.

Unfortunately no, the already adopted sensors before the final converter version still do not report battery details. They still show battery_low, but all battery details are N/A. The device adopted with the final converter does show it's details correctly.

Koenkk, 2023-11-20:
with final converter you mean https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea , everything works with that?

Klapperman, 2023-11-20:
> with final converter you mean https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea , everything works with that?

Only the last added device shows battery details. The earlier ones still work with all converters including the final one but won't report battery details as the logs show. All of them work as a contact sensor which is the primary concern. As shown I've tried reinterview and readding the early adds but that does not make a difference. They reconnect fine but no battery details. For now not a problem and others I this thread seem to have no issues except for the devices not reporting status without usage which I also have. As everything else works I'm fine with the current result.

Koenkk, 2023-11-21:
With what converter did the battery results show? Was it this one? https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea/df7c7620e447a8742454232e285156fc63ebc04b

Klapperman, 2023-11-23:
> With what converter did the battery results show? Was it this one? https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea/df7c7620e447a8742454232e285156fc63ebc04b

Yes, that was the active version when I onboarded the final sensor which is showing it's battery details succesfully.

dusan-ivanco, 2023-11-23:
@Koenkk - Is it somehow solvable?

[image]

> "Okno: komora" is a device directly connected to the coordinator, thus correctly testing its "online" status. The "Okno: spálňa 1/2" devices are connected via a router and were opened 13 hours ago, [...]

Koenkk, 2023-11-23:
@Klapperman can you try if battery reporting works and configure succeeds with [this converter](https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea)?

@dusan-ivanco I expect this is solved once battery reporting works properly (since you will receive a message regularly then)

Klapperman, 2023-11-25:
> @Klapperman can you try if battery reporting works and configure succeeds with [this converter](https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea)?
> 
> @dusan-ivanco I expect this is solved once battery reporting works properly (since you will receive a message regularly then)

@Koenkk no joy unfortunately. First tried reinterview but that gave immediate error. Then removed device from Zigbee, first soft then forced. Afterwards held reset button for 10 seconds, then re-added device. For now no battery stats (and the one joined with later converter showed it's battery details instantly with every update. I also retried with that succesvol converter, but no matter how I remove them from the zigbee network they seemingly always join back-up without working battery updates. Maybe if I removed the battery and waited a while, but as the setup does work I'm not that keen on breaking it again. Plus my luck with Aqara devices and battery updates has been somewhat hit & miss in the past.

```
debug 2023-11-25 09:44:07: Received Zigbee message from 'Contact Sensor Brievenbus Binnen', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":33}' from endpoint 1 with groupID 0
info  2023-11-25 09:44:07: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen', payload '{"battery":null,"battery_low":false,"contact":false,"linkquality":84,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
debug 2023-11-25 09:44:07: Received Zigbee message from 'Contact Sensor Brievenbus Binnen', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
info  2023-11-25 09:44:07: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen', payload '{"battery":null,"battery_low":false,"contact":true,"linkquality":84,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
debug 2023-11-25 10:49:51: Passive device 'Contact Sensor Brievenbus Buiten' was last seen '25.00' hours ago.
info  2023-11-25 10:49:51: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Buiten/availability', payload 'offline'
debug 2023-11-25 12:34:57: Received Zigbee message from 'Contact Sensor Brievenbus Binnen', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":33}' from endpoint 1 with groupID 0
info  2023-11-25 12:34:57: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen', payload '{"battery":null,"battery_low":false,"contact":false,"linkquality":78,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
debug 2023-11-25 12:34:58: Received Zigbee message from 'Contact Sensor Brievenbus Binnen', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
info  2023-11-25 12:34:58: MQTT publish: topic 'zigbee2mqtt/Contact Sensor Brievenbus Binnen', payload '{"battery":null,"battery_low":false,"contact":true,"linkquality":78,"tamper":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
debug 2023-11-25 15:26:52: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":false,"id":"Contact Sensor Brievenbus Binnen","transaction":"ubey4-1"}'
info  2023-11-25 15:26:52: Removing device 'Contact Sensor Brievenbus Binnen' (block: false, force: false)
error 2023-11-25 15:27:02: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'Contact Sensor Brievenbus Binnen' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)'
debug 2023-11-25 15:27:02: Error: Failed to remove device 'Contact Sensor Brievenbus Binnen' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
    at Bridge.removeEntity (/app/lib/extension/bridge.ts:567:19)
    at Bridge.onMQTTMessage (/app/lib/extension/bridge.ts:127:34)
info  2023-11-25 15:27:02: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{},"error":"Failed to remove device 'Contact Sensor Brievenbus Binnen' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)","status":"error","transaction":"ubey4-1"}'
debug 2023-11-25 15:27:16: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":true,"id":"Contact Sensor Brievenbus Binnen","transaction":"ubey4-2"}'
info  2023-11-25 15:27:16: Removing device 'Contact Sensor Brievenbus Binnen' (block: false, force: true)
debug 2023-11-25 15:27:16: Clearing Home Assistant discovery topic for 'Contact Sensor Brievenbus Binnen'
info  2023-11-25 15:27:16: MQTT publish: topic 'homeassistant/binary_sensor/0x54ef441000850010/contact/config', payload 'null'
info  2023-11-25 15:27:16: MQTT publish: topic 'homeassistant/sensor/0x54ef441000850010/battery/config', payload 'null'
info  2023-11-25 15:27:16: MQTT publish: topic 'homeassistant/sensor/0x54ef441000850010/voltage/config', payload 'null'
info  2023-11-25 15:27:16: MQTT publish: topic 'homeassistant/sensor/0x54ef441000850010/linkquality/config', payload 'null'
[... cut ...] te":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/Contact Sensor Brievenbus Binnen","name":null,"object_id":"contact_sensor_brievenbus_binnen","origin":{"name":"Zigbee2MQTT","sw":"1.33.2","url":"https://www.zigbee2mqtt.io"},"payload_install":"{\"id\": \"0x54ef441000850010\"}","state_topic":"zigbee2mqtt/Contact Sensor Brievenbus Binnen","unique_id":"0x54ef441000850010_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'

```

christopherferreira3, 2023-12-01:
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea
>
> * save this as file next to `configuration.yaml` as [...]

I managed to make it work with this one. I had 10 sensors to pair and 3 of them were struggling to get the battery status even with custom converter. Force removing it and pairing again a few times in a row made it work.

Klapperman, 2023-12-02:
> I tried https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea
> as external convertor but I still get this error:
>
> /app/node_modules/zigbee-herdsman-converters/index.js:172
>         if (converter.options) {
>                       ^
> TypeError: Cannot read properties of undefined (reading 'options')
>     at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:172:23)
>     at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
>     at new Controller (/app/lib/controller.ts:84:58)
>     at start (/app/index.js:106:18)

That issue is new and happens with the latest update. If you roll back one
version you do not get that error.

dusan-ivanco, 2024-01-22:
> @Koenkk - Is it somehow solvable?
> 
> ![Snímka obrazovky 2023-11-23 o 12 53 47](https://private-user-images.githubusercontent.com/4436584/285198550-8707641c-2f9c-4b0f-a775-5e02719d3aca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5MzE3OTMsIm5iZiI6MTcwNTkzMTQ5MywicGF0aCI6Ii80NDM2NTg0LzI4NTE5ODU1MC04NzA3NjQxYy0yZjljLTRiMGYtYTc3NS01ZTAyNzE5ZDNhY2EucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMjJUMTM1MTMzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDdiYmJjMmQwMzY3MmI5YTkzZWMzMzdiNmIxN2YxOWVlMjFmMzNmMzY4NWRmMzA5ZmFlNzkwZGM0OGJkZmYwNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.WDW5KpFgAKoPpFUGt0RaiVvuVi07JMz4fLmxJmFn8yA)
> 
> > > * Does the battery reporting work now?
> > > * I expect 720 is too low, for end devices I recommend to use 1500 which means that the device should at least send one message every 25 hours.
> > 
> > 
> > "Okno: komora" is a device directly connected to the coordinator, thus correctly testing its "online" status. The "Okno: spálňa 1/2" devices are connected via a router and were opened 13 hours ago, so they are in an online state. Others that haven't been opened recently are in an offline state despite the passive timeout of 1500. [image]

Sorry - any progress with this issue?

christopherferreira3, 2024-01-23:
> Sorry - any progress with this issue?

I think they go offline because they stop reporting to save up battery. As soon as their status changes (like opening a door) they will, don't worry.

leroyloren, 2024-01-28:
i have a few of these sensors new and i still cant get their battery status, tried ext. converters in this thread, reconfiguration, re-pairing, etc. Nothing helps.

[image]
[image]

amaciuc, 2024-01-29:
> i have a few of these sensors new and i still cant get their battery status, tried ext. converters in this thread, reconfiguration, re-pairing, etc. Nothing helps.
> 
> ![obrazek](https://private-user-images.githubusercontent.com/57643470/300289161-f5d70757-9407-4303-ab5f-122ea51d97ef.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1MTY3MTMsIm5iZiI6MTcwNjUxNjQxMywicGF0aCI6Ii81NzY0MzQ3MC8zMDAyODkxNjEtZjVkNzA3NTctOTQwNy00MzAzLWFiNWYtMTIyZWE1MWQ5N2VmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI5VDA4MjAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdlNDAwM2VjZTFiZjMyOGQzNGRhMWUwNDcxNTJkODMwMGM2OTA0ZmIxMTI2NjU3ZGYxNzNlYzI4ZmUxODk0YjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tSHsIZCgryYEUMx6sooAh_9XUVZKblagSSCCrMIBIm4) ![obrazek](https://private-user-images.githubusercontent.com/57643470/300289182-28eea32c-3d03-45b0-99e1-ee5b0e8e010a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1MTY3MTMsIm5iZiI6MTcwNjUxNjQxMywicGF0aCI6Ii81NzY0MzQ3MC8zMDAyODkxODItMjhlZWEzMmMtM2QwMy00NWIwLTk5ZTEtZWU1YjBlOGUwMTBhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI5VDA4MjAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1YmE1YzAzNzllZmQxOWMwM2E4MmFhNzBkM2FiMDUzODYzZDc0MThiYTFkNjA2MmU0ZDhmMDgxZTU2Y2RiYTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hM1jL5Hfmzm5ys7ayZJgkZZ1yo6KcA9Rd4vsBgyNqPM)

Same here :disappointed:

amaciuc, 2024-01-29:
The voltage is successfully read

[image]
 
and I suppose this line should convert the voltage to percentage 

[image]

but it doesn't. The battery shows `N/A` after a while.

dusan-ivanco, 2024-01-29:
I believe the problem is with the current firmware (https://github.com/Koenkk/zigbee-OTA/pull/268). Devices that have not yet been updated are working fine for me. How do I downgrade the firmware?

amaciuc, 2024-01-31:
@Koenkk can you please help us with this situation ?

If you have some extra ideas what we should do, please let us know.

Koenkk, 2024-01-31:
Updated https://gist.github.com/Koenkk/20a6e2592ba7444d3d51a9d393defcea
- Make sure that configure succeeds (yellow refresh button), if not try to wake-up the device right before clicking it
- Wait 24 hours and see if battery % is reported now.

amaciuc, 2024-02-01:
I tried to load it and I got an error at restarting

[image]

Is it normal? The battery has not updated yet.

amaciuc, 2024-02-01:
@Koenkk 

I succeeded to load another js file inspired from [here](https://gist.github.com/Koenkk/bedcf4cc4f3c63058349151571e7a82a), it looks like that:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const m = require('zigbee-herdsman-converters/lib/modernExtend');
const libColor = require('zigbee-herdsman-converters/lib/color');
const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.magnet.agl02'],
    model: 'MCCGQ12LM',
    vendor: 'Xiaomi',
    description: 'Aqara T1 door & window contact sensor',
    // fromZigbee: [fz.xiaomi_contact, fz.aqara_opple, fz.ias_contact_alarm_1, fz.battery],
    fromZigbee: [xiaomi.fromZigbee.xiaomi_basic, fz.xiaomi_contact, xiaomi.fromZigbee.aqara_opple, fz.ias_contact_alarm_1],
    toZigbee: [],
    meta: {battery: {voltageToPercentage: '3V_2850_3000'}},
    exposes: [e.contact(), e.battery(), e.battery_voltage()],
    // ota: ota.zigbeeOTA,
    // configure: async (device, coordinatorEndpoint, logger) => {
    //     const endpoint = device.getEndpoint(1);
    //     await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
    //     await reporting.batteryVoltage(endpoint);
    // },
};

module.exports = definition;

```
That way I avoided that error with undefined properties and the exposes of magnetic sensor looks like that:

[image]

What do you think? Does it look okay for you?

amaciuc, 2024-02-01:
I have noticed that in the latest version of `hardsman-converters`, xiaomi becomes lumi. I got rid of converter before updating Z2M to `1.35.2` and still `xiaomi` contact sensor doesn't expose the battery level. I dug a little into this [lumi.ts file](https://github.com/Koenkk/zigbee-herdsman-converters/blob/aebfa997703e22019e460e22bc0a6693186a63a9/src/devices/lumi.ts#L1913) and I found that for `magnet.agl02` you didn't define a `meta` to convert voltage into battery percentage (like in case of `flood.agl02`). Therefore, I import once again an external converter with a `meta` define like that:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const m = require('zigbee-herdsman-converters/lib/modernExtend');
const libColor = require('zigbee-herdsman-converters/lib/color');
// const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');
const lumi = require('zigbee-herdsman-converters/lib/lumi');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.magnet.agl02'],
    model: 'MCCGQ12LM',
    vendor: 'Xiaomi',
    description: 'Aqara T1 door & window contact sensor',
    whiteLabel: [{vendor: 'Aqara', model: 'DW-S03D'}],
    fromZigbee: [lumi.fromZigbee.lumi_specific, fz.lumi_contact, fz.ias_contact_alarm_1],
    toZigbee: [],
    meta: {battery: {voltageToPercentage: '3V_2850_3000'}},
    exposes: [e.contact(), e.battery(), e.battery_voltage()],
    // ota: ota.zigbeeOTA,
    // configure: async (device, coordinatorEndpoint, logger) => {
    //     const endpoint = device.getEndpoint(1);
    //     await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
    //     await reporting.batteryVoltage(endpoint);
    // },
};

module.exports = definition;
```

Now it works okay.

Koenkk, 2024-02-01:
@amaciuc integrated the fix, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

amaciuc, 2024-02-07:
This has been fixed officially in latest version on Z2M `1.53.3`.

@Koenkk I think you can close the issue

tretabyte, 2024-02-07:
Hi,
I updated to 1.53.3, but I still can't get the battery status, still shows n/a

amaciuc, 2024-02-07:
> Hi, I updated to 1.53.3, but I still can't get the battery status, still shows n/a

Din you try to re-config / rejoin the sensor ?

tretabyte, 2024-02-07:
> > Hi, I updated to 1.53.3, but I still can't get the battery status, still shows n/a
> 
> Din you try to re-config / rejoin the sensor ?

After a couple of restarts, 1 full reboot on HA host and 2 or 3 reconfigures (the bug with the UI messes things a lot), I can report I'm finally getting the battery %

amaciuc, 2024-02-07:
About UI bug ... it is not so frustrating :smile: 
Try to press the icon with light/dark mode. it should be okay after that.

tretabyte, 2024-02-07:
> About UI bug ... it is not so frustrating 😄 Try to press the icon with light/dark mode. it should be okay after that.

Lifesaver!!!

amaciuc, 2024-02-21:
> With z2m version [1.35.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.3) I still do not get values in the UI
> 
> How it looks: <img alt="Screenshot 2024-02-21 at 16 52 05" width="307" src="https://private-user-images.githubusercontent.com/731038/306666452-7ec039a8-bf24-4fed-b61d-d93de264ce26.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1Mjg0ODksIm5iZiI6MTcwODUyODE4OSwicGF0aCI6Ii83MzEwMzgvMzA2NjY2NDUyLTdlYzAzOWE4LWJmMjQtNGZlZC1iNjFkLWQ5M2RlMjY0Y2UyNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMVQxNTA5NDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNDk0MWNkZDNlMzBiMTU3YzdlYzMyZDFkM2YyYTFiMjg5ODI2Mjc1MDljNDdjOTNlMTg5M2Q0MDI0YTQzNjNiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.WcWjPO99yJtDM4BO2RodFc8nqlNusFWmyGCtxRTg84o">
> 
> If I go under `device -> state` I can see:
> 
> ```
> {
>     "battery_low": false,
>     "contact": false,
>     "linkquality": 80,
>     "tamper": false
> }
> ```
> 
> So the information exists (at least low_battery) ... but is not correctly displayed.
> 
> I'm missing something, or is this bug still valid?

Din you try to reconfigure / rejoin the sensor ?

pauldcomanici, 2024-02-22:
I tried reconfigure, I will also try to do rejoin, but there are 2 issues here that I consider that are not related to reconfigure/rejoin:
1. N/A is shown in the UI while `"battery_low": false,` is received in the object ... I consider this is a bug on the UI by not using the right properties.
2. Battery entity from HA is with status unavailable ... I consider that the issue is caused by the misconfiguration from z2m on how data is sent to HA

However, those are my assumptions based on my thinking that things should work.

Gloomyeye, 2024-02-28:
I try to rejoin or reconnect a new aqara T1 contact sensor multiple times, but I can not get any values except the link quality. I try to hit the sensor button during connection process, too. But no success.

pbrunier, 2024-07-03:
Bought 4 of the Aqara T1 Door and Window sensors.
The magnet sensor works fine but Battery doesn't respond.
Zigbee2MQTT version 1.39.0 integrated in HASS
Zigbee stick: Sonoff -- zStack3x0

Device state:
{
    "battery_low": false,
    "contact": true,
    "linkquality": 96,
    "tamper": false,
    "battery": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null,
    "voltage": null
}

[image]
[image]

Pls ask if I need to send more info.

Gloomyeye, 2024-07-03:
I have the same issue with two sensors. Remaining battery is null / not available.

theSC, 2024-11-13:
I am having this exact issue with three brand new sensors.  I have removed and readded them multiple times.  I have restarted zigbee2mqtt multiple times as well.

Some data shows but never the "contact" part which is the one that really matters lol

theSC, 2024-11-26:
Is this issue being worked on?  Is there anything I can do to help?

theSC, 2024-12-17:
Beuller?

theSC, 2025-02-06:
No matter what I try. No matter how many times I re-add the sensors (three of them) I only ever get N/A as the state for Contact.  Obviously unusable that way. 

Can anyone please help me with this?

closed 2025-04-15 by github-actions[bot]