Koenkk/zigbee2mqtt #19560: [New device support]: Wiser Odace réf: S520559

squal32, 2023-11-04:
### Link

https://www.se.com/fr/fr/product/S520559/wiser-odace-prise-2p+t-connect%C3%A9e-16a-zigbee-blanc/

### Database entry

"ieee": "94:de:b8:ff:fe:5f:e1:ff",     "nwk": 7779,     "status": 2,     "lqi": "164",     "rssi": "-59",     "last_seen": "1699117137.368798",     "node_descriptor": "01:40:8e:5e:10:52:52:00:00:2c:52:00:00",     "endpoints": [         {             "id": 6,             "status": 1,             "device_type": 9,             "profile_id": 260,             "manufacturer": "Schneider Electric",             "model": "SOCKET\/OUTLET\/1",             "output_clusters": [

### Comments

je ne sais pas ou récupéré les entrée dans la base de données, j'ai copier toute les information brut:
"ieee": "94:de:b8:ff:fe:5f:e1:ff",
    "nwk": 7779,
    "status": 2,
    "lqi": "164",
    "rssi": "-59",
    "last_seen": "1699117137.368798",
    "node_descriptor": "01:40:8e:5e:10:52:52:00:00:2c:52:00:00",
    "endpoints": [
        {
            "id": 6,
            "status": 1,
            "device_type": 9,
            "profile_id": 260,
            "manufacturer": "Schneider Electric",
            "model": "SOCKET\/OUTLET\/1",
            "output_clusters": [
                {
                    "id": 25,
                    "name": "Ota",
                    "attributes": []
                }
            ],
            "input_clusters": [
                {
                    "id": 0,
                    "name": "Basic",
                    "attributes": [
                        {
                            "id": 0,
                            "name": "zcl_version",
                            "value": 3
                        },
                        {
                            "id": 1,
                            "name": "app_version",
                            "value": 2
                        },
                        {
                            "id": 2,
                            "name": "stack_version",
                            "value": 6
                        },
                        {
                            "id": 3,
                            "name": "hw_version",
                            "value": 1
                        },
                        {
                            "id": 4,
                            "name": "manufacturer",
                            "value": "Schneider Electric"
                        },
                        {
                            "id": 5,
                            "name": "model",
                            "value": "SOCKET\/OUTLET\/1"
                        },
                        {
                            "id": 6,
                            "name": "date_code",
                            "value": ""
                        },
                        {
                            "id": 7,
                            "name": "power_source",
                            "value": 1
                        },
                        {
                            "id": 16384,
                            "name": "sw_build_id",
                            "value": "002.009.001 R"
                        }
                    ]
                },
                {
                    "id": 3,
                    "name": "Identify",
                    "attributes": []
                },
                {
                    "id": 4,
                    "name": "Groups",
                    "attributes": []
                },
                {
                    "id": 5,
                    "name": "Scenes",
                    "attributes": [
                        {
                            "id": 1,
                            "name": "current_scene",
                            "value": 0
                        },
                        {
                            "id": 2,
                            "name": "current_group",
                            "value": 0
                        },
                        {
                            "id": 3,
                            "name": "scene_valid",
                            "value": 0
                        }
                    ]
                },
                {
                    "id": 6,
                    "name": "On\/Off",
                    "attributes": [
                        {
                            "id": 0,
                            "name": "on_off",
                            "value": 1
                        }
                    ]
                },
                {
                    "id": 1794,
                    "name": "Metering",
                    "attributes": [
                        {
                            "id": 0,
                            "name": "current_summ_delivered",
                            "value": 1
                        },
                        {
                            "id": 512,
                            "name": "status",
                            "value": 0
                        },
                        {
                            "id": 516,
                            "name": "extended_status",
                            "value": 0
                        },
                        {
                            "id": 519,
                            "name": "iambient_consumption_indicator",
                            "value": 0
                        },
                        {
                            "id": 1024,
                            "name": "instantaneous_demand",
                            "value": 0
                        }
                    ]
                },
                {
                    "id": 1800,
                    "name": "DeviceManagement",
                    "attributes": []
                },
                {
                    "id": 2820,
                    "name": "Electrical Measurement",
                    "attributes": [
                        {
                            "id": 1285,
                            "name": "rms_voltage",
                            "value": 243
                        },
                        {
                            "id": 1288,
                            "name": "rms_current",
                            "value": 0
                        }
                    ]
                },
                {
                    "id": 2821,
                    "name": "Diagnostic",
                    "attributes": []
                },
                {
                    "id": 64516,
                    "name": "Manufacturer Specific",
                    "attributes": []
                }
            ]
        },
        {
            "id": 242,
            "status": 1,
            "device_type": 97,
            "profile_id": 41440,
            "manufacturer": null,
            "model": null,
            "output_clusters": [
                {
                    "id": 33,
                    "name": "GreenPowerProxy",
                    "attributes": []
                }
            ],
            "input_clusters": []
        }
    ],
    "signature": {
        "manufacturer": "Schneider Electric",
        "model": "SOCKET\/OUTLET\/1",
        "node_desc": {
            "logical_type": 1,
            "complex_descriptor_available": 0,
            "user_descriptor_available": 0,
            "reserved": 0,
            "aps_flags": 0,
            "frequency_band": 8,
            "mac_capability_flags": 142,
            "manufacturer_code": 4190,
            "maximum_buffer_size": 82,
            "maximum_incoming_transfer_size": 82,
            "server_mask": 11264,
            "maximum_outgoing_transfer_size": 82,
            "descriptor_capability_field": 0
        },
        "endpoints": {
            "6": {
                "profile_id": 260,
                "device_type": 9,
                "input_clusters": [
                    0,
                    3,
                    4,
                    5,
                    6,
                    1794,
                    1800,
                    2820,
                    2821,
                    64516
                ],
                "output_clusters": [
                    25
                ]
            },
            "242": {
                "profile_id": 41440,
                "device_type": 97,
                "input_clusters": [],
                "output_clusters": [
                    33
                ]
            }
        }
    },
    "class": "zigpy.device"
}

devildant, 2023-11-04:
it is already supported I have 4 at home, it is referenced under: https://www.zigbee2mqtt.io/devices/EKO09716.html

Tonio16, 2023-11-09:
`"class": "zigpy.device`

Tu sembles mélanger les plugin... zigpy ce n’est pas zigbee2mqtt ( ou jeezigbee, ou zigbeelinker).

Antoine

Minims, 2024-02-28:
This device support LED indicator management.

How can i add this feature to my S520559 device ? Thx

```
Can I change when the LED Indicator is on for my Wiser switch? (Zigbee and Bluetooth)
This can be set in the App. You will need to tap on the device that you wish to change this for and then from the specific device page you can set this to - Always On, Always Off, Consistent with Load (On if the Load is on) or Reverse with Load (On if the Load is off). Reverse with Load is default setting.
```

closed 2024-09-26 by github-actions[bot]