Koenkk/zigbee2mqtt #19568: [Feature request]: Vesternet VES-ZB-DIM-004 add specify parameter

devildant, 2023-11-04:
### Is your feature request related to a problem? Please describe

Good morning,

I don't know if it's due to a more recent firmware or not, but the function allowing you to define the lighting level at power on no longer works with the min set button (2 short presses, although visually this seems to be taken into account by the module, this is not affected physically)

on the other hand the information is available via the payload received from the module:
`"level_config":{"on_level":"previous"}`

complete payload : 
`'{"brightness":1,"current":0.02,"energy":6.11,"level_config":{"on_level":1},"linkquality":117,"power":0,"power_on_behavior":"previous","state":"OFF","voltage":2.4}'`

During my tests I noticed that the possible values ​​are:
1 to 255, and the "previous" value

https://www.zigbee2mqtt.io/devices/VES-ZB-DIM-004.html

### Describe the solution you'd like

add this setting inside Settings (specific)

### Describe alternatives you've considered

no

### Additional context

no

devildant, 2023-11-10:
Hi @Koenkk, 
i modify with external file

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	fingerprint: [{modelID: 'HK-SL-DIM-A', softwareBuildID: '2.5.3_r52'}, {modelID: 'HK-SL-DIM-A', softwareBuildID: '2.9.2_r54'}],
        model: 'VES-ZB-DIM-004',
        vendor: 'Vesternet',
        description: 'Zigbee dimmer',
        fromZigbee: extend.light_onoff_brightness().fromZigbee
            .concat([fz.electrical_measurement, fz.metering, fz.ignore_genOta, fz.level_config]),
        toZigbee: extend.light_onoff_brightness().toZigbee.concat([tz.power_on_behavior, tz.level_config]),
        exposes: [e.light_brightness(), e.power(), e.voltage(), e.current(), e.energy(), e.power_on_behavior(['off', 'on', 'previous']),
		 e.composite('level_config', 'level_config', ea.ALL)
                .withFeature(e.numeric('on_level', ea.ALL)
                    .withValueMin(1).withValueMax(254)
                    .withPreset('previous', 255, 'Use previous value')
                    .withDescription('Specifies the level that shall be applied, when an on/toggle command causes the light to turn on.'))
                ],
        whiteLabel: [{vendor: 'Sunricher', model: 'SR-ZG9040A'}],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint);
            await reporting.brightness(endpoint);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.activePower(endpoint);
            await reporting.rmsCurrent(endpoint, {min: 10, change: 10});
            await reporting.rmsVoltage(endpoint, {min: 10});
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.currentSummDelivered(endpoint);
        },
		
}
module.exports = definition;
```

but 2 display problem:

1) when the previous status is selected (value 255, but 'previous' in the mqtt message), after a refresh, previous is not selected and the value is empty
[image]

2) the value when it is modified by hand, via the mqtt service, it is not updated, you have to refresh or change tab (the refresh button does nothing)

Did I do something wrong in my implementation?

best regards

devildant, 2023-11-10:
also how to like level_config.on_level to number entity ? 

[image]

Koenkk, 2023-11-10:
Can you submit a PR for the change and create an issue in the frontend regarding the frontend bug? https://github.com/nurikk/zigbee2mqtt-frontend/issues

devildant, 2023-11-10:
> Can you submit a PR for the change and create an issue in the frontend regarding the frontend bug? https://github.com/nurikk/zigbee2mqtt-frontend/issues

Ok i will do that :) 
And for my last question (for have an entity with type number for the on_level)? It's not frontend trouble

Koenkk, 2023-11-10:
Home Assistant doesn't support these complex `composite` types, therefore you can only set them via the z2m frontend.

devildant, 2023-11-10:
> Home Assistant doesn't support these complex `composite` types, therefore you can only set them via the z2m frontend.

Hoo ok don't know that, I'm new to home automation and home assistant :) 
Thx for your reply i will create a PR for my modification on zigbee-herdsman-converters, and i will create the issue front end :) 

Thx again

closed 2024-06-08 by github-actions[bot]