Koenkk/zigbee2mqtt #19569: Interview failed

jessegabriels, 2023-11-04:
### Is your feature request related to a problem? Please describe

Hi, 
I am using a Raspberry Pi 3B with a Sonoff Zigbee 3.0 USB Dongle Plus. I have followed all the instructions on how to set everything up. When I am in the UI and a click on "Permit Join", my Develco AQSZB-110 pairs but it keeps failing the interview. I have tried all the tips on the FAQ but it won't work.

### Describe the solution you'd like

Is there someone willing to help me to get this right? I have been working on this for many hours now.

### Describe alternatives you've considered

I updated the firmware on my Zigbee dongle. I also tried it with a USB extention cable. Pairing it right next to the dongle. Tried pairing the device multiple times. Tried pressing the button on the sensor every 2 seconds with the interview.

### Additional context

[image]

fireheadman, 2023-11-05:
I had an issue like this... I power cycled the device and went through a 2nd reset and it corrected itself

cellerich, 2023-11-06:
Can you check if your problem may relate to this issue:
- [18834](https://github.com/Koenkk/zigbee2mqtt/issues/18834)

If your Zigbee2mqtt is restarting when you try to pair a new device, it might be this.
Unfortunately no solution to this yet.

jessegabriels, 2023-11-06:
My Zigbee2mqtt isn't crashing. The interview just fails. Nomatter what I do.

cellerich, 2023-11-06:
Take a look in the log file while you are pairing your device. Maybe there is a hint there.
Other questions:
- how did you install z2m? (native on the RPi or inside Home Assistant as a add-on)
- what Sonoff Dongle you have? (Type P or Type E)

jessegabriels, 2023-11-06:
Here is my log:

pi@pi-b300:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.33.2 start
> node index.js

Zigbee2MQTT:debug 2023-11-06 15:04:39: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2023-11-06 15:04:39: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-11-06.15-04-39' filename: log.txt
Zigbee2MQTT:info  2023-11-06 15:04:39: Starting Zigbee2MQTT version 1.33.2 (commit #9996c93)
Zigbee2MQTT:info  2023-11-06 15:04:39: Starting zigbee-herdsman (0.21.0)
Zigbee2MQTT:debug 2023-11-06 15:04:39: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[57,103,242,211,186,70,200,193],"networkKey":"HIDDEN","panID":3798},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2023-11-06 15:04:43: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-11-06 15:04:43: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:debug 2023-11-06 15:04:43: Zigbee network parameters: {"channel":11,"extendedPanID":57,"panID":3798}
Zigbee2MQTT:info  2023-11-06 15:04:43: Currently 0 devices are joined:
Zigbee2MQTT:info  2023-11-06 15:04:43: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2023-11-06 15:04:43: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:debug 2023-11-06 15:04:43: Using MQTT anonymous login
Zigbee2MQTT:info  2023-11-06 15:04:43: Connected to MQTT server
Zigbee2MQTT:info  2023-11-06 15:04:43: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-11-06 15:04:43: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2023-11-06 15:04:43: Zigbee2MQTT started!
Zigbee2MQTT:debug 2023-11-06 15:04:57: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"supke-1","value":true}'
Zigbee2MQTT:info  2023-11-06 15:04:57: Zigbee: allowing new devices to join.
Zigbee2MQTT:debug 2023-11-06 15:04:57: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,3,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Zigbee2MQTT:info  2023-11-06 15:04:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"supke-1"}'
Zigbee2MQTT:info  2023-11-06 15:05:03: Device '0x0015bc0036001c01' joined
Zigbee2MQTT:info  2023-11-06 15:05:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0036001c01","ieee_address":"0x0015bc0036001c01"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-11-06 15:05:03: Starting interview of '0x0015bc0036001c01'
Zigbee2MQTT:info  2023-11-06 15:05:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0036001c01","ieee_address":"0x0015bc0036001c01","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:error 2023-11-06 15:06:04: Failed to interview '0x0015bc0036001c01', device has not successfully been paired
Zigbee2MQTT:info  2023-11-06 15:06:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0036001c01","ieee_address":"0x0015bc0036001c01","status":"failed"},"type":"device_interview"}'

I installed Zigbee2MQTT native (no add-on)
I am using a Sonoff Dongle Type E

cellerich, 2023-11-06:
Do you have the `adapter` setting in your `configuration.yaml`

```yaml
serial:
  port: /dev/ttyACM0
  adapter: ezsp
```

jessegabriels, 2023-11-06:
[image]

closed 2024-06-04 by github-actions[bot]