Koenkk/zigbee2mqtt #19572: E12-N1E - OTA not working

fireheadman, 2023-11-05:
### What happened?

Just installed zigbee2mqtt this evening and went around the house resetting and adding all my zigbee devices.   Everything came right in on the first try, so I'm happy with that!   Reaction times are lightning fast!

My adapter is a Nortec [HUSBZB-1] and I updated it to the latest firmware last week v6.7.8

**Here's my issue:**
Have 17x -  E12-N1E (Sengled) bulbs installed.
15x of them are on v30
2x are on v21

Home assistant is attempting to update them.    I have no way to stop the update, can't skip it either.
Have rebooted the entire server and it goes right back to OTA updates.
I also have OTA updates disabled, so not sure what I can do to prevent this?
[image]

Checking the guides and compatibility, the E12-N1E (Sengled) bulbs has been supported for a while now on OTA.

[image]
[image]

[image]

[image]

### What did you expect to happen?

expected the OTA operation to update the firmware

### How to reproduce it (minimal and precise)

reboot server or restart home assistant (docker)

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Nortec [HUSBZB-1

fireheadman, 2023-11-05:
an after thought.....
If you or anyone knows the country where sengled updates/firmware might be coming from, it could be possible my firewall is block it.    If/once I know that country I can check to see if that is the case and make an exception to allow it in

fireheadman, 2023-11-05:
UPDATE:   went ahead and temporarily disable all blocking by country on my firewall (Ubiquiti UDM Pro), then restarted HA and the Zigbee2Mqtt addon.    The issue still persist.

If there is something else I can try, please let me know...
Else, let me know how I can disable OTA and get HA from trying to upgrade the firmware.

essegi65, 2023-11-05:
same problem but with device BTicino K4003C

lichtl, 2023-11-08:
having the same as well. it seems that even Z2M is not updating them HA seem to show them up incorrectly

lichtl, 2023-11-10:
Same issue as #19622 (even if this was raised earlier)

Koenkk, 2023-11-10:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/19597, should be fixed in the dev branch

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-11-10 by Koenkk

fireheadman, 2023-11-10:
curious how long it takes for changes in the "DEV" branch to hit the Production branch?

I'm more in this for stability than to be in a development cycle where it is possible for other issues to be present than the one I am most concerned about.

Koenkk, 2023-11-11:
Dev -> production = new release = every 1sth of the month

andreapx, 2023-11-19:
> [image]
> 
> same problem but with device BTicino K4003C

Was your problem resolved with the update?
@Koenkk I'm on the latest dev branch (1.33.2-dev commit: fd19c86 with docker) but my bTicino devices (switches, roller shutters, plugs) don't updates, I still get the error
`Device didn't respond to OTA request`

fireheadman, 2023-11-27:
I'm see this in Z2Q, looking good so far.
Will update this with final screenshots once it completes
[image]

[image]

[image]

fireheadman, 2023-11-27:
Shows 100%
[image]

takes a few more seconds to complete, then will see this
[image]

After:
[image]

Looks good... no more "Update device firmware" for this bulb
[image]

fireheadman, 2023-11-27:
Lastly... if you enabled the Updater in HA, it will just show this while it is updating  (no change here, so will need to review M2Q for a status on the FW update)
[image]

[image]

Think this can be closed out since it pertains to the OP Subject.